home-assistant/operating-system #1199: Rebooting Host (NUC compatible) results in shutdown instead

LucidityCrash, 2021-02-01:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

Device = Chuwi herobox

**Home Assistant OS release:**
- [x] Fresh installation of release 5.10
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | 2021.1.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.86
arch | x86_64
timezone | UTC

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4835
Installed Version | 1.10.1
Stage | running
Available Repositories | 709
Installed Repositories | 8

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.10
-- | --
update_channel | stable
supervisor_version | 2021.01.7
docker_version | 19.03.13
disk_total | 234.0 GB
disk_used | 4.7 GB
healthy | true
supported | true
board | intel-nuc
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (1.15.3), RPC Shutdown (2.2), File editor (5.2.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
mode | auto-gen
resources | 4

</details>

**Supervisor logs:**

N/A

**Journal logs:**

 
Added as attachments - 
Initial_Boot was the first power up
Reboot was after a reboot command was issued
After_Power_Up was after I then pressed the power button to turn on the host.

[Initial_Boot.txt](https://github.com/home-assistant/operating-system/files/5903416/Initial_Boot.txt)
[Reboot.txt](https://github.com/home-assistant/operating-system/files/5903418/Reboot.txt)
[After_Power_Cycle.txt](https://github.com/home-assistant/operating-system/files/5903419/After_Power_Cycle.txt)

**Kernel logs:**

Only able to get this for a successful boot after a power cycle but is attached
[dmesg.txt](https://github.com/home-assistant/operating-system/files/5903420/dmesg.txt)

**Description of problem:**

I've migrated from an unsupported Supervised install on Ubuntu to HomeAssistant OS and whenever I reboot the host using either the supervisor webui or at the debug ssh session on port 22222 just using "reboot" the device reboots but then detects the Power Button being pressed as it is in the process of rebooting and then just shuts down.
I've tried installing a base Ubuntu OS to see if the same happens but it works as you would expect with a reboot working normally and only shutting down when the power button is actually pressed.
I've tried reimaging the device and the issue is the same.

agners, 2021-02-02:
> the device reboots but then detects the Power Button being pressed as it is in the process of rebooting and then just shuts down.

How do you come to this conclusion? I can't see something in the log which indicates that this is the case.

I did a quick test on my end, when pressing the power button on my NUC I get this information in `journalctl`:

```
Feb 02 13:50:11 ha-intel-nuc systemd-logind[294]: Power key pressed.
Feb 02 13:50:11 ha-intel-nuc systemd-logind[294]: Powering Off...
Feb 02 13:50:11 ha-intel-nuc acpid[203355]: System power button was pressed - shutting down...
Feb 02 13:50:11 ha-intel-nuc systemd-logind[294]: System is powering down.
```

In your case the reboot seems to be initiated from the Supervisor. It also seems that a Snapshot restore has been executed? I think a reboot after restore is intentional...
```
Feb 01 09:02:25 homeassistant 5bbe24019bf8[397]: 21-02-01 09:02:25 INFO (MainThread) [supervisor.snapshots] Full-Restore 1db14188 done
Feb 01 09:02:25 homeassistant hassos-supervisor[3811]: 21-02-01 09:02:25 INFO (MainThread) [supervisor.snapshots] Full-Restore 1db14188 done
...
Feb 01 09:06:57 homeassistant 5bbe24019bf8[397]: 21-02-01 09:06:57 INFO (MainThread) [supervisor.host.control] Initialize host reboot using logind
```

LucidityCrash, 2021-02-03:
> How do you come to this conclusion? I can't see something in the log which indicates that this is the case.

Oops got the files round the wrong way when I named them.  Initial_boot is the reboot and Reboot is the Initial Boot ... sorry.

line 1133 in this file : 
[Initial_Boot.txt](https://github.com/home-assistant/operating-system/files/5903416/Initial_Boot.txt)

`Feb 01 09:08:53 homeassistant systemd-logind[322]: Power key pressed.`

agners, 2021-02-03:
Hm, I see, interesting. The fact that `acpid` as well as `systemd-logind` sees the event seems to indicate that the kernel indeed reports a power key event. But its weird that Ubuntu seems not to see/get such events. Maybe only the kernel version we are using right now is affected...

What Ubuntu version (and kernel) did you test? 

I will update the development branch to Linux 5.10 soon, once a development build is ready a test using that version would be interesting.

LucidityCrash, 2021-02-03:
It was actually a very old live USB key I had lying about that I use in emergencies :) - 18.04 LTS so 4.15 or at most 5.3

agners, 2021-02-17:
There are development builds using the 5.10 kernel available now. If you like, you can test by changing to the development channel:

```
ha supervisor options --channel dev
ha supervisor reload
ha os update --version 6.0.dev20210209
# Don't forget to switch back to stable.
ha supervisor options --channel stable
```

Another idea is maybe its actually a bug in the BIOS/firmware of that device? Maybe a BIOS/firmware update helps?

LucidityCrash, 2021-02-17:
I've acutually just returned the device ... the support from the manufacturer was pitiful.  There was no way in the BIOS to turn on after a power failure, I'd asked if there was a BIOS upgrade that would enable it ... no response for 2 weeks and then finally a "no it can't do that" ... so back it has gone ... I'll close this off as there is nothing to do any more.

closed 2021-02-17 by LucidityCrash