home-assistant/operating-system #1201: OS 5.11 freezes my Pi4 on SSD

bschatzow, 2021-02-03:
**Hardware Environment**

- [x ] Raspberry Pi [4] 4G Kingston 120G SSD, StarTech controller
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x ] Updated from version 5.11
- Additional information (if accessible):

➜  ~ ha info
This report was generated after the last crash and the return to 5.3
arch: aarch64
channel: stable
docker: 19.03.11
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.3"
homeassistant: 2021.1.5
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: HassOS 5.3
supervisor: 2021.01.7
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: America/New_York

**Supervisor logs:**

No errors in SU or CORE logs
**Journal logs:**

c9', 'pin': '', 'name': ''}, {'_id': '5daf274b8c9d263151bd3bc8', 'pin': '', 'name': ''}, {'_id': '5daf274b8c9d263151bd3bc7', 'pin': '', 'name': ''}, {'_id': '5daf274b8c9d263151bd3bc6', 'pin': '', 'name': ''}], 'duress': {'pin': ''}, 'master': {'pin': '0622'}}}, 'basestationStatus': {'lastUpdated': '2021-02-02T17:25:13.012Z', 'rfJamming': False, 'ethernetStatus': 4, 'gsmRssi': -51, 'gsmStatus': 3, 'backupBattery': 5732, 'wallPower': 6414, 'wifiRssi': -37, 'wifiStatus': 1, '_id': '60198af976ebb81de26040c0', 'encryptionErrors': []}, 'lastUpdated': 1612286713}
Feb 02 17:25:16 homeassistant 118990b09036[375]: 2021-02-02 12:25:16 INFO (MainThread) [engineio.client] Sending packet PING data None
Feb 02 17:25:16 homeassistant 118990b09036[375]: 2021-02-02 12:25:16 INFO (MainThread) [engineio.client] Received packet PONG data None
Feb 02 17:25:16 homeassistant 118990b09036[375]: 2021-02-02 12:25:16 DEBUG (Thread-LeaseMaintainer) [google.cloud.pubsub_v1.subscriber._protocol.leaser] The current deadline value is 10 seconds.
Feb 02 17:25:16 homeassistant 118990b09036[375]: 2021-02-02 12:25:16 DEBUG (Thread-LeaseMaintainer) [google.cloud.pubsub_v1.subscriber._protocol.leaser] Snoozing lease management for 7.475591 seconds.
Feb 02 17:25:17 homeassistant 118990b09036[375]: 2021-02-02 12:25:17 DEBUG (MainThread) [simplipy] Data received from /ss3/subscriptions/2785102/sensors: {'account': '001138f7', 'success': True, 'sensors': [{'serial': '008caba3', 'type': 1, 'status': {}, 'name': 'Master BR', 'deviceGroupID': 0, 'setting': {'lowPowerMode': False, 'alarm': 1}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '008cb08f', 'type': 1, 'status': {}, 'name': 'Mudroom', 'deviceGroupID': 0, 'setting': {'lowPowerMode': False, 'alarm': 1}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '005d4393', 'type': 2, 'status': {}, 'name': 'Bill', 'deviceGroupID': 0, 'setting': {'disarmEnabled': True, 'alarm': 1}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '005d43b7', 'type': 2, 'status': {}, 'name': 'Charlotte', 'deviceGroupID': 0, 'setting': {'disarmEnabled': True, 'alarm': 1}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '007bf4db', 'type': 4, 'status': {}, 'name': 'Front Door', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 0, 'home': 0, 'off': 2}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '007bdc3b', 'type': 4, 'status': {}, 'name': 'Garage', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 0, 'home': 2, 'off': 2}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '007bf95e', 'type': 4, 'status': {}, 'name': 'Hallway', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 0, 'home': 0, 'off': 0}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '007947c7', 'type': 4, 'status': {}, 'name': 'Living Room', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 0, 'home': 0, 'off': 0}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '0076781d', 'type': 5, 'status': {'triggered': False}, 'name': 'Front Door', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 1, 'home': 1, 'off': 2}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '00726ece', 'type': 5, 'status': {'triggered': False}, 'name': 'Garage', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 1, 'home': 1, 'off': 2}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '00767944', 'type': 5, 'status': {'triggered': False}, 'name': 'GarageBack', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 1, 'home': 1, 'off': 2}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '00a4468f', 'type': 5, 'status': {'triggered': False}, 'name': 'Kitchen Deck', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 0, 'away': 1, 'home2': 0, 'home': 1, 'off': 0}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 1, 'home': 2, 'off': 2}, 'deviceGroupID': 0, 'name': 'Master Closet', 'status': {'triggered': True}, 'type': 5, 'serial': '0076770b', 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}, {'serial': '00767816', 'type': 5, 'status': {'triggered': False}, 'name': 'Mudroom', 'deviceGroupID': 0, 'setting': {'instantTrigger': False, 'away2': 1, 'away': 1, 'home2': 1, 'home': 1, 'off': 0}, 'flags': {'swingerShutdown': False, 'lowBattery': False, 'offline': False}}], 'lastUpdated': 1612286717, 'lastSynced': 1612286717, 'lastStatusUpdate': 1612286715}
Feb 02 17:25:17 homeassistant 118990b09036[375]: 2021-02-02 12:25:17 DEBUG (MainThread) [homeassistant.components.simplisafe] Finished fetching bschatzow@gmail.com data in 5.816 seconds
Feb 02 17:25:18 homeassistant 118990b09036[375]: 2021-02-02 12:25:18 ERROR (Recorder) [homeassistant.components.recorder] Error executing query: (MySQLdb._exceptions.OperationalError) (1366, "Incorrect string value: '\\xF0\\x9F\\x9A\\xAB h...' for column `homeassistant`.`states`.`state` at row 1")
Feb 02 17:25:18 homeassistant 118990b09036[375]: [SQL: INSERT INTO states (domain, entity_id, state, attributes, event_id, last_changed, last_updated, created, old_state_id) VALUES (%s, %s, %s, %s, %s, %s, %s, %s, %s)]
Feb 02 17:25:18 homeassistant 118990b09036[375]: [parameters: ('sensor', 'sensor.sm_t813_last_notification', 'No stick, no problem for Shesty. 🚫 https://t.co/gFghirXilz', '{"android.appInfo": "ApplicationInfo{e27102f com.twitter.android}", "android.bigText": "No stick, no problem for Shesty. \\ud83d\\udeab https://t.co/ ... (585 characters truncated) ... ongoing": false, "package": "com.twitter.android", "post_time": 1612280297856, "friendly_name": "SM-T813 Last Notification", "icon": "mdi:bell-ring"}', 23095025, datetime.datetime(2021, 2, 2, 15, 38, 21, 132989, tzinfo=datetime.timezone.utc), datetime.datetime(2021, 2, 2, 15, 38, 21, 132989, tzinfo=datetime.timezone.utc), datetime.datetime(2021, 2, 2, 15, 38, 21, 132989, tzinfo=datetime.timezone.utc), 22529671)]
Feb 02 17:25:18 homeassistant 118990b09036[375]: (Background on this error at: http://sqlalche.me/e/13/e3q8)
Feb 02 17:25:18 homeassistant 118990b09036[375]: 2021-02-02 12:25:18 ERROR (Recorder) [homeassistant.components.recorder] Error in database connectivity during commit: (MySQLdb._exceptions.OperationalError) (1366, "Incorrect string value: '\\xF0\\x9F\\x9A\\xAB h...' for column `homeassistant`.`states`.`state` at row 1")
Feb 02 17:25:18 homeassistant 118990b09036[375]: [SQL: INSERT INTO states (domain, entity_id, state, attributes, event_id, last_changed, last_updated, created, old_state_id) VALUES (%s, %s, %s, %s, %s, %s, %s, %s, %s)]
Feb 02 17:25:18 homeassistant 118990b09036[375]: [parameters: ('sensor', 'sensor.sm_t813_last_notification', 'No stick, no problem for Shesty. 🚫 https://t.co/gFghirXilz', '{"android.appInfo": "ApplicationInfo{e27102f com.twitter.android}", "android.bigText": "No stick, no problem for Shesty. \\ud83d\\udeab https://t.co/ ... (585 characte [... cut ...]
[ 2951.453347] mmc0: sdhci: Resp[0]:   0x00000000 | Resp[1]:  0x00000000
[ 2951.455962] mmc0: sdhci: Resp[2]:   0x00000000 | Resp[3]:  0x00000000
[ 2951.458489] mmc0: sdhci: Host ctl2: 0x00000000
[ 2951.461036] mmc0: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x00000000
[ 2951.463623] mmc0: sdhci: ============================================
#

**Description of problem:**

All versions of the OS after 5.3 cause my Pi4 on SSD to freeze anytime from 1 hour to a couple of days.  The identical system on 5.2 or 5.3 has not had any issues.

alex88, 2021-02-03:
This is probably a duplicate of #1119, quite a few people are experiencing this bug

bschatzow, 2021-02-03:
Ii realize this.  was told to open a new one.

agners, 2021-02-03:
For the 

```
mmc0: Timeout waiting for hardware cmd interrupt.
```

messages a fix is available, see: https://github.com/home-assistant/operating-system/tree/dev/Documentation/boards/raspberrypi#usb-boot

Do you experience problems even after setting `sd_poll_once=on`?

agners, 2021-02-03:
The Linux kernel has been updated to Linux 5.4 with HAOS release 5.5, and I think some problems might be related to that. But it sounds like HAOS 5.4 already causes problems for you? If the above does not help, can you recheck if it starts with HAOS 5.4 or 5.5?

bschatzow, 2021-02-03:
I did not try 5.4, but others in the 1119 did and it did not work for them. I read the changes and it looked like the core was changed on 5.4 not 5.5 (unless I read the notes incorrectly, aways a possibility.) I'm willing to test anything if it helps get to the root cause and corrective action.  By the way a spreadsheet was created on 1119 where users are putting in all their information. If you get a chance can you take a look and add a column for any additional information you need.

agners, 2021-02-03:
> Are you saying to change this sd_poll_once=on? This seems related to boot on SSD, which is not my issue.

But you are booting from SSD right? Or do you have a SD card as well?

If you do boot from SD, then all this `mmc0: Timeout waiting for hardware cmd interrupt.` messages should disappear with `sd_poll_once=on`. It could be that this is the root cause of the problems you are seeing.

> I am also confuse on when the kernel was changed It shows it on HassOS Release-4 build 16 (Stable).

HAOS updates the kernel quite often, but often its only minor security/bug fixes. However in HA OS release 5.5 we moved from Linux 4.19 to 5.4 for all Raspberry Pi's, so this can change behavior etc...

bschatzow, 2021-02-03:
Only boot Home Assistant on SSD. If I need to update the firmware I use a SD card. I have updated to 5.4.  If this works I can try the mmc0 and then try the 5.11 again. I don't understand why this change is needed for the kernel change?

bschatzow, 2021-02-03:
I figured out how to change the config.txt file and added dtparam=sd_poll_once=on.
OS 5.4 seemed fine before the change.  I just made the change then upgraded the OS to 5.11 at 6PM EST.  I'll let you know if the config.txt change fixes the issue. 
Thanks for the help.

bschatzow, 2021-02-04:
hardware cmd interrupt. messages should disappear with sd_poll_once=on. It could be that this is the root cause of the problems you are seeing.
@agners 
Made the changes and system froze after 7 hrs 57 minutes. Going back to 5.4. No errors in any logs. Seems unrelated to SSD and controller.
Not sure what to try next.

bschatzow, 2021-02-04:
New dmseg

# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.83-v8 (builder@465349041c38) (gcc version 9.3.0 (Buildroot -g1a551c07)) #1 SMP PREEMPT Sun Jan 31 04:08:28 UTC 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012480
[    0.000000]   DMA zone: 3788 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242432 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 31 pages/cpu s89432 r8192 d29352 u126976
[    0.000000] pcpu-alloc: s89432 r8192 d29352 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996660
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=4247b5e639ba4e3887466667a2fb6eaf cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=592 bcm2708_fb.fbheight=448 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:75:DC:C6 vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37300000-0x3b300000] (64MB)
[    0.000000] Memory: 3819772K/4049920K available (11004K kernel code, 1176K rwdata, 3744K rodata, 1088K init, 1232K bss, 164612K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36308 entries in 142 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x320/0x4b0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000223] Console: colour dummy device 80x25
[    0.000649] printk: console [tty1] enabled
[    0.000705] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000738] pid_max: default: 32768 minimum: 301
[    0.000875] LSM: Security Framework initializing
[    0.001001] AppArmor: AppArmor initialized
[    0.001207] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001277] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.003547] ASID allocator initialised with 32768 entries
[    0.003756] rcu: Hierarchical SRCU implementation.
[    0.004470] EFI services will not be available.
[    0.004926] smp: Bringing up secondary CPUs ...
[    0.005804] Detected PIPT I-cache on CPU1
[    0.005861] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.006846] Detected PIPT I-cache on CPU2
[    0.006884] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.007802] Detected PIPT I-cache on CPU3
[    0.007841] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.007974] smp: Brought up 1 node, 4 CPUs
[    0.008061] SMP: Total of 4 processors activated.
[    0.008082] CPU features: detected: 32-bit EL0 Support
[    0.008105] CPU features: detected: CRC32 instructions
[    0.033608] CPU: All CPU(s) started at EL2
[    0.033677] alternatives: patching kernel code
[    0.035268] devtmpfs: initialized
[    0.045214] Enabled cp15_barrier support
[    0.045265] Enabled setend support
[    0.045672] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[... 552 lines cut ...]
[  255.640989] hassio: port 13(veth4301d91) entered disabled state
[  255.642061] device veth4301d91 entered promiscuous mode
[  256.508760] eth0: renamed from vethe5d760a
[  256.526091] IPv6: ADDRCONF(NETDEV_CHANGE): veth4301d91: link becomes ready
[  256.526230] hassio: port 13(veth4301d91) entered blocking state
[  256.526237] hassio: port 13(veth4301d91) entered forwarding state
[  257.473740] hassio: port 14(veth57903b1) entered blocking state
[... cut ...]
[  258.384660] audit: type=1325 audit(1612429643.966:284): table=nat family=2 entries=12
[  258.390879] audit: type=1325 audit(1612429643.970:285): table=nat family=2 entries=13
[  258.397498] audit: type=1325 audit(1612429643.978:286): table=nat family=2 entries=14
[  258.436938] eth0: renamed from vethbecdfbb
[  258.461257] IPv6: ADDRCONF(NETDEV_CHANGE): veth57903b1: link becomes ready
[  258.461386] hassio: port 14(veth57903b1) entered blocking state
[  258.461392] hassio: port 14(veth57903b1) entered forwarding state
#

bschatzow, 2021-02-04:
New journalctl --since "2021-02-04 02:00:00" --until "2021-02-04 10:00:00" >/mnt/data/supervisor/homeassistant/jour2.txt  2>&1
[jour2.txt](https://github.com/home-assistant/operating-system/files/5924666/jour2.txt)

This is the 8 hour span it was running.  Looks link it lost the network at the end?

agners, 2021-02-04:
Hm, I don't see anything unusual in that log. Btw, you can use `journalctl -b -1` to display the complete log of the last boot.

It really seems like the system just freezes for some reason. What would be interesting if a Raspberry Pi OS installation on the SSD behaves the same way (in HAOS 5.10 we use the very same Linux kernel version as the December release of Raspberry Pi OS, and in the HAOS 5.11 release we use the same kernel as in their January release).

Maybe also worth a try: Is the 32-bit variant stable?

bschatzow, 2021-02-04:
@agners 
> Hm, I don't see anything unusual in that log. Btw, you can use `journalctl -b -1` to display the complete log of the last boot.
I understood this.  I didn't want you to have to go through more than what was necessary.  

> It really seems like the system just freezes for some reason. What would be interesting if a Raspberry Pi OS installation on the SSD behaves the same way (in HAOS 5.10 we use the very same Linux [...]
I only use the Raspberry PI OS to update the EEPROM.  Very rare and not up for more than a couple of minutes.  Also I use the 32 bit version.  I need the 64bit version version on Home Assistant as I was told this is the only version that supports SSD boot.
Is there a way to get a custom with all the OS updates except using the Kernel from prior to 5.5? 

In the spreadsheet in the 1119 issue there are currently 10 users.  All but 2 are using the 64bit version.  I had no issues with the 32 bit on the SD card and OS 5.8 (last that I tried before updating to SSD).

bschatzow, 2021-02-07:
Just had 5.4 freeze.  On terminal it says wlan link is not ready.  Disconnected lan cable reconnect and it works again.

agners, 2021-02-08:
So, that is not a freeze, but just a networking issue then?

bschatzow, 2021-02-08:
On 5.4 it was a network issue.  Never had this with 5.2 or 5.3.  This happened on day four.  5.3 never had this issue. On 5.8 and above it is a freeze.  No activity.  Plug /unplug does nothing.  System is dead except for the network light on the pi blinking rapidly.  Only recovery is remove power.

bschatzow, 2021-02-08:
I don't believe this is the proper place to ask.  PM me  and I'll try to help.  You can also a ask in the forums or discord.

agners, 2021-02-08:
@bschatzow that HAOS 5.4 does not show the freeze (but a different network issue) makes the theory that it is related to Linux 5.4 more plausible.

Yes you can compile Home Assistant OS on your end to test things. It takes a while since it downloads and builds everything from scratch (depending on your connectivity and machine 30min to several hours). There is a [getting started development](https://github.com/home-assistant/operating-system/blob/dev/Documentation/getting_started_development.md) file in the Documentation section.

Make sure to checkout branch `rel-5` and revert the kernel bump commit (846512a2830b96ddd897533d458b82c85ccf6c16).

bschatzow, 2021-02-08:
> Yes you can compile Home Assistant OS on your end to test things. It takes a while since it downloads and builds everything from scratch (depending on your connectivity and machine 30min to several hours). There is a [getting started development](https://github.com/home-assistant/operating-system/blob/dev/Documentation/getting_started_development.md) file in the Documentation section.
> 
> Make sure to checkout branch `rel-5` and revert the RPi kernel bump commits ([846512a](https://github.com/home-assistant/operating-system/commit/846512a2830b96ddd897533d458b82c85ccf6c16) and friends).

Thanks.  A couple of problems, I do not have a Linux system and a bigger one, this requires a lot more knowledge than I currently have.  I was hoping there was a way to build it on the github server with a script file and than download it when completed.  Similar to my custom Android files that I download on the lineage servers.  I'll ask around the "1119" issue and see if anyone has knowledge there and a system to build it on.

bschatzow, 2021-02-10:
@agners I see that the Pi OS is moving to 5.10 .  As far as I know the current release is using 5.83 or 5.84.  Is there a beta available that is using the newer 5.10 kernel?  If so, I'll install it and see if it fixes my freeze issue.
Thanks.

agners, 2021-02-11:
> the current release is using 5.83 or 5.84

There is a 4 missing, 5.4.83/5.4.84.

Moving the RPi's to 5.10 is on my todo list, probably next week.

bschatzow, 2021-02-11:
> > the current release is using 5.83 or 5.84
> 
> There is a 4 missing, 5.4.83/5.4.84.
> 
> Moving the RPi's to 5.10 is on my todo list, probably next week.
I would be happy to beta test and capture information for you.  If you need others, I can ask on issue 1119. Let me know.

bschatzow, 2021-02-17:
@agners I see that there is a new beta for various hardware but none for the RPI4.  Is this in the works?  Let me know when it is available and I will test it.

bschatzow, 2021-02-18:
@agners ,
I see version 5.12 is now released as stable.  I followed the release notes and it doesn't look like the kernel was updated like the other hardware was.  I wanted to try it anyway and I get an error; Error: Unknown error, see supervisor.  I loaded the beta supervisor and tried again.  Got the same error.

bschatzow, 2021-02-18:
@agners - 5.12 just froze at 6 hours.  No log errors.  Same as before.

bschatzow, 2021-02-26:
@agners,
Is there a developers version you would like me to try to see if this fixes the freeze issue?

bschatzow, 2021-03-01:
@agners ,
I have setup my system with the split configuration and so far no freezes with the latest OS, pi4 Feb firmware and the latest Mar Home assistant beta.  If this freezes I'll open a new issue and capture the logs.  I kept the old SSD so if you want me to test something with just the the SSD connected, I can.  Thanks for your help.

closed 2021-03-01 by bschatzow