home-assistant/operating-system #1203: Boot hang after reboot on Raspberry Pi 4 (2GB) 32-bit installation

agners, 2021-02-03:
**Hardware Environment**

- [x] Raspberry Pi 4 (32-bit installation)
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.6 -> 5.10 -> 5.11
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

U-Boot seems to hang on reboot
```
U-Boot 2020.10 (Jan 01 2021 - 16:45:27 +0000)

DRAM:  (fixing) 1.9 GiB
RPI 4 Model B (0xb03112)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Host not halted after 16000 microseconds.
<freeze>
```

Mariusthvdb, 2021-02-09:
is this an issue to worry about before clicking the update button? going from 5.10 to 5.11
Don't want to brick the setups yet again ...
thx.

agners, 2021-02-17:
It happened to me only on reboot. Power cycling afterwards makes it boot without a problem, so its not bricking it, just not coming up by itself. But only sometimes, which makes the bug very hard to reproduce and fix...

agners, 2021-09-03:
Most likely resolved by the same patch which fixes #1490.

closed 2021-09-03 by agners