home-assistant/operating-system #1206: Outdated documentation for changing NTP configuration

hubbergit, 2021-02-03:
Hi,

the documentation describing how to change the NTP configuration (https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md) seems to be outdated.

> You can manual edit the systemd timesync file on `/etc/systemd/timesyncd.conf`

There is no file `/etc/systemd/timesyncd.conf` on my system. There is even no directory `/etc/systemd`.

I access the root file system using the SSH & Web Terminal add-on provided by the community.

**Hardware Environment**
- Raspberry Pi 3+

**Home Assistant OS release:**
- Fresh installation of release core-2021.1.5

Thank you in advance for looking into this.

agners, 2021-02-04:
The "SSH & Web Terminal add-on" does not give you root file system access. For that you need to use the local console or via SSH port 22222 (see https://developers.home-assistant.io/docs/operating-system/debugging/).

closed 2021-02-04 by agners

hubbergit, 2021-02-04:
Thank you for this clarification. I was not aware that the add-on works this way.