home-assistant/operating-system #1207: Check Configuration: Fatal error while loading config: [Errno 2] No such file or directory:

woodmj74, 2021-02-04:
**Hardware Environment**

- ODROID N2(+)

**Home Assistant OS release:**
- Updated from version 5.10
- Additional information (if accessible):
- issue occurred when moving to 2020.2 (may not be an os issue, please advise and will raise in the appropriate place:-) )

## System Health

version | core-2021.2.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.9.16
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4750
Installed Version | 1.10.1
Stage | running
Available Repositories | 720
Installed Repositories | 20

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 14 February 2021, 0:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.10
-- | --
update_channel | stable
supervisor_version | supervisor-2021.01.7
docker_version | 19.03.13
disk_total | 28.0 GB
disk_used | 7.9 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Samba share (9.3.0), Terminal & SSH (9.0.0), deCONZ (6.6.4), Home Assistant Google Drive Backup (0.103.1), Check Home Assistant configuration (3.6.0), Mosquitto broker (5.1), Log Viewer (0.9.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 4
-- | --
resources | 15
views | 15
mode | storage

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Hopefully I've put this in the correct place - if not please let me know and I'll happily raise in the appropriate location.

Prior to upgrading from 2020.1.5 to 2020.02 ran Check Home Assistant configuration which threw the following errors. Raised in OS as it seems to referencing directories on the machine and (forgive my lack of knowledge) wasn't sure is this was managed by the OS rather than Core. The following log was written:

INFO:homeassistant.util.package:Attempting install of colorlog==4.6.2
[01mTesting configuration at /tmp/config[0m
INFO:homeassistant.util.package:Attempting install of aiohttp_cors==0.7.0
INFO:homeassistant.util.package:Attempting install of zeroconf==0.28.8
INFO:homeassistant.util.package:Attempting install of mutagen==1.45.1
INFO:homeassistant.util.package:Attempting install of gTTS==2.2.1
INFO:homeassistant.util.package:Attempting install of bimmer_connected==0.7.14
[31mFatal error while loading config:[0m [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/chardet-4.0.0.dist-info/METADATA'
[01;37mFailed config[0m
  [01;31mGeneral Errors:[0m [31m [31m
    - [Errno 2] No such file or directory: '/usr/local/lib/python3.8/site-packages/chardet-4.0.0.dist-info/METADATA'
[0m
[01;37mSuccessful config (partial)[0m

Thank you for your assistance and ongoing awesome work!

agners, 2021-02-04:
Hm, that seems like a problem with the Core container, a corruption of some sort. Can you install the SSH/Terminal Add-On and use `ha core update --version 2021.2.0` or/and `ha core rebuild`.

woodmj74, 2021-02-04:
Hi @agners - thanks for the comments. I've run `ha core update --version 2021.2.0` (same error) followed by `ha core rebuild`. The original error appears to have been resolved but am now throwing a different one.

Not sure if this is related (or nothing to worry about, or something to raise elsewhere). Error below, and thanks again.

```
[15:20:27] ERROR: The configuration check did not pass!
[15:20:27] ERROR: See the output below for more details.
Testing configuration at /tmp/config
Fatal error while loading config: (idna 3.1 (/usr/local/lib/python3.8/site-packages), Requirement.parse('idna<3,>=2.5'), {'requests'})
Failed config
  General Errors: 
    - (idna 3.1 (/usr/local/lib/python3.8/site-packages), Requirement.parse('idna<3,>=2.5'), {'requests'})

Successful config (partial)
[15:20:27] INFO: The full output has been written to /share/check_config.txt
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
```