home-assistant/operating-system #1208: Rpi PoE hat goes brrrrr on Rpi 4b

majkrzak, 2021-02-04:
**Hardware Environment**

- [*] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [*] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

```
[...] rpi-poe-fan rpi-poe-fan@0: Failed to get default PWM value: -5
[...] rpi-poe-fan: probe of rpi-poe-fan@0 failed with error -5
```

**Description of problem:**
_
Fan of RPi PoE hat intalled on top of Raspbery Pi 4 B runs at full speed no matter what options are set in config.txt.

agners, 2021-02-04:
What configs did you try in `config.txt`? Which version of Home Assistant OS are you using?

We are using the same kernel as well as the same overlays as Raspberry Pi OS, so if it works with Raspberry Pi, it really also should work with HAOS...

majkrzak, 2021-02-04:
Latest image downloaded from the https://github.com/home-assistant/operating-system/releases/download/5.10/hassos_rpi4-5.10.img.xz

In config.txt I was checking various setups, but with same result. The last one was:

```
dtoverlay=rpi-poe,poe_fan_temp0=65000,poe_fan_temp0_hyst=1000,poe_fan_temp1=70000,poe_fan_temp1_hyst=2500,poe_fan_temp2=80000,poe_fan_temp2_hyst=5000,poe_fan_temp3=90000,poe_fan_temp3_hyst=5000
```

I managed to google some post describing similar problem, but solution there was to update boot loader package.

majkrzak, 2021-02-04:
Here is most recent similar issue from rpi forums:
https://www.raspberrypi.org/forums/viewtopic.php?t=298985

majkrzak, 2021-02-04:
It can be replicated with latest raspbian: https://downloads.raspberrypi.org/raspios_full_armhf/images/raspios_full_armhf-2021-01-12/2021-01-11-raspios-buster-armhf-full.zip

majkrzak, 2021-02-04:
Updating with current state of master does not help.

agners, 2021-02-04:
@majkrzak I guess then it is a problem with the overlay/Raspberry Pi Linux 5.4 kernel. I would report it with the Raspberry Pi folks. Not 100% which bug tracker this should go, but I think the overlays are part of the kernel tree it should be reported at https://github.com/raspberrypi/linux/issues.

agners, 2021-02-04:
Oh I see you already reported it in the firmware repo :+1:  I guess they can move if necessary.

majkrzak, 2021-02-06:
Hah, I'm confused now :D According to the ticket on raspberrypi/firmware it might be related to the commit: https://github.com/raspberrypi/firmware/commit/c78f3ef45229ab722ec6b858f39b078535d88bee which is from 15th of January, but hassoss 5.10 was released at 1st.

agners, 2021-02-08:
@majkrzak yeah 5.10 was using a version before that: https://github.com/home-assistant/operating-system/blob/5.10/buildroot/package/rpi-firmware/rpi-firmware.mk#L7

closed 2021-05-12 by majkrzak