home-assistant/operating-system #1216: HA Blue - HDMI-CEC not working

DCSBL, 2021-02-06:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release os-5.9.16, core-2021.2.1, supervisor-2021.01.7
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.2.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.9.16
arch | aarch64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4939
Installed Version | 1.10.1
Stage | running
Available Repositories | 709
Installed Repositories | 4

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.10
-- | --
update_channel | stable
supervisor_version | supervisor-2021.01.7
docker_version | 19.03.13
disk_total | 113.9 GB
disk_used | 11.0 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | CEC Scanner (2.4), deCONZ (6.6.2), Let's Encrypt (4.11.0), File editor (5.2.0), Home Assistant Google Drive Backup (0.103.1), AdGuard Home (3.0.0), FTP (4.0.0), SSH & Web Terminal (8.0.0)

</details>

<details><summary>Lovelace</summary>

[... 194 lines cut ...]

This issue is reported to the add-on repository (https://github.com/home-assistant/addons/issues/1771), but maybe it it related to the OS. To summarize:
After starting the CEC Scanner add-on, it immediately stops and has the following in the log file:
```
[12:41:49] INFO: Starting CEC client scan...
* failed to open vchiq instance
```

I can also see this in the Home Assistant log when I've added the `hdmi_cec` integration to `configuration.yaml`, and then calling the `hdmi_cec.power_on` service:
```
2021-02-06 22:26:08 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2021-02-06 22:26:09 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2021-02-06 22:26:10 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2021-02-06 22:26:11 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
etc..
```

I have:
- Tried with and without HDMI cable connected.
- Tried connecting to TV directly or though amplifier.
- Tried 'rebooting -> connecting cable' and 'connecting cable -> reboot' order.
- Tried different cables.
- When HDMI cable is connected, TV displays OS log messages (`Welcome to Home Assistant(\n) Homeassistant login: _`)
- Disconnected all other hardware than ethernet, power and HDMI cable, to be sure there is no interference.
- Removed ('official HA Blue') case to make sure cable is connected properly.

agners, 2021-02-08:
Thanks for the detailed report! Hm, it seems the the drivers are enabled and the HDMI CEC access is given, but for some reason its not properly communicating.

adeepn, 2021-02-12:
Seems, that docker containers lost write access to devices after some changes in supervisor/core

oramirite, 2021-02-15:
This is identical behavior to what I'm experiencing on my Pi 3B+. I just got this and am a fairly new HA user so I was trying to ensure it wasn't user error. CEC does seem to be active, because rebooting my Pi sends an input signal and switches my AVR over to HDMI5 where my Pi is plugged in.

As a Docker guy myself, is it possible this is related to containerizing everything, and that the CEC device isn't properly accessible? Just a stab in the dark...

oramirite, 2021-02-18:
I just ran a duplicate of the CEC extension container manually with an interactive terminal in Privileged mode. A CEC scan works fine when this is done.

A workaround could be to allow this extension to run in privileged mode via the Supervisor panel, but that's not ideal since you'd, like... need that for the extension to work :P

It sounds like there's just some kind of device path or privilege missing from the container to get full access to CEC.

DCSBL, 2021-05-12:
Well, I don't think it is fixed (yet)

andybjackson, 2021-05-14:
I have the same experience.

N36, 2021-05-23:
Same situation here on pi 3b+. HDMI cec suddenly stopped working recently. Caused HA to repeatedly reboot on loop. Had to remove hdmi_cec: from configuration.yaml to get HA stable again.

boszme, 2021-06-16:
I've made some research and I believe the current docker image is compiled only for RPI based devices: /dev/vchiq is very specific for RPI:
https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/misc/vc04_services/Kconfig
https://raspberrypi.stackexchange.com/a/54590
https://github.com/Pulse-Eight/libcec/blob/76551ea1dd9a55f0ce1533e440dc12dbc594f7ba/src/libcec/adapter/RPi/RPiCECAdapterDetection.cpp#L51

I think the CEC addon should be blocked from installing on non-RPI based installations and a separate docker image should be created for Odroid based devices.

Based on this my understanding is that libcec need to be compiled separately for the various OS/HWs setups, not sure if multiple detection modes could be added at the same time:
https://github.com/Pulse-Eight/libcec/blob/master/docs/README.linux.md

boszme, 2021-06-16:
Took a second look at this: I'm not a C/C++ expert, however it seems to me that libcec supports multiple detections to be compiled with; probably just need to recompile libcec with Odroid(even though I'm not sure yet which is the Odroid specific one) AND RPI specific cmake flags

https://github.com/Pulse-Eight/libcec/blob/76551ea1dd9a55f0ce1533e440dc12dbc594f7ba/src/libcec/adapter/AdapterFactory.cpp

boszme, 2021-06-16:
> I just ran a duplicate of the CEC extension container manually with an interactive terminal in Privileged mode. A CEC scan works fine when this is done.
>
> A workaround could be to allow this [...]

@oramirite you've tested this on RPI and NOT on Odroid, right?

boszme, 2021-06-16:
So it seems it is already compiled with multiple flags, however it is still failing on searching for the /dev/vchiq which is only exists on RPI:
https://github.com/home-assistant/addons/blob/master/cec_scan/Dockerfile#L35

strace output:

```
rt_sigprocmask(SIG_BLOCK, ~[RTMIN RT_1 RT_2], [], 8) = 0
clone(child_stack=0xffffaf127a70, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID|0x400000, parent_tid=[17], tls=0xffffaf127b88, child_tidptr=0xffffaf5540c4) = 17
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
futex(0xaaaae41b716c, FUTEX_WAKE_PRIVATE, 1) = 1
openat(AT_FDCWD, "/dev/vchiq", O_RDWR|O_LARGEFILE) = -1 ENOENT (No such file or directory)
writev(1, [{iov_base="* failed to open vchiq instance", iov_len=31}, {iov_base="\n", iov_len=1}], 2* failed to open vchiq instance
) = 32
exit_group(-1)                          = ?
+++ exited with 255 +++
bash-5.0#
```

boszme, 2021-06-16:
Found the root cause: it is failing due to the RPI userland code. The AdapterFactory.cpp doesn't check in runtime which OS/HW configuration is currently running on; if the RPI cmake flag is compiled in it will try to invoke those anyways.
However the bcm_host_init() function calls an `exit(-1);` if the initialization fails 🤯 which causing the issue on other OS/HW setups than RPI..

https://github.com/Pulse-Eight/libcec/blob/76551ea1dd9a55f0ce1533e440dc12dbc594f7ba/src/libcec/adapter/RPi/RPiCECAdapterCommunication.cpp#L458
https://github.com/raspberrypi/userland/blob/master/host_applications/linux/libs/bcm_host/bcm_host.c#L99

boszme, 2021-06-17:
Reported to libCEC project: https://github.com/Pulse-Eight/libcec/issues/572

agners, 2021-06-17:
@boszme nice find! So does returning in that case instead of exiting makes it work for other platforms?

It looks that libcec is rather stale lately. Are there maybe forks of it which are also better suited for other hardware?

agners, 2021-06-17:
Btw, we do compile libcec in our base container, see: https://github.com/home-assistant/docker/blob/master/Dockerfile#L116

So if a patch fixes the problem, we can add it there for now.

boszme, 2021-06-20:
I've checked if I can fix the code with some patch, but it seems way more complicated than I thought (or I'm lost in libCEC's code :D ).
So as a quick check I've re-built a local docker image without the RPI flags, here are the results:

```bash
➜  ~ docker run --privileged -it --entrypoint /bin/bash local/cec-addon-aarch64:2.4
bash-5.0# ./run.sh
[18:31:55] INFO: Starting CEC client scan...
opening a connection to the CEC adapter...
requesting CEC bus information ...
ERROR:   [          106011]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
ERROR:   [          106011]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
ERROR:   [          106011]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
CEC bus information
===================
device #1: Recorder 1
address:       2.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
ERROR:   [          106011]     CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
bash-5.0# cec-client -l
libCEC version: 4.0.3, git revision: libcec-4.0.3, compiled on Sun Jun 20 18:15:35 UTC 2021 by root@localhost on Linux 5.10.25-linuxkit (aarch64), features: P8_USB, DRM, P8_detect, Exynos, Linux, AOCEC
Found devices: 1

device:              1
com port:            Linux
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

bash-5.0#
```

Two conclusions I've made (but I might be wrong):
1. I was right and the RPI code which is failing on Odroid devices
2. We have another issue too, as it seems the Exynos adapter didn't initialize itself on my N2+ properly (or N2+ is supposed to use a different adapter)

boszme, 2021-06-20:
Another test run: built another image without Linux and AOCEC detection and it couldn't find anything :(

```
➜  ~ docker run --privileged -it --entrypoint /bin/bash local/cec-addon-aarch64:2.4
bash-5.0# cec-client -l
libCEC version: 4.0.3, git revision: libcec-4.0.3, compiled on Sun Jun 20 20:34:00 UTC 2021 by root@localhost on Linux 5.10.25-linuxkit (aarch64), features: P8_USB, DRM, P8_detect, Exynos
Found devices: NONE
bash-5.0# ./run.sh
[20:39:12] INFO: Starting CEC client scan...
autodetect FAILED
bash-5.0#
```

boszme, 2021-06-21:
And it is pretty much the same with AOCEC, so I'm not sure now which adapter should be used on N2+

```bash
bash-5.0# cec-client -l
libCEC version: 4.0.3, git revision: libcec-4.0.3, compiled on Mon Jun 21 05:01:50 UTC 2021 by root@localhost on Linux 5.10.25-linuxkit (aarch64), features: P8_USB, DRM, P8_detect, AOCEC
Found devices: NONE
bash-5.0# ./run.sh
[05:08:02] INFO: Starting CEC client scan...
autodetect FAILED
bash-5.0#
```

boszme, 2021-06-21:
@agners help me clarify one thing: what is the reason that both the OS both the cec-addon has libCEC?

I was investigating the cec-addon, not the base OS. I thought that the OS just need to pass the devices files (/dev/cec0, /dev/vchiq, etc..) to the docker image and everything should be handled inside docker after that.
My understanding was that only the kernel modules required at OS level to populate the /dev/ system with the needed device files. libCEC is supposed to contain also the kernel modules?

agners, 2021-06-21:
> what is the reason that both the OS and the cec-addon has libCEC?

The OS itself comes without any CEC library. However the Home Assistant Core container ships with one (see link above). Since the add-on and core are in separate container, they both need to bring the library themselfs.

> My understanding was that only the kernel modules required at OS level to populate the /dev/ system with the needed device files. libCEC is supposed to contain also the kernel modules?

In upstream Linux CEC support requires a kernel driver which exposes `/dev/cecX`. That driver seem to be available and enabled in ODROID-N2, since there is `/dev/cec0`. From there on, user space can use `/dev/cec0` in a standardized way to use CEC.

Unfortunately, there are drivers/hardware which use non-"standard"/upstream way to do CEC. From the looks it seems that Raspberry Pi (used to?) have its own way of doing CEC. I've not looked in how that exactly works, but it could involve all sorts of hackery.

samueltardieu, 2021-07-28:
The problem still exists in HomeAssistant 2021.7.4 on Odroid N2+. Running `echo scan | cec-client -s -d 1` from within the HomeAssistant container yields:

```
opening a connection to the CEC adapter...
requesting CEC bus information ...

CEC bus information
===================
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng

currently active source: unknown (-1)
```

with the following repeated error all along the process:

```
ERROR:   [           27693]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=64
```

Note that the errno is 64 (`ENONET`) while @boszme seemed to have an errno of 22 (`EINVAL`) back in June, but I'm using `cec-client` 6.0.2 from the base image while they were using an older 4.0.3 from the addon.

boszme, 2021-07-28:
@samueltardieu which adapters were enabled in your environment?
when I tried it, I got errno 22 when I disabled the RPI adapter (needed to re-compile the cec-addon docker image without RPI cmake flag).
In that case I believe it tried to you use the Linux adapter which gave the errno 22 (but I'm not confirmed), but I'm still not sure which adapter is supposed to work on the Odroid N2+.

```bash
cec-client -l
libCEC version: 4.0.3, git revision: libcec-4.0.3, compiled on Sun Jun 20 18:15:35 UTC 2021 by root@localhost on Linux 5.10.25-linuxkit (aarch64), features: P8_USB, DRM, P8_detect, Exynos, Linux, AOCEC
```

samueltardieu, 2021-07-28:
@bozme The base image no longer includes anything but the kernel-supported adapters since 2021.7.0. Moreover, strace shows that `/dev/cec0` is indeed used by `cec-client`.

Btw:

```bash
# cec-client -l
libCEC version: 6.0.2, git revision: libcec-6.0.2, compiled on 2021-06-17 11:49:16 by root@db06922c0afe on Linux 5.8.0-1033-azure (aarch64), features: P8_USB, DRM, P8_detect, Linux
Found devices: 1

device:              1
com port:            Linux
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
```

boszme, 2021-08-03:
@samueltardieu how did you run cec-client: have you enabled direct SSH and invoked from the base OS (https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host)?

samueltardieu, 2021-08-03:
Yes I did exactly that.

pvizeli, 2021-09-04:
I need only update the add-on to work with new hw layer + libcec version. We only support the Linux standard CEC drivers and also rpi added support for the standard kernel CEC protocol. However libcec ist stale and current there is no better library around for python

agners, 2021-10-13:
From the OS side, update 6.3 includes some additional drivers which I am pretty sure are required forHDMI CEC. I haven't tested it though. So testing with HAOS 6.3 or newer is required.

andybjackson, 2021-12-17:
Still not working in OS 7.0 & 2021.12.2
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[11:43:46] INFO: Starting CEC client scan...
opening a connection to the CEC adapter...
requesting CEC bus information ...
ERROR:   [          110439]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
ERROR:   [          110439]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
ERROR:   [          110439]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
CEC bus information
===================
device #1: Recorder 1
address:       2.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng
currently active source: unknown (-1)
ERROR:   [          110439]	CLinuxCECAdapterCommunication::Write - ioctl CEC_TRANSMIT failed - tx_status=00 errno=22
[cont-finish.d] executing container finish scripts...
[cont-finish.d] done.
[s6-finish] waiting for services.
[s6-finish] sending all processes the TERM signal.
```

agners, 2022-01-13:
Started to investigate into this issue a bit more.

One thing which is interesting is that Linux 5.10 seems to expose two CEC controllers:

```
# ls -la /dev/cec*
crw-rw----    1 root     video     250,   0 Jan 13 12:15 /dev/cec0
crw-rw----    1 root     video     250,   1 Jan 13 12:15 /dev/cec1
# find /sys -name "cec*"
/sys/devices/platform/soc/ff600000.bus/ff600000.hdmi-tx/cec0
/sys/devices/platform/soc/ff800000.bus/ff800280.cec/cec1
...
```

`/dev/cec0` seems to be provided by the Synopsis HDMI IP (through driver `CONFIG_DRM_DW_HDMI_CEC`), and the other, `/dev/cec1` seems to be a Amlogic specific device (compiled as a module `ao_cec_g12a`, through driver `CONFIG_CEC_MESON_G12A_AO`). Btw, there seems to be even a third IP but it [seems to be disabled in the ODROID-N2 device tree](https://github.com/torvalds/linux/blob/v5.10/arch/arm64/boot/dts/amlogic/meson-g12b-odroid-n2.dtsi#L349-L354), so I assume its not meant to be used. The driver `CONFIG_CEC_MESON_AO` is available as `ao_cec` module though.

Since Pascal updated the CEC Scanner addon in https://github.com/home-assistant/addons/pull/2240 it comes with the "vanilla" libcec 6.0.2 (at this point still the latest release).

Starting the add-on manually allows to run `cec-client` with all output enabled. 

```
# docker run -it --rm --privileged --entrypoint /bin/bash homeassistant/aarch64-addon-cec_scan:3.0
bash-5.1# cec-client -s
```

However, for me this did not really work well. It seems that messages get sent, but time out all the time (see dmesg)

<details>

```
bash-5.1# cec-client -s                                                                                                                                                                                                                                                                                                       
opening a connection to the CEC adapter...                                                                                                                                                                                                                                                                                    
DEBUG:   [               7]     Broadcast (F): osd name set to 'Broadcast'                                                                                                                                                                                                                                                    
DEBUG:   [               7]     CLinuxCECAdapterCommunication::Open - m_fd=3 bStartListening=1                                                                                                                                                                                                                                
DEBUG:   [               7]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_G_PHYS_ADDR - addr=2400                                                                                                                                                                                                                  
DEBUG:   [               7]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=0000 num_log_addrs=0                                                                                                                                                                                         
DEBUG:   [               8]     CLinuxCECAdapterCommunication::Open - ioctl CEC_ADAP_S_LOG_ADDRS - log_addr_mask=8000 num_log_addrs=1                                                                                                                                                                                         
NOTICE:  [               8]     connection opened                                                                                                                                                                                                                                                                             
DEBUG:   [               8]     CLinuxCECAdapterCommunication::Process - CEC_DQEVENT - CEC_EVENT_STATE_CHANGE - log_addr_mask=8000 phys_addr=2400                                                                                                                                                                             
DEBUG:   [               8]     processor thread started                                                                                                                                                                                                                                                                      
DEBUG:   [               8]     << Broadcast (F) -> TV (0): POLL                                                                                                                                                                                                                                                              
TRAFFIC: [               8]     << f0                                                                                                                                                                                                                                                                                         
[... cut ...]
[  133.341151] cec-meson_g12a_ao_cec: message 11 timed out
[  135.644925] cec-meson_g12a_ao_cec: message 11 timed out
[  137.948725] cec-meson_g12a_ao_cec: message 22 timed out
[  140.252534] cec-meson_g12a_ao_cec: message 22 timed out
[  142.556341] cec-meson_g12a_ao_cec: message 99 timed out
[  144.860180] cec-meson_g12a_ao_cec: message 99 timed out
```

I am a bit out of ideas here. I wonder if HDMI CEC every worked on upstream Linux on this platform. Is anyone aware if it works with Armbian/LibreELEC or similar distributions?

agners, 2022-01-13:
It seems that LIbreELEC doesn't support ODROID-N2+ (anymore). However, CoreELEC 19.4-Matrix_rc1 does, but they seem to use the downstream 4.9.113 kernel :spider_web: 

<details>

```
CoreELEC:~ # cec-client -s
opening a connection to the CEC adapter...
DEBUG:   [              14]     Broadcast (F): osd name set to 'Broadcast'
NOTICE:  [              14]     connection opened
DEBUG:   [              14]     << Broadcast (F) -> TV (0): POLL
TRAFFIC: [              14]     << f0
DEBUG:   [              14]     processor thread started
DEBUG:   [              43]     >> POLL sent
DEBUG:   [              43]     TV (0): device status changed into 'present'
DEBUG:   [              43]     << requesting vendor ID of 'TV' (0)
TRAFFIC: [              43]     << f0:8c
TRAFFIC: [             247]     >> 0f:87:00:e0:91
CoreELEC:~ # dmesg | grep cec
[    0.917788@4] cectx ff80023c.aocec: cec driver date:2020/03/16:reduece no msg in sleep time
[    0.922455@5] cectx ff80023c.aocec: compatible:amlogic, aocec-g12a
[    0.928353@5] cectx ff80023c.aocec: cecb_ver:0x1
[    0.932922@5] cectx ff80023c.aocec: line_reg:0x1
[    0.937546@5] cectx ff80023c.aocec: line_bit:0x3
[    0.942230@5] cectx ff80023c.aocec: ee_to_ao:0x1
[    0.946893@5] input: cec_input as /devices/virtual/input/input0
[    0.947058@4] cectx ff80023c.aocec: not find 'port_num'
[    0.951920@4] cectx ff80023c.aocec: using cec:1
[    0.956480@4] cectx ff80023c.aocec: no hdmirx regs
[    0.961182@4] cectx ff80023c.aocec: no hhi regs
[    0.965710@4] cectx ff80023c.aocec: not find 'arc_port_mask'
[    0.971374@4] cectx ff80023c.aocec: not find 'output'
[    0.977994@3] cectx ff80023c.aocec: wakeup_reason:0x0
[    0.981444@3] cectx ff80023c.aocec: cev val1: 0x0;val2: 0x0
[    0.986927@3] cectx ff80023c.aocec: aml_cec_probe success end
[    5.501066@1] cectx ff80023c.aocec: bad iniator with self 0x0
```

</details>

It seems that this old kernel has a widely different CEC stack using the input subsystem to expose the device. CoreELEC does use libCEC 6.0.2. So from that exercise I know that my setup does support HDMI CEC and that HAOS should be using the `CONFIG_CEC_MESON_G12A_AO` driver (or `/dev/cec1`).

oramirite, 2022-01-13:
Thank you Stefan for looking into this so deeply.

I just want to mention that a clue of versions of HA ago, I was able to
manually run the CEC scanner container in privileged mode and correctly
scan for devices. Simply changing to privileged and running the same
command worked. I was never able to get the CEC integration to work though.

agners, 2022-01-13:
@oramirite from what I see that was on Raspberry Pi 3 correct? This issue is really about ODROID-N2. While errors could look similar, its likely a different root cause as the two boards use different kernel versions and drivers. If that is indeed still a problem and you'd still like to use HDMI CEC on Raspberry Pi, then we should open a separate issue for that.

agners, 2022-01-13:
Disabling the DW HDMI CEC driver in #1216 does help to make `/dev/cec0` to be the correct device, however, at least in my setup HDMI CEC still seems not to work :cry: .

andybjackson, 2022-01-14:
Thanks @agners & @oramirite. I have tried both the Odroid & the Pi and get different errors with ultimately the same outcome. Makes me wonder how the 7 reporting users of this feature got it to work. Look forward to testing the changed code and will update ASAP.

rigrig, 2022-01-14:
> Makes me wonder how the 7 reporting users of this feature got it to work.

As (I think) one of those data points: I didn't.
I installed the addon & integration a while ago on my odroid-n2+, noticed they didn't work, found this issue, and simply haven't removed them yet (hoping for a fix).

agners, 2022-01-15:
I got a hint on the mailing list that Arch Linux ARM with Linux 5.15.13 works. It turns out that Arch Linux compiles some drivers as a module, which seems to change behavior such that it works :man_shrugging: 

I'll prepare a PR to get this fixed.

agners, 2022-01-15:
Btw, it seem that there is a more "upstream" Linux friendly tool called `cec-ctl` which is part of the `v4l-utils` package in Alpine Linux. We should consider using it instead of libcec.

agners, 2022-01-17:
HDMI CEC should now be functional with HAOS 7.2 (currently in beta channel).

rigrig, 2022-01-17:
I installed the beta (and reinstalled the addon), but both the addon and the integration are still broken for me :-(

<details><summary>Supervisor log</summary>

```
22-01-17 22:46:28 INFO (SyncWorker_9) [supervisor.docker.interface] Cleaning addon_core_cec_scan application
22-01-17 22:46:28 INFO (SyncWorker_9) [supervisor.docker.interface] Removing image homeassistant/aarch64-addon-cec_scan with latest and 3.0
22-01-17 22:46:28 INFO (MainThread) [supervisor.addons.addon] Removing add-on data folder /data/addons/data/core_cec_scan
22-01-17 22:46:28 INFO (MainThread) [supervisor.addons] Add-on 'core_cec_scan' successfully removed
22-01-17 22:46:46 INFO (MainThread) [supervisor.addons] Creating Home Assistant add-on data folder /data/addons/data/core_cec_scan
22-01-17 22:46:46 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image homeassistant/aarch64-addon-cec_scan with tag 3.0.
22-01-17 22:46:50 INFO (MainThread) [supervisor.addons] Add-on 'core_cec_scan' successfully installed
22-01-17 22:46:53 INFO (SyncWorker_1) [supervisor.docker.addon] Starting Docker add-on homeassistant/aarch64-addon-cec_scan with version 3.0
```
</details>

<details><summary>CEC Scanner Addon log</summary>

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[22:46:53] INFO: Starting CEC client scan...
opening a connection to the CEC adapter...
ERROR:   [               7]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [               7]	could not open a connection (try 1)
ERROR:   [            1008]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            1008]	could not open a connection (try 2)
ERROR:   [            2008]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            2008]	could not open a connection (try 3)
ERROR:   [            3008]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            3008]	could not open a connection (try 4)
ERROR:   [            4009]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            4009]	could not open a connection (try 5)
ERROR:   [            5009]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            5009]	could not open a connection (try 6)
ERROR:   [            6009]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            6009]	could not open a connection (try 7)
ERROR:   [            7010]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            7010]	could not open a connection (try 8)
ERROR:   [            8010]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            8010]	could not open a connection (try 9)
ERROR:   [            9011]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [            9011]	could not open a connection (try 10)
unable to open the device on port Linux
ERROR:   [           10011]	could not start CEC communications
```

</details>
<details><summary>ha core log</summary>

```
...
2022-01-17 22:46:51 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:52 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:53 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:54 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:54 ERROR (ThreadPoolExecutor-0_0) [pycec.cec] failed to open a connection to the CEC adapter
2022-01-17 22:46:55 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:56 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:57 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:58 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:46:59 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
2022-01-17 22:47:00 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.
...
```
</details>

## System Health

version | core-2021.12.9
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.91
arch | aarch64
timezone | Europe/Amsterdam

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4850
Installed Version | 1.19.3
Stage | running
Available Repositories | 944
Downloaded Repositories | 12

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.2
-- | --
update_channel | beta
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 13.8 GB
disk_used | 11.1 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Network UPS Tools (0.9.0), motionEye (0.16.0), Studio Code Server (4.1.0), Let's Encrypt (4.12.0), CEC Scanner (3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 6
-- | --
resources | 7
views | 10
mode | storage

</details>

agners, 2022-01-17:
Just double checked here, works fine with HAOS 7.2 and CEC Scanner 3.0.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[23:30:34] INFO: Starting CEC client scan...
opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        LG
osd string:    TV
CEC version:   1.3a
power status:  standby
language:      eng
...
```

Errno 16 means "Device or resource busy", it sounds as if something else is using the device. Do you maybe have a CEC configuration active in HAOS? Any other Add-on which might be using the CEC device?

rigrig, 2022-01-17:
> Do you maybe have a CEC configuration active in HAOS?

Yes, that was it, thanks!
I didn't realise mixing the addon and the integration would cause problems.

agners, 2022-01-18:
> I didn't realise mixing the addon and the integration would cause problems.

I'd say, it didn't cause problems: The system prevented the second user from interfering, by returning error 16. libcec should really print a nicer message here :)

```
ERROR:   [               7]	CLinuxCECAdapterCommunication::Open - ioctl CEC_S_MODE failed - errno=16
ERROR:   [               7]	could not open a connection (try 1)
```

But yeah, I know what you mean, it is not obvious that you can't use the two of them at the same time.

andybjackson, 2022-01-18:
Where does one find the Integration? The CEC Scanner add-on isn't working for me either.

rigrig, 2022-01-18:
You can add it in your `configuration.yaml`: https://www.home-assistant.io/integrations/hdmi_cec/#configuration-example

andybjackson, 2022-01-18:
2022-01-18 17:25:13 WARNING (SyncWorker_0) [pycec] Not initialized. Waiting for init.

That's with the following YAML:
`hdmi_cec:
  devices:
    TV: 0.0.0.0
    HABlue: 1.0.0.0`

andybjackson, 2022-11-21:
Afraid it looks like this bug is back. Had to rebuild my Home Assistant on HA Blue (ODroid 2+) after a failed update.

The log messages I got to know so well are there in all their glory.
`Not initialized. Waiting for init.
21:53:43 – (WARNING) /usr/local/lib/python3.10/site-packages/pycec/network.py - message first occurred at 21:35:03 and shows up 1118 times
failed to open a connection to the CEC adapter
21:53:15 – (ERROR) /usr/local/lib/python3.10/site-packages/pycec/cec.py - message first occurred at 21:35:05 and shows up 10 times
Error doing job: Exception in callback CecAdapter.standby_devices()
21:35:18 – (ERROR) /usr/src/homeassistant/homeassistant/runner.py`

I have the integration enabled by the entired in configuraiton.yaml as before. Ihave tried with and without CEC Scanner enabled. I have been back through all of the issues and can't din what setting I might have changed.

agners, 2022-11-22:
What HAOS version are you using?

Anyone else actively using HDMI CEC on ODROID-N2?

inytar, 2022-11-22:
I'm using HDMI CEC it on the Odroid-N2 (HA Blue), without any issues.
I had CEC issues with SU 2022.11.0, but that related to: https://github.com/home-assistant/supervisor/issues/4004, and was fixed once I moved to SU 2022.11.2.

```
Home Assistant 2022.11.4
Supervisor 2022.11.2
Operating System 9.3
Frontend 20221108.0 - latest
```

andybjackson, 2022-11-22:
Thanks @inytar. Upgrading SU to 2022.11.2 solved the problem for me too. And thank you @agners for reacting so promptly.

This can be closed again.

closed 2022-11-22 by agners