home-assistant/operating-system #1221: problem to reboot after HASSOS upgrade on RPI3b+

pilot1981, 2021-02-10:
**Hardware Environment**
- [X] Raspberry Pi 3b+ with HASSOS booting from SSD disk USD attached

**Home Assistant OS release:**
- [X] Updated from version HASSOS 5.10 and Home Assistant 2021.2.2, core-2021.2.2,Versione Frontend: 20210127.7
- Additional information (if accessible):
please see attached screenshoot
[image]

**Supervisor logs:**
please attached screenshoot

[image]

**Journal logs:**
journalctl don't exist

**Kernel logs:**
[image]

**Description of problem:**
from many months, many releases, everytime I upgrade HASSOS, RPI3b+ stop on reboot phase (as you can see in picture). then I have to manually power off raspberry and then attach again power supply and it come back alive booting fine....
Reboot works fine when I use manual command and when I upgrade supervisor!

agners, 2021-02-11:
Is that reproducible when manually trigger an update or downgrade? E.g. when you now move back to OS 5.10 using `ha os update --version 5.10` or/and when moving back to 5.11 using  `ha os update --version 5.11`?

pilot1981, 2021-02-12:
Yes, sure!

pilot1981, 2021-02-12:
@agners you need more info?

agners, 2021-02-18:
Just tested with 5.12, downgrading to 5.11 and upgrading to 5.12 seems to work fine when booting from SD card. I guess it must be related to USB boot. Did you consider using the `datactl` approach? https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk

pilot1981, 2021-02-18:
Hi,
thanks for reply!
I also think this could be related to use of USB disk, but my firend with the same architecture don't have same issue...

I wouldn't use both SD and USB disk...

we have any other way to try to solve it?

agners, 2021-02-18:
> [..] but my firend with the same architecture don't have same issue...

That is a the problem with USB SSD support: There are so many combinations, and some work and some don't, and some just behave weird (like yours, works but doesnt reboot :man_shrugging: ). That is why we kinda try to move away from USB SSD support, as it is just hard to properly support.

What I would do is check if it also happens with Raspberry Pi OS. If so, then its likely a general kernel/firmware issue, and needs to be reported with the Raspberry Pi project.

Maybe forums help, e.g. a quick Google brought up this one: https://www.raspberrypi.org/forums/viewtopic.php?t=213394

pilot1981, 2021-02-18:
I moved to SSD after 3 Sd Card broken...

agners, 2021-02-18:
Sorry, I should have been a bit more specific,  USB SSD boot support.

Booting from SD card and using USB SSD just for data is more reliable as only LInux has to deal with the USB SSD (what I referred above with datactl approach, see https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk). We only write to the SD card when updating, as the root file system is read-only with Home Assistant OS.  So SD cards will wear-out a lot slower.

pilot1981, 2022-01-04:
problem solved changing USB adapter and enabling TRIM!

I think i's sufficient change USB adapter to solved this issue!

closed 2022-01-04 by pilot1981