home-assistant/operating-system #1223: REQUEST : Add Canbus support for like can2usb dongles

pergolafabio, 2021-02-12:
Hi, this is a request for canbus ,canbus is widely used in car industry, but also in home automations...
There are dongles like https://canable.io/ to talk with the system

But we need extra tools in the HassOS , like candump/ slcand , package can be found here : 

https://github.com/linux-can/can-utils / 
https://pypi.org/project/python-can/

but most of important , some kernel modules need to load also : 

sudo modprobe can
sudo modprobe can_raw
sudo modprobe slcan

is this possible as a request?

more info : http://pascal-walter.blogspot.com/2015/08/installing-lawicel-canusb-on-linux.html

thnx and appreciated

pvizeli, 2021-02-15:
You can make an PR again https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/kernel/device-support.config - please test it before you make the PR public.

pergolafabio, 2021-02-15:
ok, interesting allthough not sure what lines i need to add there? where can i find these for canbus?

pergolafabio, 2021-02-15:
is it enought by adding like below? : 

config_can=y
config_vcan=y
config_can_raw=y
config_slcan=y

pvizeli, 2021-02-15:
https://github.com/home-assistant/operating-system/tree/dev/Documentation

pergolafabio, 2021-02-15:
ok, but if i make a PR wit these lines, you need to bring it in DEV before i can test it right?

```
CONFIG_CAN=m
CONFIG_CAN_RAW=m
CONFIG_CAN_VCAN=m
CONFIG_CAN_SLCAN=m
CONFIG_CAN_DEV=m
```

pergolafabio, 2021-02-15:
done : 

https://github.com/home-assistant/operating-system/pull/1228

pergolafabio, 2021-02-17:
merged in  : https://github.com/home-assistant/operating-system/pull/1228

closed 2021-02-17 by pergolafabio