home-assistant/operating-system #1226: Updating OS ERROR on Task exception was never retrieved

d4s1r, 2021-02-15:
**Hardware Environment**

Raspberry Pi [3]
Operating System Family | Linux
Operating System Version | 5.4.79-v7
CPU Architecture | armv7l

**Home Assistant OS release:**
Version | 2021.1.5
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.7
- Additional information (if accessible):
Host Operating System | Home Assistant OS 5.10
Update Channel | stable
Supervisor Version | 2021.02.9
Docker Version | 19.03.13
Disk Total | 13.9 GB
Disk Used | 6.4 GB
Healthy | true
Supported | true
Board | rpi3
Supervisor API | ok
Version API | ok
Installed Add-ons | Terminal & SSH (8.9.1), Mosquitto broker (5.1), File editor (5.2.0), Happy Bubbles (3.1.4), Zigbee2mqtt (1.16.2.2)

**Supervisor logs:**
21-02-15 11:42:04 ERROR (MainThread) [supervisor.hassos] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.1: Opening output device /dev/disk/by-partlabel/hassos-kernel1 failed: Device or resource busy
21-02-15 11:42:04 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-1578' coro=<process_lock.<locals>.wrap_api() done, defined at /usr/src/supervisor/supervisor/utils/__init__.py:24> exception=HassOSUpdateError()>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/utils/__init__.py", line 33, in wrap_api
    return await method(api, *args, **kwargs)
  File "/usr/src/supervisor/supervisor/hassos.py", line 177, in update
    raise HassOSUpdateError()
supervisor.exceptions.HassOSUpdateError

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**

I tried to update the Core and the OS. It always stops throwing this exception. The changelogs refer to an old version of python no longer being available, which might hint to a connection to my issue. I will provide the missing logs afterwards if needed.

agners, 2021-02-17:
It seems that the system has problems reading data in general:

```
[ 2397.210416] SQUASHFS error: zlib decompression failed, data probably corrupt
[ 2397.210555] SQUASHFS error: squashfs_read_data failed to read block 0x1fc1be
```

This sounds like a broken SD card... I would try to create a snapshot, download it and move to a new card asap.

d4s1r, 2021-02-19:
@agners Thank you very much. I was able to create a snapshot and recover the data on a new SD Card.

closed 2021-02-19 by d4s1r