home-assistant/operating-system #1227: ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed

KrisRux, 2021-02-15:
**Hardware Environment**

- [X] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

~ $ ha info
arch: armv7
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.11"
homeassistant: 2021.2.3
hostname: hassio
logging: info
machine: raspberrypi3
operating_system: Home Assistant OS 5.11
state: running
supervisor: 2021.02.9
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/Rome

**Supervisor logs:**

`~ $ ha su logs
21-02-15 13:58:27 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
21-02-15 13:58:30 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/armv7-hassio-cli with version 2021.02.1 - 172.30.32.5
21-02-15 13:58:30 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-observer with version 2020.10.1
21-02-15 13:58:30 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-hassio-multicast with version 3
21-02-15 13:58:30 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-02-15 13:58:30 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_multicast application
21-02-15 13:58:32 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast homeassistant/armv7-hassio-multicast with version 3 - Host
21-02-15 13:58:32 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-15 13:58:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi3-homeassistant with version 2021.2.3
21-02-15 13:58:33 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 28 Home Assistant secrets
21-02-15 13:58:33 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.11 / BootSlot B
21-02-15 13:58:34 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'startup' with 'before' is deprecated. Please report this to the maintainer of Dropbox Sync
21-02-15 13:58:35 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-15 13:58:35 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/7be23ff5 repository
21-02-15 13:58:35 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/03d23610 repository
21-02-15 13:58:35 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-02-15 13:58:35 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-02-15 13:58:42 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'startup' with 'before' is deprecated. Please report this to the maintainer of Dropbox Sync
21-02-15 13:58:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-02-15 13:58:43 INFO (MainThread) [supervisor.store] Loading add-ons from store: 67 all - 67 new - 0 remove
21-02-15 13:58:43 INFO (MainThread) [supervisor.addons] Found 10 installed add-ons
21-02-15 13:58:43 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-samba with version 9.3.0
21-02-15 13:58:43 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to dwelch2101/dropbox-sync-armhf with version 1.3.0
21-02-15 13:58:43 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to hassioaddons/ide-armv7 with version 2.2.0
21-02-15 13:58:43 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-duckdns with version 1.12.5
21-02-15 13:58:43 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to danimtb/armv7-dasshio with version 0.3.8
21-02-15 13:58:43 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/armv7-addon-ssh with version 9.0.1
21-02-15 13:58:43 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to esphome/esphome-hassio-armv7 with version 1.16.1
21-02-15 13:58:43 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/node-red/armv7 with version 8.1.1
[... 642 lines cut ...]
[  152.596316] audit: type=1700 audit(1613397558.615:201): dev=vethdaa9a5e prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  154.660063] audit: type=1325 audit(1613397560.685:202): table=nat family=2 entries=0
[  154.701421] audit: type=1325 audit(1613397560.725:203): table=filter family=2 entries=0
[  154.762290] audit: type=1325 audit(1613397560.785:204): table=nat family=2 entries=5
[  154.780120] audit: type=1325 audit(1613397560.805:205): table=nat family=2 entries=7
[  154.810916] audit: type=1325 audit(1613397560.835:206): table=nat family=2 entries=8
[  154.838822] audit: type=1325 audit(1613397560.865:207): table=nat family=2 entries=10
[  154.860978] audit: type=1325 audit(1613397560.885:208): table=nat family=2 entries=11
[  154.908923] audit: type=1325 audit(1613397560.935:209): table=nat family=2 entries=12
[  154.937458] audit: type=1325 audit(1613397560.965:210): table=nat family=2 entries=13
[  155.058363] eth0: renamed from vethd4097f2
[  155.102258] IPv6: ADDRCONF(NETDEV_CHANGE): vethdaa9a5e: link becomes ready
[  155.102651] hassio: port 11(vethdaa9a5e) entered blocking state
[  155.102669] hassio: port 11(vethdaa9a5e) entered forwarding state
[  161.663806] device wlan0 entered promiscuous mode
[  161.664006] kauditd_printk_skb: 1 callbacks suppressed
[  161.664016] audit: type=1700 audit(1613397567.685:212): dev=wlan0 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295

**Description of problem:**
ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed!
[ +0,000102] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[ +2,563903] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[ +2,566418] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110

[IMG_20210215_144530](https://user-images.githubusercontent.com/17131080/107960845-6afa2f00-6fa5-11eb-98e4-abc65a8fd980.jpg)

agners, 2021-02-17:
Hm, this seems to be similar to https://github.com/raspberrypi/linux/issues/3849, however that is about Raspberry Pi 4. But there are reports in that thread about that issue with Raspberry Pi 3 Model B. But from what I can tell, no solution so far :disappointed:

mir-asp, 2022-01-04:
Unfortunately, the problem still occurs in my case on RPi4.
According to some users in [bug reported on raspberrypi/linux](https://github.com/raspberrypi/linux/issues/3849) it should be solved by latest firmware. HASSIO is using the same (latest) firmware:
`[    3.881919] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e`
and the problem still occures.
Any advices ???

agners, 2022-01-12:
> Any advices ???

How fast is that reproducible for you? Maybe testing with Raspberry Pi OS to see if it happens there too? If so, then reporting on the Raspberry Pi repositories might help to get it resolved.

KrisRux, 2022-02-27:
Unfortunately, the problem still occurs again
[image]
.

KrisRux, 2022-05-19:
This problem occurs with core-2022.5.4 - Home Assistant OS 8.0 - 32bit or 64bit OS

agners, 2022-05-19:
In OS 8.0, we use the latest firmware version from https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi, which from what I understand is pretty much the latest version from what Raspberry Pi distributes.

Which Raspberry Pi 3 model are you using exactly?

Can you reproduce the problem on Raspberry Pi OS?

agners, 2022-05-19:
Hm, -123 means `ENOMEDIUM`, basically it seems that the WiFi chip disappears from the SDIO bus. Yeah I am sorry, really no idea how/if that can be fixed. I'd guess that this will be reproducible on Raspberry Pi OS (since we use the exact same firmware and kernel), and if so you can report it in the Raspberry Pi bug tracker ([raspberrypi/linux ](https://github.com/raspberrypi/linux/)). Maybe its really a hardware issue. Is your power supply powerful enough? WiFi can cause quite some power spikes.

KrisRux, 2022-05-19:
Hi @agners,
thanks for your reply. I have changed my power supply with a new original once, but the issue continues.
About power spikes of wifi, how do check this?

Krautmaster, 2022-06-25:
since I did a rpi-update, my Pi Z 2 Wifi also crashes. Mostly when switching within the mesh.

`Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: deleting address fe80::9470:7258:3ba7:df8d
Jun 25 11:43:42 pi3a avahi-daemon[300]: Withdrawing address record for fe80::9470:7258:3ba7:df8d on wlan0.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::9470:7258:3ba7:df8d.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Interface wlan0.IPv6 no longer relevant for mDNS.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Withdrawing address record for 192.168.2.150 on wlan0.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.2.150.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Interface wlan0.IPv4 no longer relevant for mDNS.
Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: deleting route to 192.168.2.0/24
Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: deleting default route via 192.168.2.10
Jun 25 11:43:42 pi3a avahi-daemon[300]: Got SIGHUP, reloading.
Jun 25 11:43:42 pi3a avahi-daemon[300]: No service file found in /etc/avahi/services.
Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: carrier acquired
Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: IAID 01:58:07:f7
Jun 25 11:43:42 pi3a dhcpcd[570]: wlan0: adding address fe80::9470:7258:3ba7:df8d
Jun 25 11:43:42 pi3a avahi-daemon[300]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::9470:7258:3ba7:df8d.
Jun 25 11:43:42 pi3a avahi-daemon[300]: New relevant interface wlan0.IPv6 for mDNS.
Jun 25 11:43:42 pi3a avahi-daemon[300]: Registering new address record for fe80::9470:7258:3ba7:df8d on wlan0.*.
Jun 25 11:43:43 pi3a dhcpcd[570]: wlan0: soliciting an IPv6 router
Jun 25 11:43:43 pi3a dhcpcd[570]: wlan0: rebinding lease of 192.168.2.150
Jun 25 11:43:44 pi3a ntpd[811]: Deleting interface #3 wlan0, 192.168.2.150#123, interface stats: received=147, sent=147, dropped=0, active_time=195 secs
Jun 25 11:43:44 pi3a ntpd[811]: 185.168.228.58 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 141.82.25.201 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 81.7.16.52 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 167.86.115.96 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 116.202.26.144 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 3.64.117.201 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 78.46.60.40 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 217.91.44.17 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 176.9.102.215 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 141.95.116.43 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 85.214.58.202 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 162.159.200.123 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 129.250.35.251 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 212.18.3.18 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 78.47.249.55 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 131.188.3.221 local addr 192.168.2.150 -> <null>
Jun 25 11:43:44 pi3a ntpd[811]: 192.168.2.10 local addr 192.168.2.150 -> <null>
Jun 25 11:43:48 pi3a dhcpcd[570]: wlan0: probing for an IPv4LL address
Jun 25 11:43:48 pi3a dhcpcd[570]: wlan0: DHCP lease expired
Jun 25 11:43:48 pi3a dhcpcd[570]: wlan0: soliciting a DHCP lease
Jun 25 11:43:53 pi3a dhcpcd[570]: wlan0: using IPv4LL address 169.254.47.171
Jun 25 11:43:53 pi3a avahi-daemon[300]: Joining mDNS multicast group on interface wlan0.IPv4 with address 169.254.47.171.
Jun 25 11:43:53 pi3a dhcpcd[570]: wlan0: adding route to 169.254.0.0/16
Jun 25 11:43:53 pi3a dhcpcd[570]: wlan0: adding default route
Jun 25 11:43:53 pi3a avahi-daemon[300]: New relevant interface wlan0.IPv4 for mDNS.
Jun 25 11:43:53 pi3a avahi-daemon[300]: Registering new address record for 169.254.47.171 on wlan0.IPv4.
Jun 25 11:43:55 pi3a ntpd[811]: Listen normally on 6 wlan0 169.254.47.171:123
Jun 25 11:43:55 pi3a ntpd[811]: new interface(s) found: waking up resolver
Jun 25 11:43:56 pi3a dhcpcd[570]: wlan0: no IPv6 Routers available
Jun 25 11:47:28 pi3a kernel: [  544.751249] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jun 25 11:47:28 pi3a kernel: [  544.835372] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jun 25 11:47:28 pi3a kernel: [  544.835399] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Jun 25 11:47:28 pi3a kernel: [  545.275350] ieee80211 phy0: brcmf_netdev_start_xmit: xmit rejected state=0
Jun 25 11:47:28 pi3a avahi-daemon[300]: Interface wlan0.IPv6 no longer relevant for mDNS.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::9470:7258:3ba7:df8d.
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: carrier lost
Jun 25 11:47:28 pi3a avahi-daemon[300]: Interface wlan0.IPv4 no longer relevant for mDNS.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 169.254.47.171.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Withdrawing address record for fe80::9470:7258:3ba7:df8d on wlan0.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Withdrawing address record for 169.254.47.171 on wlan0.
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: deleting address fe80::9470:7258:3ba7:df8d
[... cut ...]
Jun 25 11:47:30 pi3a ntpd[811]: 192.168.2.10 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 131.188.3.221 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 78.47.249.55 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 212.18.3.18 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 129.250.35.251 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 162.159.200.123 local addr 169.254.47.171 -> <null>
Jun 25 11:47:30 pi3a ntpd[811]: 85.214.58.202 local addr 169.254.47.171 -> <null>
`

closed 2022-09-30 by github-actions[bot]