home-assistant/operating-system #1229: hassio_audio container causing higher CPU usage

NdR91, 2021-02-17:
**Hardware Environment**

- [X] Raspberry Pi 4 4GB
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [X] Updated from version 2021.2.1 to 2021.2.3. Supervisor Version: 2021.02.9
- Additional information (if accessible): 

## System Health

version | core-2021.2.3
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.10.11-v7l+
arch | armv7l
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4440
Installed Version | 1.11.3
Stage | running
Available Repositories | 760
Installed Repositories | 52

</details>

<details><summary>Hass.io</summary>

host_os | Raspbian GNU/Linux 10 (buster)
-- | --
update_channel | stable
supervisor_version | supervisor-2021.02.9
docker_version | 20.10.3
disk_total | 58.4 GB
disk_used | 17.5 GB
healthy | true
supported | failed to load: Unsupported
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.6.0), Duck DNS (1.12.5), Mosquitto broker (5.1), Zigbee2mqtt Edge (test), Home Assistant Google Drive Backup (0.103.1), File editor (5.2.0), Zigbee2mqtt (1.17.1-4), Log Viewer (0.9.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 6
-- | --
resources | 50
views | 46
mode | storage

</details>

**Supervisor logs:**

<details><summary>Logs</summary>
[... 87 lines cut ...]
D: [pulseaudio] backend-native.c: Bluetooth Headset Backend API support using the native backend,
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPAGProfile 00001112-0000-1000-8000-00805f9b34fb,
D: [pulseaudio] module-udev-detect.c: /dev/snd/controlC0 is accessible: yes,
D: [pulseaudio] module-udev-detect.c: Resuming all sinks and sources of card alsa_card.platform-bcm2835_audio.,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
I: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files,
D: [pulseaudio] bluez5-util.c: oFono is running: no,
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPHSProfile 00001108-0000-1000-8000-00805f9b34fb,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
</details>

**Description of problem:**

- Has this been working before (is this a regression?) YES
- Has there been attempt to rule out harware issues? (different SD card etc.) No, same HW
Starting from some days, I had issues with the SMB file transfer to and from HDDs connected to my Pi (2 HDDs and 1 SSD). The problem was generalized and not specific to one drive. Also the CPU load was unusual with a 30/40% of load instead of 5/10%.

So I tried stopping each container I had on Docker and I found that the hassio_audio is the problem. 
If I stop it, my CPU load return to 5/10% and also my file transfer rate come to the normal speed.

I cannot kill the container, as it restart by itself.

agners, 2021-02-17:
Hm, are the logs of that container saying anything specific (via frontend -> Supervisor -> System -> Audio logs or `docker logs hassio_audio`)

pvizeli, 2021-02-17:
`ha audio stats`

NdR91, 2021-02-17:
> Hm, are the logs of that container saying anything specific (via frontend -> Supervisor -> System -> Audio logs or `docker logs hassio_audio`)

Sure, you can find it in my main post under "hassio_audio Logs". I will paste theme here for praticity:

<details><summary>Logs</summary>

D: [pulseaudio] module-switch-on-connect.c: Trying to switch to new sink alsa_output.platform-bcm2835_audio.stereo-fallback,
I: [pulseaudio] core.c: configured_default_sink: (unset) -> alsa_output.platform-bcm2835_audio.stereo-fallback,
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.,
I: [pulseaudio] core.c: default_sink: (unset) -> alsa_output.platform-bcm2835_audio.stereo-fallback,
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to change event.,
I: [pulseaudio] module.c: Loaded "module-alsa-card" (index: #6; argument: "device_id="0" name="platform-bcm2835_audio" card_name="alsa_card.platform-bcm2835_audio" namereg_fail=false tsched=no fixed_latency_range=no ignore_dB=no deferred_volume=yes use_ucm=yes avoid_resampling=no card_properties="module-udev-detect.discovered=1"").,
I: [pulseaudio] module-udev-detect.c: Card /devices/platform/soc/fe00b840.mailbox/bcm2835_audio/sound/card0 (alsa_card.platform-bcm2835_audio) module loaded.,
I: [pulseaudio] module-udev-detect.c: Found 1 cards.,
I: [pulseaudio] module.c: Loaded "module-udev-detect" (index: #5; argument: "tsched=0").,
D: [pulseaudio] module.c: Checking for existence of '/usr/lib/pulse-14.2/modules/module-bluez5-discover.so': success,
D: [pulseaudio] dbus-util.c: Successfully connected to D-Bus system bus a8edc321ce64f04d03c0757e602c1a0e as :1.46,
I: [pulseaudio] module.c: Loaded "module-bluez5-discover" (index: #8; argument: "").,
I: [pulseaudio] module.c: Loaded "module-bluetooth-discover" (index: #7; argument: "").,
I: [pulseaudio] module.c: Loaded "module-native-protocol-unix" (index: #9; argument: "auth-anonymous=1 auth-cookie-enabled=0 socket=/data/external/pulse.sock").,
I: [pulseaudio] module-default-device-restore.c: Manually configured default sink, not overwriting.,
I: [pulseaudio] module-default-device-restore.c: No previous default source setting, ignoring.,
I: [pulseaudio] module.c: Loaded "module-default-device-restore" (index: #10; argument: "").,
E: [pulseaudio] module-rescue-streams.c: module-rescue-stream is obsolete and should no longer be loaded. Please remove it from your configuration.,
I: [pulseaudio] module.c: Loaded "module-rescue-streams" (index: #11; argument: "").,
I: [pulseaudio] module.c: Loaded "module-always-sink" (index: #12; argument: "").,
I: [pulseaudio] module.c: Loaded "module-position-event-sounds" (index: #13; argument: "").,
E: [pulseaudio] main.c: Failed to acquire org.pulseaudio.Server: org.freedesktop.DBus.Error.AccessDenied: Connection ":1.46" is not allowed to own the service "org.pulseaudio.Server" due to security policies in the configuration file,
I: [pulseaudio] main.c: Daemon startup complete.,
I: [pulseaudio] module.c: Unloading "module-rescue-streams" (index: #11).,
I: [pulseaudio] module.c: Unloaded "module-rescue-streams" (index: #11).,
D: [pulseaudio] core-subscribe.c: Dropped redundant event due to remove event.,
D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.AgentManager1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.ProfileManager1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.HealthManager1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Introspectable found, skipping,
D: [pulseaudio] bluez5-util.c: Adapter /org/bluez/hci0 found,
D: [pulseaudio] bluez5-util.c: Registering /MediaEndpoint/A2DPSink/sbc on adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Registering /MediaEndpoint/A2DPSource/sbc on adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Unknown interface org.freedesktop.DBus.Properties found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.GattManager1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.LEAdvertisingManager1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.Media1 found, skipping,
D: [pulseaudio] bluez5-util.c: Unknown interface org.bluez.NetworkServer1 found, skipping,
D: [pulseaudio] backend-native.c: Bluetooth Headset Backend API support using the native backend,
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPAGProfile 00001112-0000-1000-8000-00805f9b34fb,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] module-udev-detect.c: /dev/snd/controlC0 is accessible: yes,
D: [pulseaudio] module-udev-detect.c: Resuming all sinks and sources of card alsa_card.platform-bcm2835_audio.,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
I: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files,
D: [pulseaudio] bluez5-util.c: oFono is running: no,
D: [pulseaudio] backend-native.c: Registering Profile /Profile/HSPHSProfile 00001108-0000-1000-8000-00805f9b34fb,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
D: [pulseaudio] bluez5-util.c: Properties changed in adapter /org/bluez/hci0,
[... cut ...]
I: [pulseaudio] client.c: Created 5 "Native client (UNIX socket client)",
I: [pulseaudio] protocol-native.c: Client authenticated anonymously.,
D: [pulseaudio] protocol-native.c: Protocol version: remote 34, local 34,
D: [pulseaudio] protocol-native.c: SHM possible: no,
D: [pulseaudio] protocol-native.c: Negotiated SHM: no,
D: [pulseaudio] protocol-native.c: Disabling srbchannel, reason: No SHM support,
I: [pulseaudio] client.c: Freed 5 "supervisor",
I: [pulseaudio] protocol-native.c: Connection died.,</details>

agners, 2021-02-17:
> Sure, you can find it in my main post under "hassio_audio Logs". I will paste them here for praticity:

Whoops I see. Hm, yeah nothing really out of the ordinary in my eyes.

> > `ha audio stats`
> 
> What do you mean?

This prints the statistics such as CPU usage of the Audio Container specifically.

NdR91, 2021-02-17:
> > Sure, you can find it in my main post under "hassio_audio Logs". I will paste them here for praticity:
> 
> Whoops I see. Hm, yeah nothing really out of the ordinary in my eyes.
> 
> > > `ha audio stats`
> > 
> > 
> > What do you mean?
> 
> This prints the statistics such as CPU usage of the Audio Container specifically.

Here the output of "ha audio stats":

` ha audio stats
blk_read: 14684160
blk_write: 57344
cpu_percent: 0.17
memory_limit: 0
memory_percent: 0
memory_usage: 0
network_rx: 6841170
network_tx: 690`

What I think is that the issue isn't the CPU usage of the container itself, but it's something like if he is writing continously on the disk.

I also found this thread on the community forum: [https://community.home-assistant.io/t/audio-container/219512](url)
The Author says: 
> I have one particular setup monitored with Grafana and can show you what happens when this Container fails: HDD writes up to 150MB every 5 minutes.

I don't know if this is the same issue or wat. But I tried again in order to be 110% sure: if I stop only thos container, my file transfer speed go to 100 Mb/s... until it restarts again. After that the transfer rate go to 30 Mb/s.

agners, 2021-02-17:
> What I think is that the issue isn't the CPU usage of the container itself, but it's something like if he is writing continously on the disk.

But writes of that container also don't seem particularly high.

The output of `ha audio stats` is basically what Docker reports, you can also look at the data using `docker stats hassio_audio` (or without `hassio_audio` argument to see the stats of all containers).

> I don't know if this is the same issue or wat. But I tried again in order to be 110% sure: if I stop only this container, my file transfer speed go to 100 Mb/s... until it restarts again. After that the transfer rate go to 30 Mb/s.

Maybe audio container running causes high load on another component on the system? `docker stats` should give you a better picture what exactly happens on the system between the two states.

NdR91, 2021-02-17:
> Maybe audio container running causes high load on another component on the system? `docker stats` should give you a better picture what exactly happens on the system between the two states.

I don't see nothing of unusual using docker stats.
Here you can see the output:

[image]

That's really incredible, I didn't have this issue one week ago...

pvizeli, 2021-02-17:
Using a docker container outside of the Add-on ecosystem is very dangerous. I would deinstall all containers which should fix the issues. You have now 2 audio system: our backend and plex - not sure if there is the issue

closed 2021-02-17 by pvizeli

metropt, 2021-02-18:
I'm running HA supervised on thinker board and found the same problem today.
[image]
There are more people here https://community.home-assistant.io/t/hassio-audio-high-cpu/280604

agners, 2021-02-18:
@metropt hm, interesting so in this case no other container is running. This is most likely caused by the recent audio plug-in update. Can you open an issue in https://github.com/home-assistant/plugin-audio. Thanks!

NdR91, 2021-02-19:
@metropt do you plan to open the Issue?

metropt, 2021-02-19:
> @metropt do you plan to open the Issue?

Someone opened a issue.My system is unsupported since I'm running it with armbian.
https://github.com/home-assistant/plugin-audio/issues/53