home-assistant/operating-system #1235: CONFIG_RT_GROUP_SCHED=y leaves docker with no available cpu.rt_runtime_us resources to share

jens-maus, 2021-02-18:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [X] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of development version 6.0dev
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

The use of `CONFIG_RT_GROUP_SCHED=y` in [docker.config](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/kernel/docker.config#L5) results in docker and its child containers to not be able to use system functions like `sched_setscheduler()` to raise a process' realtime priority in case of time critical applications even thought `SYS_NICE` capability is assigned to a container.

Note the following default /sys output in a fresh HomeAssistantOS install:

1. cpu.rt_runtime_use cgroup of system:
   ```sh
   # cat /sys/fs/cgroup/cpu/cpu.rt_runtime_us 
   950000
   ```
2. cpu.rt_runtime_use cgroup of docker:
   ```sh
   # cat /sys/fs/cgroup/cpu/docker/cpu.rt_runtime_us 
   0
   ```
3. cpu.rt_runtime_use cgroup of a started docker container:
    ```sh
    # cat /sys/fs/cgroup/cpu/docker/fc74c542b573e272a600b46c76fbd32dd0d74c8e04914f00cfe504871e97e9b9/cpu.rt_runtime_us 
    0
    ```    

Please note that the output of (2) and (3) suggests that not only docker itself is not allowed to use functions like `sched_setscheduler()` to raise the realtime priority, but also its started containers won't be able to use processes which try to raise the real-time priority of the scheduler. One example of an add-on trying to use a process which uses `sched_setscheduler()` can be found here: https://github.com/jens-maus/RaspberryMatic/issues/1087#issuecomment-780888083. However, other applications might exist which require a process to raise the realtime priority of processes (e.g. like real-time audio apps, etc.)

In addition, due to these reasons it still seems to be a general best practice to generally disable `CONFIG_RT_GROUP_SCHED` on systems running with `systemd`, as can be read up here:

https://github.com/systemd/systemd/blob/main/README#L109-L116
https://forum.armbian.com/topic/16489-config_rt_group_schedy-harmuflull-for-real-time-applications/
https://github.com/systemd/systemd/issues/329

I would therefore like to suggest to remove `CONFIG_RT_GROUP_SCHED=y` from [docker.config](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/kernel/docker.config#L5). I already verified in an own fork that removing it works and allows applications to use `sched_setscheduler()` correctly and without any `Operation not permitted` error.

If wanted I can perfectly submit a pull request removing  `CONFIG_RT_GROUP_SCHED=y`, but before doing so I thought about raising an issue here beforehand to potentially discuss the matter

agners, 2021-02-18:
Thanks for raising this issue and the detailed report, interesting find.

After skimming through, I feel that we should be able to allocate CPU bandwidth somehow, to make `sched_setscheduler()` work in the container again. I'll have a closer look tomorrow.

jens-maus, 2021-02-18:
Of course it is possible to manually set `cpu.rt_runtime_us` to `950000` (e.g. using `echo 950000 >sys/fs/cgroup/cpu/docker/cpu.rt_runtime_us`) for docker and perhaps also for the started containers separately. However, I still think that this might be a principle question here that needs to be answered first.

Also in standard Ubuntu/Linux no `CONFIG_RT_GROUP_SCHED` is enabled and definitely for a good reason I think. And then the developers of `systemd` also probably tinkered around for long time I guess and added this comment to their README for a good reason. And as HomeAssistantOS is using `systemd` itself it IMHO might be best to simply follow the best practices here and disable `CONFIG_RT_GROUP_SCHED` all together as otherwise we might end up sooner or later again here identifying issues that the systemd crew or Ubuntu devs also identified in the past as good reasons not to enable `CONFIG_RT_GROUP_SCHED`.

agners, 2021-02-19:
We could just follow systemd's recommendation/follow other distros and drop it indeed.

Or we could use the feature, which would allow us make sure Add-Ons do not consume to much CPU time. Starting the Docker daemon with `--cpu-rt-runtime=950000` makes sure that its cgroup gets the time allocated:

```
# cat /sys/fs/cgroup/cpu/docker/cpu.rt_runtime_us
950000
```

We still would have to allocate CPU time per container, even when starting `--privileged` it seems that Docker does not allocate any CPU time. But with `docker run -it --privileged --cpu-rt-runtime=950000 --rm debian`, I was able to schedule a real-time task just fine, and also cgroup looks good:

```
# cat /sys/fs/cgroup/cpu/docker/5ba3f6e5ba01e008b70211facd637f9844ebf5f0aaad7de976e72703de08e517/cpu.rt_runtime_us
950000
```

This would need Supervisor support of course. But with that we could also add `-ulimit rtprio=99` and `--cap-add=sys_nice`, which should make `sched_setscheduler()` available without `--privileged`.

See also: https://docs.docker.com/config/containers/resource_constraints/#configure-the-realtime-scheduler

@jens-maus do you need `--privileged` anyways or is this the only reason?

jens-maus, 2021-02-19:
> @jens-maus do you need `--privileged` anyways or is this the only reason?

Currently I don't need `--privileged` for the add-on I am working on. So I am of course in principle fine if you apply your suggested changes and assign containers full `950000` `cpu.rt_runtime_us` per default. I still feel, however, that this imposes too much complexity for something that might not be really used at all (the limiting of rt-runtime) in the end. Also please note, that `cpu.rt_runtime_us` doesn't actually limit CPU time in the sense an ordinary user would understand it. It actually limits the real-time scheduling amount and this might be a tedious and unclear job/decision for add-on developers to decide. And if you anyway assign "950000" to each container you can also completely remove `CONFIG_RT_GROUP_SCHED` as then there is no real benefit in keeping it enabled. And I think that this is exactly the reason why Ubuntu and other distributions decided not to enable `CONFIG_RT_GROUP_SCHED` at all because otherwise it could cause too much trouble if it is incorrectly setup of `cpu.rt_runtime_us`.

But if you decide otherwise I am also fine with your proposed changes of course. So when do you think you can integrate these changes because I want to proceed with my add-on development  ASAP.

agners, 2021-02-25:
From what I understand, you need  `--privileged` or adjusted capabilities and ulimits (`rtprio`) to make use of the real-time scheduler no matter if `CONFIG_RT_GROUP_SCHED` is enabled or disabled. Both have been implemented now in Supervisor (see https://github.com/home-assistant/supervisor/pull/2593 and https://github.com/home-assistant/supervisor/pull/2605). This issue triggered those changes, so already some progress in supporting add-ons with real-time support :+1: .

If we support real-time CPU bandwidth allocation via `CONFIG_RT_GROUP_SCHED` is a separate question. Technically it sounds like a good feature, especially since we use containers to enable 3rdparty addons. Also the Supervisor now has rudimentary support for it (see https://github.com/home-assistant/supervisor/pull/2602).

At this point, thank @pvizeli for all that work!

> Also please note, that cpu.rt_runtime_us doesn't actually limit CPU time in the sense an ordinary user would understand it. It actually limits the real-time scheduling amount and this might be a tedious and unclear job/decision for add-on developers to decide. 

The problem with the real-time schedulers is that they really can monopolize a CPU completely. Weather that is due to a bug or just high usage doesn't really matter, but it will lead to a unresponsive system (from HA point of view). Granted, a sensible value is hard to find, and the "magic default" of 950000 is rather arbitrary. But it should make sure that a system stays somewhat responsive since a RT task will only every be allowed to monopolize 95% of CPU time. If an Add-On uses too much RT time, this 5% will allow to diagnose what task uses so much CPU time.

Given that we have the feature now already setup in Supervisor, I'll try to make the necessary changes to use the feature. Let's see how it goes.

> So when do you think you can integrate these changes because I want to proceed with my add-on development ASAP.

The 2021.03 supervisor will have the necessary changes. As for the OS, the changes are rather minimal. We can pack it into the next stable release 5.13. I'll probably schedule it to be ready around the next supervisor release.

jens-maus, 2021-02-25:
> The 2021.03 supervisor will have the necessary changes. As for the OS, the changes are rather minimal. We can pack it into the next stable release 5.13. I'll probably schedule it to be ready around [...]

That sounds great and I am looking forward to use that feature in my RaspberryMatic add-on as soon as you released the new supervisor+HAos version. And of course I will try to give feedback if it works as expected. Thanks again to you both (@agners and @pvizeli) for quickly integrating it. Looking forward to send in more suggestions and PRs, especially to HAos.

feutl, 2021-02-25:
@agners 
Can these things be tested already, before it gets "rolled" out to productive environments. I know there is a Beta Channel for "Supervisor"
Is this the way to get this features already so I can help @jens-maus testing the add-on?

agners, 2021-02-25:
@feutl yes, there are development builds. I triggered a build with the changes in, they are available at https://os-builds.home-assistant.io/6.0.dev20210225/. You have to switch the Supervisor to the dev channel, and then update

```
ha su options --channel dev
ha su reload
ha os update --version 6.0.dev20210225
```

You'll also need a Supervisor with the latest fixes (https://github.com/home-assistant/supervisor/pull/2630).

agners, 2021-02-26:
The necessary changes hit the dev branch and will get backported for release 5. This will be addressed in 5.13 and later. Closing.

closed 2021-02-26 by agners

jens-maus, 2021-02-27:
Sorry, posted this to the PR instead of here. So here again:

@agners @pvizeli

Sorry to bring this up here again, but I actually tried to check if the changes mentioned here (in HAos and the supervisor) actually work so that the new realtime option in an add-on actually brings up the container with enough real-time cpu time now. For this I installed my own dev version of HAos (from latest dev branch) with the changes from @agners and switched immediately to the beta channel so that supervisor 2021.02.12 was installed and is running (which should contain the changes from @pvizeli). In fact, while I can see that dockerd is perfectly running with --cpu-rt-runtime and the new SUPERVISOR_CPU_RT env variable is propagated to the supervisor, my test add-on container is still outputting the following

```
# cat /sys/fs/cgroup/cpu/cpu.rt_runtime_us
0
```

which suggests that the cpu.rt_runtime_us setup within supervisor does not work as expected even thought I have "realtime": true defined in the config.json of the test add-on. So did someone from you actually test that the new combination of the HAos changes work nicely together with the new 2021.02.12 changes regarding real-time changes for add-ons?

agners, 2021-02-27:
Hm, not sure where that comes from, seems to work here:

```
# docker exec -it addon_local_rt_test /bin/sh
# cat /sys/fs/cgroup/cpu/cpu.rt_runtime_us
950000
```

Using this add-on: https://github.com/agners/home-assistant-addon-rt-tests/. Can you reproduce the issue also with this simple test add-on?

jens-maus, 2021-02-27:
> Using this add-on: https://github.com/agners/home-assistant-addon-rt-tests/. Can you reproduce the issue also with this simple test add-on?

How exactly can I install that addon in HAos? Sorry for asking but here I usually install using the UI Add-on Store functionality.

Counterwise you could also try to install my raspberrymatic test addon using the following URL as a new Add-on store:

https://github.com/jens-maus/ha-addon-raspmatic

Then simply install the `RaspberryMatic CCU` add-on and let it start/run. You should end up with `/sys/fs/cgroup/cpu/cpu.rt_runtime_us` returning `0` even thought the config.json of the add-on has the necessary `realtime` entry. At least that's the case here with using HAos 6.0.dev20210225 and supervisor-2021.02.dev2605

jens-maus, 2021-02-28:
@agners I just updated to `supervisor-2021.02.dev2801` and still have the same issue here. And I just wondered, can it be that this might be actually caused by the fact that the supervisor itself also has `/sys/fs/cgroup/cpu/cpu.rt_runtime_us` of zero and due to the point that the supervisor is the one starting the add-on containers it won't actually start it with the necessary options or permissions to setup `cpu.rt_runtime_us` correctly?!?

agners, 2021-03-01:
> How exactly can I install that addon in HAos? Sorry for asking but here I usually install using the UI Add-on Store functionality.

There are many ways, but a easy way is to use the Supervisor's local build/installation capability. install the Terminal add-on. then do:

```
cd /addons
git clone https://github.com/agners/home-assistant-addon-rt-tests/
```

Go to Supervisor -> Add-on Store, reload via the tripple dot menu on the top right corner. You should see the add-on appearing in the top `Local add-ons` section. With that you should be able to build and start it.

I'll give ha-addon-raspmatic a try on my test setup here.

> And I just wondered, can it be that this might be actually caused by the fact that the supervisor itself also has `/sys/fs/cgroup/cpu/cpu.rt_runtime_us` of zero [...]

The supervisor communicates with the Docker daemon via socket, All `cpu.rt_runtime_us` allocation is done by the Docker daemon, so it should really not matter what time the supervisor container has assigned.

agners, 2021-03-01:
@jens-maus hm, now using the fixed Supervisor version I see that allocation for the add-on actually fails:

```
21-03-01 09:03:22 ERROR (SyncWorker_6) [supervisor.docker] Can't start addon_local_rt_test: 500 Server Error for http+docker://localhost/v1.40/containers/724049edf8a14e75acdd77a2a516909c3224f037052c6a63c286a3253500b3e3/start: Internal Server Error ("OCI runtime create failed: container_linux.go:370: starting container process caused: process_linux.go:326: applying cgroup configuration for process caused: failed to write "950000" to "/sys/fs/cgroup/cpu,cpuacct/docker/724049edf8a14e75acdd77a2a516909c3224f037052c6a63c286a3253500b3e3/cpu.rt_runtime_us": write /sys/fs/cgroup/cpu,cpuacct/docker/724049edf8a14e75acdd77a2a516909c3224f037052c6a63c286a3253500b3e3/cpu.rt_runtime_us: invalid argument: unknown")
```

It seems that `cpu.rt_runtime_us` is the guaranteed limit, and we can't simply allocate `950000` for all the realtime add-ons. I do understand Lennert's comment, the interface is a bit weird to use. I still think its a valuable feature for our use case. I am working on a solution.

agners, 2021-03-01:
The problem I saw will be fixed in the next nightly. I assume that fixes your case as well :crossed_fingers:

jens-maus, 2021-03-01:
> The problem I saw will be fixed in the next nightly. I assume that fixes your case as well 🤞

Ok, thanks for the hint and the potential fix. I will try to test this again as soon as it is shows up in the next nightly build of the supervisor.

However, I am still highly skeptical about this simple distribution of real-time runtime like you implemented. The point being, you cannot simply divide the max. real-time runtime shares to a max of e.g. 5 containers as this might simply leave not enough real-time runtime to a single container, which in fact really requires as much real-time runtime as possible. And this brings me back to my initial suggestion of simply disabling `CONFIG_RT_GROUP_SCHED` altogether in HAos, because I still have the feeling that exactly this special behaviour and dependency in how to distribute the real-time runtime slices is the reason why major linux distributions like Ubuntu, etc. have it disabled as well and that's also why the systemd developers suggest to disable it also. And I still feel we (HAos) should not try to be smarter than these other OS projects where a user can also perfectly start multiple containers and could easily run into the same issue. And probably Ubuntu&Co have learned their lession as well, that the theory behind `CONFIG_RT_GROUP_SCHED` might sound great and fitable to a project like HAos. But the practical implications/complications behind using it might actually disqualify it. And I am almost certain that this won't the the last time that you will have to touch that thing in future again because either 5 might simple not be enough or because a certain add-on (potentially already the raspberrymatic one i am working on) requires more real-time runtime to work stable.

jens-maus, 2021-03-02:
@agners Ok, I could now actually verify that with your latest changes to supervisor the `cpu.cpu_rt_runtime_us` is now non-zero:

```
root@339c13ac-raspberrymatic:~# cat /sys/fs/cgroup/cpu/cpu.rt_runtime_us 
190000
```

Together with the changes here in HAos I was now able to verify that the raspberrymatic add-on I am currently developing is now able to change the real-time scheduler priority of applications it is using. If `190000` is enough have to be seen as soon as the add-on goes beta, thought. However, thanks again for all your work and the discussions here!

agners, 2021-03-02:
>  I am still highly skeptical about this simple distribution of real-time runtime like you implemented. The point being, you cannot simply divide the max. real-time runtime shares to a max of e.g. 5 containers as this might simply leave not enough real-time runtime to a single container,

This obviously will depend on the use case. But most often, real-time operations are not very CPU time consuming, they are just very repetitive and need to be done in a timely manner. Quite often real-time task come by with a 1-5% CPU time of a single core. With the current allocation you can consume almost 80% of a single core on a quadcore system, which really should be sufficient. I think if a real-time task uses more than that, then maybe some operations should be moved to a non-RT task.

If we encounter a reasonable use case which needs much more CPU time, then we can still reconsider, get rid of CPU time allocation or add more settings to allocate smarter.

> I still have the feeling that exactly this special behaviour and dependency in how to distribute the real-time runtime slices is the reason why major linux distributions like Ubuntu, etc. have it disabled as well and that's also why the systemd developers suggest to disable it also.

I get your point, and the interface is definitely a bit weird to use and maybe does not have the flexibility we ideally want. But then, there is Docker which has it implemented and allows to make use of it. Ubuntu as well as systemd write general purpose software for vastly different use cases etc. For them a kernel interface must be generally useful, which often requires a flexible interface. Otherwise they just won't use it like in this case.

> And I still feel we (HAos) should not try to be smarter than these other OS projects where a user can also perfectly start multiple containers and could easily run into the same issue. 

We are not trying to copy Ubuntu or any other distro. If we would, we could just use Ubuntu or any other distro.

Quite the opposite frankly: Running our own OS only make sense if we can add value. Using squashfs is one of those, the RAUC image based update or in this case: Protecting the Core from Add-Ons monopolizing the CPU. Weather that is by design or accident doesn't really matter, I'd rather prefer to have a problem with an Add-On rather than the Core web interface not being accessible or very slow due to Add-Ons.

Let's see how it goes. If it really proves not usable for our use case as well, we can just disable it.

jens-maus, 2021-03-02:
> If we encounter a reasonable use case which needs much more CPU time, then we can still reconsider, get rid of CPU time allocation or add more settings to allocate smarter.

Is there at least a way that you move away from a hardcoded value of 5 to the use of some configuration value somewhere in the HA UI which a user could then change on his own? That would help to provide some kind of support in case users report issues so that we could instruct them to change it to, e.g. 2 (including the internal audio add-on) to increase CPU time allocation gradually. IMHO that would be a good enhancement and finalization of this whole realtime enhancement lately.

> Let's see how it goes. If it really proves not usable for our use case as well, we can just disable it.

Ok, lets' see. And thanks again, really good and professional work 👍

agners, 2021-03-02:
> Is there at least a way that you move away from a hardcoded value of 5 to the use of some configuration value somewhere in the HA UI which a user could then change on his own? 

TBH, I rather not spend time on such a feature. I think if that is required we failed with that feature and we should fix/disable it.

What I'd like to do is distribute all the time to the current realtime addons. That would then give much more CPU time if only 2 are running. But for this we need to be able to adjust allocated time at runtime, and currently code is missing in the Docker Python SDK to adjust this values at runtime. It shouldn't be hard to extend it, I am planning to look into it. With that we'll have the necessary code in place to improve the current static value of 5.

> That would help to provide some kind of support in case users report issues so that we could instruct them to change it to [..]

That is a bit a problem: Even if we add that magic knob, how do we know when to even use it? Not sure if it is possible, I haven't done any research, but what would be cool is if we could detect when an Add-On hits the limit.

> Ok, lets' see. And thanks again, really good and professional work +1

Thank you, like wise! :+1: