home-assistant/operating-system #1237: RasPi GPIO and Bluetooth co-existence requires missing /boot/config.txt options

GitPetri, 2021-02-20:
**Hardware Environment**

- [x] Raspberry Pi [4]

**Home Assistant OS release:**

- Additional information (if accessible):
System Health

version: core-2021.2.3
installation_type: Home Assistant OS
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.7
os_name: Linux
os_version: 5.4.83-v7l
arch: armv7l
timezone: UTC

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: ok

host_os: Home Assistant OS 5.11
update_channel: stable
supervisor_version: supervisor-2021.02.11
docker_version: 19.03.13
disk_total: 14.1 GB
disk_used: 4.5 GB
healthy: true
supported: true
board: rpi4
supervisor_api: ok
version_api: ok
installed_addons: File editor (5.2.0), SSH & Web Terminal (8.0.1), Z-Wave JS to MQTT (0.4.2), Samba share (9.3.0), Ruuvitag (0.3.0), RuuviTag Discovery (0.2.8), Z-Wave JS (0.1.7)

dashboards: 2
resources: 0
views: 3
mode: storage

**Supervisor logs:**
 N/A

**Journal logs:**
N/A

**Kernel logs:**
N/A

**Description of problem:**
Default /boot/config.txt configuration doesn't allow parallel GPIO and Bluetooth usage for Raspberry Pi. I don't know whether the config.txt configuration is responsibility of operating-system or some other component, If this issue is reported for wrong component, please help me identify the correct one (or assign directly to correct one).

In order to use RasPi for example with RazBerry2 Z-Wave HAT and simultaneously still use Bluetooth to monitor for example RuuviTag's the following parameters need to be set in the /boot/config.txt:

> dtoverlay=miniuart-bt
> force_turbo=1

The 1st line (surprisingly) switches BT to the miniUART and the 2nd line is needed with miniUART BT to set the VPU core clock to a fixed frequency, this is required based on https://www.raspberrypi.org/documentation/configuration/uart.md / UARTs and Device Tree -section, 3rd chapter.

agners, 2021-02-26:
> Default /boot/config.txt configuration doesn't allow parallel GPIO  [...]

I guess with GPIO usage you mean using the PL011 UART on the 40 pin header (which confusingly always gets called GPIO header).

This is not a standard setting, so manual `config.txt` adjustments are necessary. Using the mini UART for Bluetooth does limit the maximum speed (from https://www.raspberrypi.org/documentation/configuration/uart.md):

> [..] Note that this may reduce the maximum usable baud rate (see mini UART limitations below) [...]

I'd rather prefer to not have that as a default setting. I suspect that more people are using Bluetooth than the full UART on the GPIO header.

closed 2021-02-26 by agners

GitPetri, 2021-02-26:
> I'd rather prefer to not have that as a default setting. I suspect that more people are using Bluetooth than the full UART on the GPIO header.

Fair enough.