home-assistant/operating-system #1240: Raspberry Pi4 crashes

webnology-ch, 2021-02-23:
Hi, my Raspberry Pi4 crashes after a fiew minutes having installed an addon. It does not matter which addon, I have tried multiple. Rebooting does not solve the problem. 
I have tried the 32-bit version and the 64-bit version with same result. I used a 32GB SD Card on all these tests.
Now I installed the 32-bit version on a 128GB SDcard ant let it run without any addons etc, just the initial configuration.
If it keeps working I will install the mosquitto addon for testing.
Does anybody else have similar issues?

## System Health

version | core-2021.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.83-v7l
arch | armv7l
timezone | UTC

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | undefined
can_reach_cloud_auth | undefined
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.02.11
docker_version | 19.03.13
disk_total | 115.2 GB
disk_used | 1.8 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | undefined
installed_addons | 

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

agners, 2021-02-25:
Not aware of such an issue. How did your tests go with the new SD-card?

When you say "crashing" do you see anything on the HDMI output?

webnology-ch, 2021-02-25:
the first attempts I did were withouta display. I tried two different 32GB sdcards replicating the same error.
to debug I connected a display and saw read write error messages for the logical volume partition. That was the reason
I changed to 128GB sdcard. Now it works fine, with and without addons installed.
I do not know if the sdcard problem was because of the size (32gb) or because both were no name cards (quality related).
My 128gb card is branded.
I still have ony installed the ssh addon and now starting to add more. Hope it continues working fine.

agners, 2021-02-26:
Ok, this sounds like a SD card failure then, nothing uncommon, unfortunately.

closed 2021-02-26 by agners

webnology-ch, 2021-02-27:
well I have used this 32gb sdcard with rasbian without any issue. is maybe the 32gb size too small for home assistant? I have no branded 32gb card to test it by my own.