home-assistant/operating-system #1241: Can Hypver-V's Dynamic Memory feature be fixed in HassOS?

jant90, 2021-02-23:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [X] OVA, VHDX image in Hyper-V on Windows 10

**Home Assistant OS release:**
- [X] Fresh installation of release 5.12, 4.20, 3.13
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**
```
dmesg | grep  hv_
[   0.111196] hv_vmbus: Vmbus version:5.0
[   0.176555] hv_vmbus: registering driver hyperv_fb 
[   0.186906] hv_vmbus: registering driver hv_storvsc 
[   0.188264] hv_vmbus: registering driver hv_netvsc 
[   0.711577] hv_vmbus: registering driver hv_balloon 
[   0.712043] hv_utils: Registering HyperV Utility Drier
[   0.712418] hv_balloon: Using Dynamic Memory protocol version 2.0 
[   0.712530] hv_vmbus: registering driver hv_utils
[   0.713553] hv_vmbus: registering driver hyperv_keyboard 
[   0.714301] hv_utils: Heartbeat IC version 3.0
[   0.714904] hv_utils: Shutdown IC version 3.0 
[   0.715339] hv_utils: TimeSync IC version 4.0
[  48.809874] hv_balloon: Max. dynamic memory size: 2048 MB 
[  50.067095] hv_balloon: Memory hot add failed
```

**Description of problem:**
Even though though the hv_balloon kernel module is present in HassOS it's not working. The dmesg output above shows the error.

I tested this with various HassOS 5.X releases (including 5.12) and 4.20 and 3.13 and all give the same error (`hv_balloon: Memory hot add failed`). Can this be fixed? Maybe the hv_vmbus module is required? I don't see it with `lsmod`:
```
lsmod
Module				Size	Used by 
xfrm_user			36864	1
hv_balloon			32768	0
hyperv_keyboard			16384	0
hv_utils			28672	0
```

agners, 2021-02-26:
Hm, from what I can tell the necessary options should be enabled, also `CONFIG_MEMORY_HOTPLUG` is set to `y`.

It sounds similar to this problem: https://community.clearlinux.org/t/dynamic-memory-broken-on-hyper-v/3891

Did you try starting with maximum memory?

jant90, 2021-03-01:
@agners the 2GB limit is the upper limit I set for this particular VM in Hyper-V. The issue is different than what's described in the the linked thread because it's not scaling down either when starting with a lot of memory.

Anyways, I built the today's dev version myself (mainly because initially I didn't see dev builds were available, but hosted elsewhere) and Hyper-V Dynamic Memory works properly with that release, possibly because it runs on a newer kernel (5.10)?

When can we expect kernel 5.10 in a stable version of HassOS? Will v6 be the next stable version?

agners, 2021-03-02:
Thanks for the update, that is good news. I guess it got fixed in upstream Linux then.

The nightly builds are available at https://os-builds.home-assistant.io/.

Instead of using your own build you can use that nightly build, it will allow you to update to the stable once it is out. But having things backed up etc is still advisable :)

There will probably be 1-2 bug fix releases with version 5. I expect release 6 to become stable towards end of March/early April.

closed 2021-03-17 by agners