home-assistant/operating-system #1242: WiFi dongle drivers fail to load

yozik04, 2021-02-23:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 2020.x
- Additional information (if accessible):

Version | core-2021.2.1
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.7
Operating System Family | Linux
Operating System Version | 5.4.83-v7
CPU Architecture | armv7l
Timezone | Europe/Tallinn

**Supervisor logs:**

Irrelevant
**Journal logs:**

Irrelevant
**Kernel logs:**

Relevant log:
```
[   12.485748] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   12.485905] rt2800usb 1-1.2:1.0: Direct firmware load for rt2870.bin failed with error -2
[   12.485928] ieee80211 phy0: rt2x00lib_request_firmware: Error - Failed to request Firmware
```

Full log:
```
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 5.4.83-v7 (builder@afcfacd4d5bd) (gcc version 9.3.0 (Buildroot -g1a551c07)) #1 SMP Sun Jan 31 02:31:00 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49868 r8192 d23860 u81920
[    0.000000] pcpu-alloc: s49868 r8192 d23860 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=57921e056a7342698680b0a0da47af91 cgroup_enable=memory fsck.repair=yes coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootfstype=squashfs ro rauc.slot=B
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 879412K/970752K available (10240K kernel code, 731K rwdata, 2896K rodata, 1024K init, 844K bss, 25804K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32551 entries in 64 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[... 539 lines cut ...]
[  235.373938] usb 1-1.4: Manufacturer: Apple, Inc.
[  235.373956] usb 1-1.4: SerialNumber: 000000000000
[  235.376157] hub 1-1.4:1.0: USB hub found
[  235.376525] hub 1-1.4:1.0: 3 ports detected
[  235.693195] usb 1-1.4.2: new low-speed USB device number 8 using dwc_otg
[  235.841772] usb 1-1.4.2: New USB device found, idVendor=05ac, idProduct=0221, bcdDevice= 0.69
[  235.841808] usb 1-1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  235.841830] usb 1-1.4.2: Product: Apple Keyboard
[  235.841849] usb 1-1.4.2: Manufacturer: Apple, Inc
[  235.999489] input: Apple, Inc Apple Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2:1.0/0003:05AC:0221.0001/input/input0
[  236.074462] apple 0003:05AC:0221.0001: input,hidraw0: USB HID v1.11 Keyboard [Apple, Inc Apple Keyboard] on usb-3f980000.usb-1.4.2/input0
[  236.077121] apple 0003:05AC:0221.0002: Fn key not found (Apple Wireless Keyboard clone?), disabling Fn key handling
[  236.077784] input: Apple, Inc Apple Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4.2/1-1.4.2:1.1/0003:05AC:0221.0002/input/input1
[  236.217967] apple 0003:05AC:0221.0002: input,hidraw1: USB HID v1.11 Device [Apple, Inc Apple Keyboard] on usb-3f980000.usb-1.4.2/input1
[  241.433480] Voltage normalised (0x00000000)
[  290.257592] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[  290.259237] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xD1E1
```
**Description of problem:**

- Problem is reproducible on RPI2
- WiFi dongle was working before on previous year versions
- WiFi dongle works properly on other hardware

WiFi dongle on RPI2 stopped working after update.

agners, 2021-05-19:
This is fixed in 6.0.rc1 and later.

yozik04, 2021-05-19:
Thank you @agners

keanew, 2021-07-28:
Hi, 

I'm trying to use my dongle on the Home Assistant OS 6.1 version with a RPI 4. SO it's a version after the 6.0.rc1
I have a Ralink RT5370 DONGLE Wifi USB adapter but the driver isn't found. 
 
Output log of dmesg command : 
```
[11630.291528] ieee80211 phy1: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[11630.291611] rt2800usb 1-1.3:1.0: Direct firmware load for rt2870.bin failed with error -2
[11630.291631] ieee80211 phy1: rt2x00lib_request_firmware: Error - Failed to request Firmware
```

How can I make it run ?

agners, 2021-07-29:
@keanew use HAOS 6.2. Currently available in the beta channel

keanew, 2021-07-29:
Ok thanks. I'll try later. 

Do you know when the 6.2 will be released in a stable version ?

agners, 2021-07-29:
Probably tomorrow.

keanew, 2021-07-29:
It works! Thank you! :)