home-assistant/operating-system #1243: Lost WiFi after 5.9->5.12 upgrade

Masterz69, 2021-02-25:
**Hardware Environment**

- [X ] Raspberry Pi [1/2/3/**4**]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ X] Updated from version 5.9->5.12
- Additional information (if accessible):

➜  ~ ha info
arch: aarch64
channel: stable
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.12"
homeassistant: 2021.2.3
hostname: homeassistant
logging: info
machine: raspberrypi4-64
operating_system: Home Assistant OS 5.12
state: running
supervisor: 2021.02.11
supported: true
supported_arch:
- aarch64
- armv7
- armhf
timezone: Europe/Riga

➜  ~ ha su logs
[s6-finish] sending all processes the KILL signal and exiting.
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[07:35:51] INFO: Setup udev backend inside container
[07:35:51] INFO: Update udev information
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[07:35:52] INFO: Starting local supervisor watchdog...
21-02-25 07:35:55 INFO (MainThread) [__main__] Initializing Supervisor setup
21-02-25 07:35:55 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-02-25 07:35:55 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
21-02-25 07:35:55 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.02.11
21-02-25 07:35:55 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-02-25 07:35:55 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-02-25 07:35:55 INFO (MainThread) [__main__] Setting up Supervisor
21-02-25 07:35:55 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-02-25 07:35:55 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-02-25 07:35:56 INFO (MainThread) [supervisor.host.info] Updating local host information
21-02-25 07:35:56 INFO (MainThread) [supervisor.host.services] Updating service information
21-02-25 07:35:56 INFO (MainThread) [supervisor.host.network] Updating local network information
21-02-25 07:35:58 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-02-25 07:35:58 INFO (MainThread) [supervisor.host] Host information reload completed
21-02-25 07:35:58 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-02-25 07:35:58 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
[... 601 lines cut ...]
[   78.216988] hassio: port 7(veth80d8eac) entered blocking state
[   78.216997] hassio: port 7(veth80d8eac) entered forwarding state
[   78.326425] hassio: port 7(veth80d8eac) entered disabled state
[   78.849449] eth0: renamed from veth9041a2f
[   78.884531] IPv6: ADDRCONF(NETDEV_CHANGE): veth80d8eac: link becomes ready
[   78.884695] hassio: port 7(veth80d8eac) entered blocking state
[   78.884702] hassio: port 7(veth80d8eac) entered forwarding state
[   84.085453] hassio: port 6(veth60287d2) entered disabled state
[   84.089396] vetha2aa102: renamed from eth0
[   84.181698] hassio: port 6(veth60287d2) entered disabled state
[   84.185457] device veth60287d2 left promiscuous mode
[   84.185486] hassio: port 6(veth60287d2) entered disabled state
[   84.185532] kauditd_printk_skb: 12 callbacks suppressed
[   84.185536] audit: type=1700 audit(1614238724.126:135): dev=veth60287d2 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[  263.090797] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  280.099633] bcmgenet fd580000.ethernet eth0: Link is Down
[  323.108393] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[  370.082476] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

On 5.9 both eth0 & wlan0 ports was configured for static IP.
After upgrade to 5.12 wlan0 config lost and UI does not allow configure DHCP or static IP on wlan0 
- always Disabled both IPv4 & IPv6.

While eth0 connected - not possible to configure wlan0 IP (DHCP ot static).
On eth0 disconnected - have no idea, as no connectivity.

agners, 2021-02-26:
Just tried to reproduce this: Installed 5.9 from scratch, setup WiFi with fixed IP, updated to 5.12. But things have still been connected, all good here.

Is this still happening on your end? Can you reproduce the problem? Have there been updates on other components?

Masterz69, 2021-02-26:
Have there been updates on other components? AFAIK - not.

On 5.9 was able to in "one shot" connect to WiFi, set wlan0 static IP, save - that's all.

On 5.12 only way found : connect to WiFi, reboot, set wlan0 to DHCP, reboot, switch to static and change IP.
After that configuration still in place after reboot and working fine.

If set wlan0 ipv6 to dhcp, f.e. - all wlan0 config disappears.
Tried twice.

agners, 2021-02-26:
> If set wlan0 ipv6 to dhcp, f.e. - all wlan0 config disappears.

That I can reproduce here indeed as well. Interestingly, though it seems that the NetworkManager has the settings setup accordingly (e.g. `nmcli device show wlan0` shows the static IPv4 address and the IPv6 addresses). But the frontend as well as `ha network info` is somewhat empty:

```
- connected: true
  enabled: true
  interface: wlan0
  ipv4:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  ipv6:
    address: []
    gateway: null
    method: disabled
    nameservers: []
  primary: false
  type: wireless
  vlan: null
  wifi:
    auth: wpa-psk
    mode: infrastructure
    signal: 67
    ssid: The Promised LAN 5G
```

When I then just reboot, settings seem to show up correctly/

I am moving this to the Supervisor repository as this seems more a Supervisor issue.