home-assistant/operating-system #1244: BT dies occasionallyon RasPi 4

GitPetri, 2021-02-25:
**Hardware Environment**

- [x ] Raspberry Pi 4

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

Version | core-2021.2.3
-- | --
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.7
Operating System Family | Linux
Operating System Version | 5.4.83-v7l
CPU Architecture | armv7l
Timezone | Europe/Helsinki

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

```
~ dmesg | grep brcmfmac
[    2.220103] brcmfmac: F1 signature read @0x18000000=0x15264345
[    2.228747] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    2.229003] usbcore: registered new interface driver brcmfmac
[    2.304168] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    2.496690] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    2.509738] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    5.357514] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

```
~ dmesg | grep Bluet
[    6.413122] Bluetooth: Core ver 2.22
[    6.413174] Bluetooth: HCI device and connection manager initialized
[    6.413191] Bluetooth: HCI socket layer initialized
[    6.413200] Bluetooth: L2CAP socket layer initialized
[    6.413214] Bluetooth: SCO socket layer initialized
[    6.420476] Bluetooth: HCI UART driver ver 2.3
[    6.420486] Bluetooth: HCI UART protocol H4 registered
[    6.420525] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    6.420669] Bluetooth: HCI UART protocol Broadcom registered
[    6.569544] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    6.569553] Bluetooth: BNEP filters: protocol multicast
[    6.569570] Bluetooth: BNEP socket layer initialized
[   45.476413] Bluetooth: RFCOMM TTY layer initialized
[   45.476430] Bluetooth: RFCOMM socket layer initialized
[   45.476446] Bluetooth: RFCOMM ver 1.11
[  583.993489] Bluetooth: hci0: advertising data len corrected
[  996.874863] Bluetooth: hci0: Frame reassembly failed (-84)
[  996.875269] Bluetooth: hci0: Frame reassembly failed (-84)
[ 2758.942447] Bluetooth: hci0: Dropping invalid advertising data
[ 3612.676209] Bluetooth: hci0: Frame reassembly failed (-84)
[ 3612.676225] Bluetooth: hci0: Dropping invalid advertising data
[ 3612.676232] Bluetooth: hci0: unknown advertising packet type: 0x32
[ 3612.677836] Bluetooth: hci0: Frame reassembly failed (-84)
```

**Description of problem:**

I have /boot/config.txt configured for simultaneous GPIO and Bluetooth (see https://github.com/home-assistant/operating-system/issues/1237) with
```
dtoverlay=miniuart-bt
force_turbo=1
```
and the Bluetooth device dies occasionally. I am not sure whether the GPIO/BT co-existence is the reason or just BT and WiFi.
To my understanding the BT issue is fixed in bluez-firmware 1.2-4+rpt8 according to
https://github.com/RPi-Distro/firmware-nonfree/issues/8#issuecomment-770057829
and 
https://github.com/RPi-Distro/bluez-firmware/blob/master/debian/changelog

I assume that current HA OS doesn't have the fix in. As this is OS level problem, restarting HA doesn't help but the Pi requires rebooting to recover.

agners, 2021-02-26:
Hm, using the mini UART does not have flow control and a very shallow FIFO (see also my comment on https://github.com/home-assistant/operating-system/issues/1237#issuecomment-786728822 and the linked RPi documentation). Put differently: Using a UART device on the primary PL011 and Bluetooth at the same time on Raspberry Pi is kinda problematic.

That said, theoretically follow Raspberry Pi's guideline it should setup things such that it kinda works... Not sure if something goes wrong in the HAOS case now.

> To my understanding the BT issue is fixed in bluez-firmware 1.2-4+rpt8 according to

The changelog mentions a Spectra fix, which is a security issue. I don't think that this is related what you are seeing.

agners, 2021-02-26:
What you can try is using `core_freq=250` instead of `force_turbo=1`, maybe this helps to mitigate the problem.

I'll update the Bluetooth firmware as those security fixes are worthwhile. But I don't expect it to solve this issue tbh.

agners, 2021-02-26:
Firmware bump: https://github.com/home-assistant/operating-system/pull/1250

GitPetri, 2021-02-26:
I'll test the core_freq as soon as I have time this weekend and monitor BT.

GitPetri, 2021-02-27:
Zero BT error messages visible in dmesg after the replacement above as of now.

agners, 2021-03-03:
So it seems `core_freq=250` does fix the communication issues then :+1:

closed 2021-03-03 by agners

GitPetri, 2021-03-03:
> So it seems `core_freq=250` does fix the communication issues then 👍

Yes, zero BT errors visible as of now after 4 days