home-assistant/operating-system #1249: Problem booting HassOS 5.12 on NUC8i5BEK with a USB modem Huawei E3531i-2 connected

browetd, 2021-02-26:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release HA 2021.2.3, supervisor 2021.02.11, HassOS 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.99
arch | x86_64
timezone | Europe/Brussels

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4849
Installed Version | 1.11.3
Stage | running
Available Repositories | 751
Installed Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.02.11
docker_version | 19.03.13
disk_total | 228.5 GB
disk_used | 12.9 GB
healthy | true
supported | true
board | intel-nuc
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (8.0.3), Check Home Assistant configuration (3.6.0), File editor (5.2.0), Log Viewer (0.9.1), Mosquitto broker (5.1), Samba share (9.3.0), Visual Studio Code (3.1.0), Duck DNS (1.12.5), NGINX Home Assistant SSL proxy (3.0.1), MariaDB (2.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 7
views | 24
mode | yaml

</details>
**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

The problem: When I boot on my new Nuc with the USB modem attached (Huawei E3531i-2), I am getting an error message: "efi-block-io handle-000000007607ce98: Cannot read MBR/partition table".  I remove the modem, the NUC is booting.  I attach than the modem while booting or afterwards and everything is going well: modem is working as expected...  See the picture for more details...  Any additional information/journal/trace you need ?  (by the way I opened a support ticket from Intel not knowing if the problem is in the Bios or somewhere else)...   Thanks !

[image]

agners, 2021-02-26:
Hm, is this modem maybe exposing a USB mass storage device, and the firmware tries to boot from it?

Does adjusting the boot order maybe help?

browetd, 2021-02-26:
That's what I thought so I disabled "secure boot", "boot from usb" and "legacy boot" from the default values in the Bios (I installed also the latest version of the Bios)... So the only visible device in the list is the SSD drive and it is the first on the list... but it does not boot if modem is connected... I am lost... 

[image]
[image]
[image]

browetd, 2021-02-26:
The modem is configured in serial mode (12d1:1001) and recognised like this by HassOS but I tried another modem (same model) and the problem is the same.  Those models can also be configured as a Hilink Modem (12d1:14dc) or as mass storage (12d1:1f01), this is the default...

agners, 2021-02-26:
Yeah not sure, it really seems that Barebox is picking that USB mass storage device somehow. From what I understand the UEFI BIOS should mount the ESP where it picked up the bootloader and then go from there (see https://www.barebox.org/doc/2020.04.0/boards/efi.html#efi-file-io-interface). Maybe the Barebox shell allows to debug the situation a bit better.

agners, 2021-02-26:
Hm, this might be just due to the fact that we have MBR partition table disabled. I created a build which enables it, can you try this development build? https://os-builds.home-assistant.io/hassos_intel-nuc-6.0.dev20210226.img.xz

browetd, 2021-02-27:
I tried this new image, same issue... but what I noticed: if I reboot the node using the supervisor menu, it is shutting down and rebooting without issues (modem connected)... The problem occurs only if I switch off/on the NUC using the power button in front of the NUC...

agners, 2021-03-01:
So you get "Cannot read MBR/partition table" as well?

browetd, 2021-03-01:
Yes I got the same message with the development build: "Cannot Read MBR/partition table"...  I was surprised too that the NUC started without problem after a "reboot"... but still not after a power off/power on...  Strange indeed

browetd, 2021-03-03:
Just wondering if the installation procedure of HassOS could be impacting the boot ?...  I used "gnome-disk" to restore the image to the SSD and "gparted" to expand the last partition to maximize the disk space... before booting to the SSD

browetd, 2021-03-12:
@agners Yesterday, Intel Customer Support asked me to create a video showing the boot with and without modem... I was unable to create that video on my screen (Samsung LU28E570DS, screen definition 3840x2160), characters are too small... So I decided to change screen and use an ACER H223HQ (screen definition 1920x1080)... And surprise !  The NUC is booting on the ACER with the modem connected but is unable to boot when the ACER is replaced by the Samsung (but is booting with the Samsung screen if the modem is disconnected)... Strange, strange, strange... Any explanations ?  Thank you for your help

browetd, 2021-03-14:
Finally the problem was due to the screen attached to the NUC.  The NUC can boot with the modem but without screen attached or with a low definition screen (fullHD f.i.).... So problem solved.  Thanks to Stefan fot his help !

closed 2021-03-14 by browetd