home-assistant/operating-system #1251: Spontaneous popping/clicking audio from USB speaker after upgrading to HA OS 5.12

jodur, 2021-02-27:
**Hardware Environment**

- [X] Raspberry Pi 4
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

After upgrading to OS 5.12 i had a popping/clicking audio noise on my USB speaker. 
It is a repeating  single pop/click after a few minutes while nothing is being played.

Reverting back to OS 5.11 resolved the problem

agners, 2021-02-27:
Hm, the only thing which changed is the firmware. It seems strange to me that this influences USB, let alone a USB speaker...

Are you sure the OS is the reason and not another component? What type of speakers are those? Is the log (like Audio log in Supervisor) saying anything special?

jodur, 2021-02-28:
Just switched back to 5.12 again to check if the problem appears again and it did immediately when 5.12 was running.
I'am using a usb speaker from adafruit [(https://www.adafruit.com/product/3369)](https://www.adafruit.com/product/3369)
After switching back again to 5.11 the popping sound was gone. I looked in the audio log, OS log and supervisor log, but nothing weirds that could point to this issue. Maybe a issue in the firmware that is causing this problem?

agners, 2021-03-01:
Hm, it seems like. You can test if that is the case by downgrading to 5.11, and use the firmwares used in 5.12 (e.g. by coping them from the boot partition, downgrade, and copy them back using a PC, or just downgrade to 5.11 and replace the files with the ones from the 5.11 release, which you can find here: https://github.com/raspberrypi/firmware/tree/2ba11f2a07760588546821aed578010252c9ecb3/boot).

closed 2022-01-11 by github-actions[bot]