home-assistant/operating-system #1253: CoreState.FREEZE How to fix this issue?

kzeidan, 2021-03-01:
**Hardware Environment**

- [ x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ x] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

21-03-01 12:03:27 ERROR (MainThread) [supervisor.api.ingress] Ingress error: Cannot connect to host 172.30.32.1:64402 ssl:default [Connect call failed ('172.30.32.1', 64402)]
21-03-01 12:46:28 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
21-03-01 12:49:45 INFO (MainThread) [supervisor.snapshots] Found 1 snapshot files
21-03-01 12:53:11 INFO (MainThread) [supervisor.snapshots] Removed snapshot file 364dd5d6
21-03-01 12:57:55 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-8230' coro=<_websocket_forward() done, defined at /usr/src/supervisor/supervisor/api/ingress.py:277> exception=ConnectionResetError('Cannot write to closing transport')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/api/ingress.py", line 282, in _websocket_forward
    await ws_to.send_str(msg.data)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/web_ws.py", line 300, in send_str
    await self._writer.send(data, binary=False, compress=compress)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/http_websocket.py", line 687, in send
    await self._send_frame(message, WSMsgType.TEXT, compress)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/http_websocket.py", line 650, in _send_frame
    self._write(header + message)
  File "/usr/local/lib/python3.8/site-packages/aiohttp/http_websocket.py", line 660, in _write
    raise ConnectionResetError("Cannot write to closing transport")
ConnectionResetError: Cannot write to closing transport
21-03-01 13:02:01 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
21-03-01 13:02:36 WARNING (MainThread) [supervisor.jobs] 'SnapshotManager.do_snapshot_full' blocked from execution, system is not running - CoreState.FREEZE
21-03-01 13:03:57 INFO (SyncWorker_1) [supervisor.docker.interface] Restarting homeassistant/raspberrypi4-homeassistant
21-03-01 13:09:14 INFO (MainThread) [supervisor.homeassistant.core] Wait until Home Assistant is ready
21-03-01 13:11:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-03-01 13:11:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
21-03-01 13:11:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-03-01 13:11:04 INFO (MainThread) [supervisor.homeassistant.core] Detect a running Home Assistant instance
21-03-01 13:11:04 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-03-01 13:11:04 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-03-01 13:11:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not str
21-03-01 13:11:57 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
21-03-01 13:12:02 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-03-01 13:12:02 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-03-01 13:12:03 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-03-01 13:15:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not str
21-03-01 13:15:13 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
21-03-01 13:15:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-03-01 13:15:18 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-03-01 13:15:18 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-03-01 13:16:05 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API error: Received message 257:None is not str
21-03-01 13:16:05 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API connection is closed
21-03-01 13:16:10 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-03-01 13:16:10 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-03-01 13:16:10 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
21-03-01 13:18:30 WARNING (MainThread) [supervisor.jobs] 'SnapshotManager.do_snapshot_full' blocked from execution, system is not running - CoreState.FREEZE
**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2021-03-02:
Seems like a problem in Supervisor.

kzeidan, 2021-03-06:
The problem seems to be that i rebooted the host while a snapshot was being created. rebooting again resolved the issue

closed 2021-03-17 by agners