home-assistant/operating-system #1254: SPI MySensors GW stopped working

lustasag, 2021-03-01:
**Hardware Environment**

-  Raspberry Pi 4

**Home Assistant OS release:**
- Updated from version 5.11
- Additional information (if accessible):
```
arch: armv7
channel: beta
docker: 19.03.13
features:
- reboot
- shutdown
- services
- network
- hostname
- hassos
hassos: "5.12"
homeassistant: 2021.3.0b3
hostname: homeassistant
logging: info
machine: raspberrypi4
operating_system: Home Assistant OS 5.12
state: running
supervisor: 2021.02.12
supported: true
supported_arch:
- armv7
- armhf
timezone: Europe/Prague
```
I have spidev enabled in config.txt. 

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
I have been running a mysensors ethernet gateway in a separate docker container to talk to a sensor via RF24. It has stopped working at around the time that I upgraded to 5.12. It uses the /dev/spidev0.0 interface. I always had a flaky connection, but after the update it completely stopped working.

Nothing shows up in the system logs (as expected). I keep track of packet logs in the mysensors gw image, and they show no signs of communication. The transport is initialized and the broadcast messages are sent, yet nothing is received from the sensor.

I've spent 2 days debugging without avail. So I decided to post a report, perhaps others have experienced the same issue.

agners, 2021-03-02:
There was no change to the Raspberry Pi kernel in that release, so it seems a bit strange that this is due to the OS update.

There have also been changes in hardware handling recently. Can you try to downgrade OS again and see if that makes it work again?

```
ha os update --version 5.11
```

lustasag, 2021-03-02:
Oh, thanks for the tip, I didn't think it was so easy to downgrade! I downgraded (both the OS and the core) and the problem still persisted, so the issue must be something else than the OS. I'm closing this issue then.

closed 2021-03-02 by lustasag