home-assistant/operating-system #1255: Can't run hassos in virtualbox on windows 10?

UnbendableStraw, 2021-03-01:
**Hardware Environment**
Windows 10 Pro (up to date), Oracle VirtualBox 6.1.18
Virtualbox appliance configured for 4 cores, 4gb ram, bridged network adapter, EFI

**Home Assistant OS release:**
HassOS 5.12 for Windows VirtualBox
-->

[image]

**Description of problem:**

agners, 2021-03-02:
I assume you are following the instructions provided here?
https://www.home-assistant.io/installation/windows

Not sure, but maybe the IDE Secondary Device, the optical drive causes problems. Can you disable it?

UnbendableStraw, 2021-03-02:
The VirtualBox instructions leave little to miss;

```Create a new virtual machine
Select “Other Linux (64Bit)
Select “Use an existing virtual hard disk file”, select the VDI file from above
Edit the “Settings” of the VM and go “System” then Motherboard and Enable EFI
Then “Network” “Adapter 1” Bridged and your adapter.```

I've moved onto HyperV and its working fine there. Thank you

closed 2021-03-02 by UnbendableStraw