home-assistant/operating-system #1256: Raspberry Pi locks up on OS above 5.4 using SD and SSD per instructions

bschatzow, 2021-03-02:
**Hardware Environment**

- [ X] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

arch: aarch64
audio_input: None
audio_output: None
boot: true
image: homeassistant/raspberrypi4-64-homeassistant
ip_address: 172.30.32.1
last_version: 2021.3.0b5
machine: raspberrypi4-64
port: 8123
ssl: true
update_available: false
version: 2021.3.0b5
version_latest: 2021.3.0b5
wait_boot: 600
watchdog: true

**Supervisor logs:**
21-03-02 00:22:45 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-03-02 00:22:45 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2021.01.0
21-03-02 00:22:45 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-03-02 00:22:45 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_dns application
21-03-02 00:22:45 INFO (SyncWorker_1) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2021.01.0 - 172.30.32.3
21-03-02 00:22:45 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-03-02 00:22:45 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 2021.02.1
21-03-02 00:22:45 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-03-02 00:22:45 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
21-03-02 00:22:46 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/aarch64-hassio-audio with version 2021.02.1 - 172.30.32.4
21-03-02 00:22:46 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2021.02.1
21-03-02 00:22:46 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-03-02 00:22:46 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_cli application
21-03-02 00:22:47 INFO (SyncWorker_1) [supervisor.docker.cli] Starting CLI homeassistant/aarch64-hassio-cli with version 2021.02.1 - 172.30.32.5
21-03-02 00:22:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2020.10.1
21-03-02 00:22:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 3
21-03-02 00:22:47 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-03-02 00:22:47 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_multicast application
21-03-02 00:22:47 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast homeassistant/aarch64-hassio-multicast with version 3 - Host
21-03-02 00:22:47 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/beta.json
21-03-02 00:22:48 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/raspberrypi4-64-homeassistant with version 2021.3.0b5
21-03-02 00:22:48 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 27 Home Assistant secrets
21-03-02 00:22:48 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.12 / BootSlot A
21-03-02 00:22:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/df843657 repository
21-03-02 00:22:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
21-03-02 00:22:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
21-03-02 00:22:48 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/cebe7a76 repository
21-03-02 00:22:49 INFO (MainThread) [supervisor.store] Loading add-ons from store: 111 all - 111 new - 0 remove
21-03-02 00:22:49 INFO (MainThread) [supervisor.addons] Found 17 installed add-ons
21-03-02 00:22:49 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/tasmoadmin/aarch64 with version 0.14.0
21-03-02 00:22:49 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to esphome/esphome-hassio-aarch64 with version 1.16.2
21-03-02 00:22:49 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/ssh/aarch64 with version 8.0.3
21-03-02 00:22:49 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-samba with version 9.3.1
21-03-02 00:22:49 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-duckdns with version 1.12.5
21-03-02 00:22:49 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/portainer/aarch64 with version 1.4.0
21-03-02 00:22:49 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/vscode/aarch64 with version 3.1.0
21-03-02 00:22:49 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 5.1.1
21-03-02 00:22:49 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-aarch64 with version 0.103.1
21-03-02 00:22:49 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to hassioaddons/log-viewer-aarch64 with version 0.9.1
21-03-02 00:22:49 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mariadb with version 2.2.2
[... 674 lines cut ...]
[  433.122031] hassio: port 13(veth2bbde83) entered disabled state
[  433.782967] eth0: renamed from vethca02837
[  433.806916] IPv6: ADDRCONF(NETDEV_CHANGE): veth2bbde83: link becomes ready
[  433.807064] hassio: port 13(veth2bbde83) entered blocking state
[  433.807070] hassio: port 13(veth2bbde83) entered forwarding state
[  434.617777] hassio: port 14(veth1f70882) entered blocking state
[  434.617785] hassio: port 14(veth1f70882) entered disabled state
[  434.618104] device veth1f70882 entered promiscuous mode
[  434.618894] kauditd_printk_skb: 26 callbacks suppressed
[  434.618901] audit: type=1700 audit(1614649633.598:273): dev=veth1f70882 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  435.668894] audit: type=1325 audit(1614649634.646:274): table=nat family=2 entries=0
[  435.684317] audit: type=1325 audit(1614649634.662:275): table=filter family=2 entries=0
[  435.697334] audit: type=1325 audit(1614649634.678:276): table=nat family=2 entries=5
[  435.705567] audit: type=1325 audit(1614649634.686:277): table=nat family=2 entries=7
[  435.715132] audit: type=1325 audit(1614649634.694:278): table=nat family=2 entries=8
[  435.723009] audit: type=1325 audit(1614649634.702:279): table=nat family=2 entries=10
[  435.728068] audit: type=1325 audit(1614649634.706:280): table=nat family=2 entries=11
[  435.737487] audit: type=1325 audit(1614649634.718:281): table=nat family=2 entries=12
[  435.746710] audit: type=1325 audit(1614649634.726:282): table=nat family=2 entries=13
[  435.803258] eth0: renamed from vethab64d13
[  435.828304] IPv6: ADDRCONF(NETDEV_CHANGE): veth1f70882: link becomes ready
[  435.828445] hassio: port 14(veth1f70882) entered blocking state
[  435.828452] hassio: port 14(veth1f70882) entered forwarding state

**Description of problem:**

agners, 2021-03-02:
Thank you for testing with `datactl`! As I alluded in my comment (https://github.com/home-assistant/operating-system/issues/1119#issuecomment-778632802), I kinda feared that this might be reproducible with the `datactl` case as well. It does make sense, since the crash happens when Linux is running. But it is good that this is tested, as it rules out any firmware/bootloader (U-Boot) involvement.

You wrote somewhere Startech 3.1 controller and Kingston SSD, which model is this exactly? Is it this enclosure? https://www.startech.com/en-us/hdd/s251bmu313? And which Kingston SSD?

Now the kernel we use on Raspberry Pi comes straight from the Raspberry Pi foundation (github.com/raspberrypi/kernel). What would be interesting is if the crash also is reproducible with Raspberry Pi OS (and its kernel 5.4). But Raspberry Pi OS recently moved to 5.10, and I am working on moving Home Assistant OS to it as well.

I'll do a 6.0 development build with that new kernel soon, I'd suggest you give that a try. If it fixes the issue, then we just move forward. If not, I think the way forward is to try reproduce the issue on Raspberry Pi OS and report with the Raspberry Pi kernel developers.

Btw, since you opened this issue with more/new information, can we close https://github.com/home-assistant/operating-system/issues/1177?

bschatzow, 2021-03-02:
I went to the samsung SSD Samsung SSD 860 EVO 250GB 2.5 Inch SATA III Internal SSD (MZ-76E250B/AM), as I want to have a second SSD to test.  I have used the EVO series in windows for years with no problems.  The StarTec model is USB312SAT3CB.  I also went to the 2/22 Pi firmware.

I have read in different threads were people tested what you recommended and they have zero issues using the Pi with the Raspberry software.  Again a difference here is that the Raspberry software is 32 bit vs 64.  I have asked about the freeze issue in the Raspberry forums and every answer I have received says check the Kernel.  

I have also asked repeatedly why booting from SD / SSD combo is more reliable and this was never answer.  My issue is not the boot issue as many people have / had.  My issue is after so many hours a system crash.  I still think once the unit boots it should not matter what method was used.  

My system crashed again in three hours.

Should I go back to OS 5.4?  Should I go back to SSD only boot?  I willing to test what you think can help.  There are also others in the 1119 issue as well that will test for you.  We just need directions.

agners, 2021-03-02:
> Again a difference here is that the Raspberry software is 32 bit vs 64

Sorry I lost a bit track of what all you tested, we do have a 32-bit image, did you test 32-bit?

> I have also asked repeatedly why booting from SD / SSD combo is more reliable and this was never answer. 

There are countless reports where it just doesn't boot, see #1257. SD card approach sidesteps that.

That said, if you boot from SD, initialization of controllers is done in Firmware and U-Boot. This initialization (e.g. setting up clocks, timings etc. etc.) might even influence later behavior. So testing with Firmware/U-Boot out of the game is a reasonable way to definitly pinpoint the issue to the kernel.

> My system crashed again in three hours.

:(

> Should I go back to OS 5.4? Should I go back to SSD only boot?

I would stay with SD card boot and SSD data. You can move to different versions of the OS also with this combination. You can go to OS 5.4 for now, I probably have a development release 6 with the Linux 5.10 kernel later today, I'll write once that is ready.

> Also what would be helpful is if the scrolling log was re-enabled on the PI.

Not exactly sure what you mean, kernel errors should still appear on the console. The only thing we disabled are informational messages.

bschatzow, 2021-03-02:
> > Again a difference here is that the Raspberry software is 32 bit vs 64
> 
> Sorry I lost a bit track of what all you tested, we do have a 32-bit image, did you test 32-bit?

I was under the understanding that PI 4 booting from USB was only done using the 64 bit version. I tested the 64 bit as that is what I had used for USB3 boot.  I did not want to add another variable. 

> 
> > I have also asked repeatedly why booting from SD / SSD combo is more reliable and this was never answer.
> 
> There are countless reports where it just doesn't boot, see #1257. SD card approach sidesteps that.
> 
> That said, if you boot from SD, initialization of controllers is done in Firmware and U-Boot. This initialization (e.g. setting up clocks, timings etc. etc.) might even influence later behavior. So testing with Firmware/U-Boot out of the game is a reasonable way to definitely pinpoint the issue to the kernel.

I understand your answer, but I would think that after a couple of hours of operation nothing from the boot should be causing the system lockup?

> :(
>
>
> I would stay with SD card boot and SSD data. You can move to different versions of the OS also with this combination. You can go to OS 5.4 for now, I probably have a development release 6 with [...]
If this is coming today.  I'll hold off to test this.

> > Also what would be helpful is if the scrolling log was re-enabled on the PI.
> 
> Not exactly sure what you mean, kernel errors should still appear on the console. The only thing we disabled are informational messages.
The monitor attached console comes up with two lines:
Welcome to Home Assistant
and 
Homeassistant Login: with a blinking cursor.
In the past you had to press some key (don't remember) to get to the log in.  The default was a rolling log. Also the font size is greatly reduced from what it was.

agners, 2021-03-02:
> I was under the understanding that PI 4 booting from USB was only done using the 64 bit version. I tested the 64 bit as that is what I had used for USB3 boot. I did not want to add another variable.

Initially only 64-bit worked with USB boot yes. But today 32-bit does work as well. I mean if it helps, maybe worth a try. But I am also ok to stay with 64-bit testing.

> I understand your answer, but I would think that after a couple of hours of operation nothing from the boot should be causing the system lockup?

Yeah I guess that makes it less likely to be firmware releated :man_shrugging:..

> If this is coming today. I'll hold off to test this.

Probably, doing some testing right now.

bschatzow, 2021-03-02:
Let me know when it is out and I will try it and capture any information you feel is helpful.

bschatzow, 2021-03-02:
Do you know the settings to put the monitor back into the scrolling log?

agners, 2021-03-02:
> Do you know the settings to put the monitor back into the scrolling log? Something in the config.txt file?

I am not quite clear what you mean by that, but I guess you mean the kernel logs which have been displayed on the login screen? We just lowered the log level by default once booted up, so that informational messages no longer show up and disturb any local session. Any warnings or error messages will still appear (see https://github.com/home-assistant/operating-system/pull/1054).

You can reenable all logs by using:
```
login: root
...
ha > login
# echo 7 > /proc/sys/kernel/printk
```

bschatzow, 2021-03-02:
Do you need to does this each time?  If so, on a system restart I will miss this.

soundslikeluke, 2021-03-02:
I am having the exact same issue I guess I just need to what to see if updates will help. I thought it was an issue with my ssd but turns out I am not alone. I am running a Pi 3B+ on the latest version of everything 64 bit on a ssd. It is happening multiple times per day taking a long time to reboot

soundslikeluke, 2021-03-02:
Upgrading to the latest beta version also hasn't helped it is still restarting multiple times.

bschatzow, 2021-03-02:
@agners Latest lockup
[image]

bschatzow, 2021-03-03:
@agners,
I tried to downgrade the OS and during the downgrade the system crashed.  Not sure why?  I have done this lots of times with no issues before.  It seemed something got corrupted on the Sd.  No video out.  So much for this method being more reliable.  I never had an SSD fail in 6 months using them in Home Assistant.  Had several SD fail in the year before I switched.  When you release the new version of the OS I will test it 2 ways.  First on my existing SSD and then I'll recreate a split system (Sd boot / SSD).  I would hope that someone is working on a better restore process.  I see the current process taking almost an hour, if everything goes well.  The steps I see is create the SD, restore the snapshot to the SD, Install the SSH 22222 credentials to the SD, run disk partition utility.

agners, 2021-03-03:
> I tried to downgrade the OS and during the downgrade the system crashed. Not sure why? I have done this lots of times with no issues before. It seemed something got corrupted on the Sd.

:worried: sorry to hear that.

>  The steps I see is create the SD, restore the snapshot to the SD, Install the SSH 22222 credentials to the SD, run disk partition utility.

Via snapshot should be straight forward, but yeah its time consuming.

The last development build went though now and is ready for download at https://os-builds.home-assistant.io/6.0.dev20210302/.

bschatzow, 2021-03-03:
@agners 
Thanks.  It is straight forward.i was just frustrated yesterday. 
How do I install the developers os? Never did this before. Same cli command?

agners, 2021-03-03:
@bschatzow the screen capture shows that the kernel seems to be still alive and well, but for some reason a python3 task (e.g. could be HA core) did get stuck somewhere. I assume that its related to a I/O operation (e.g. SSD write). But it doesn't really show the root cause etc... :(

Communicating here is probably the best, then we have things tracked.

For the update, you need to switch the Supervisor to the developer channel to get to OS updates from the developer channel, e.g.

```
su options --channel dev
su reload
os update --version 6.0.dev20210302
su options --channel stable
```

bschatzow, 2021-03-03:
Thanks.  All is running so far.  Do you want me to switch the SU back to stable?  Right now all software is in development, OS, core, & SU.

bschatzow, 2021-03-03:
Crashed again.  Going back to 5.4.  Not sure what is next.  Also the screen showed no information.   It stayed on login

TuomasPakkanen, 2021-03-04:
Is this helpful?
https://unix.stackexchange.com/questions/252045/rcu-preempt-detected-stalls-on-cpus-tasks-message-appears-to-continue
https://stackoverflow.com/questions/35401317/rcu-preempt-self-detected-stall-on-cpu-0
https://www.kernel.org/doc/Documentation/RCU/stallwarn.txt

agners, 2021-03-04:
@TuomasPakkanen not particularly. Yes RCU stall can happen due to user space process running on RT, but that is not the case here. This is very likely triggered by some I/O issue. If OS 5.4 wouldn't work, I would assume broken hardware or power. But the the fact that OS 5.4 seems to work stable, indicates that there is a problem with newer kernel in combination with USB 3 etc.

@bschatzow I think I asked once, but I forgot where: Did you try OS release 5.5 once? Did that freeze as well?

TuomasPakkanen, 2021-03-04:
Should we try running the ssd via usb2 port? Mine's on USB3. Asked @bschatzow if he had tried this earlier.

bschatzow, 2021-03-04:
@agners> @bschatzow I think I asked once, but I forgot where: Did you try OS release 5.5 once? Did that freeze as well?
Yes it did.  @TuomasPakkanen made a very useful spreadsheet that is linked on 1119 that shows (I believe 32 users) with the different versions tried.  I believe you can use this to try and establish a pattern.  If more columns (or different columns) are needed just ask. I'm more than happy to try different things and convince others to also if you can give us directions of what to try.

agners, 2021-03-04:
@TuomasPakkanen if it works, why not :) It will be slower of course...

@bschatzow The problem of this issues are that they are highly hardware/environment dependent. We have all kinds of tool to analyze things. But those do not help when we don't see the problem here :( We would need to buy all the hardware you guys have to actually be able to reproduce, which is not feasible.

Is the problem also reproducible with a vanialla installation, from scratch (no add-ons, just Home Assistant default config?

What would be interesting is if the problems are also reproducible with Raspberry Pi OS. I guess just installing will not help, as this likely is some race condition etc, which only happens when there is actually things going on... So best would be to install a Supervised installation for testing.

bschatzow, 2021-03-04:
@agners 
I only have one PI4 that I use.  Others in many of the issues have confirmed the same hardware with no issues with the Raspberry desktop on top of the standard Debian image that Raspberry image installs.

If you think it is related to my Controller, I can purchase a different one.  If so, which one? I currently tried both the StarTec 3.0 and 3.1.  I have also tried the Samsung SSD EVO series as well as the Kingston.  Both have been connected to the USB3 next to the power connector.  

Should I move it to the other USB3 port?  

I had no difference in reliably using the Sd boot.

I am using a standard HA PI4 installation that I have used since the beginning.  The last test was done with the latest available download on the Sd and restored my snapshot that works great with 5.4.  I never saw lockups in any versions before (pi3 & Pi 4) of course I had other issues.  Some operator error and some HA.  Never had an issue like this.

Is this the Supervised installation that you mention above that you want me to use?

agners, 2021-03-04:
> Others in many of the issues have confirmed the same hardware with no issues with the Raspberry desktop on top of the standard Debian image that Raspberry image installs.

That does not make much sense: We established that it is kernel related. We use the *very same* kernel as the Raspberry Pi OS uses. How can it be that it is kernel related yet Raspberry Pi OS doesn't show it? I think you are mixing up things. "Others" likely have different issues. It is certainly good to compare cases and try to see patterns, but be careful to not make conclusions of your case by mixing in reports from others..

I created a special build now: It is based on the current release 6 dev branch, but uses the Raspberry Pi Linux 4.19 kernel as has been used in OS release 5.4. With that, we can definitly pinpoint or rule out kernel issues:
https://os-builds.home-assistant.io/hassos_rpi4-64-6.0.dev20210304.raucb
https://os-builds.home-assistant.io/hassos_rpi4-64-6.0.dev20210304.img.xz

I think the raucb update method via USB flash drive with the name `CONFIG` should work (see https://github.com/home-assistant/operating-system/blob/dev/Documentation/configuration.md#automatic). Your system needs to be up at least 5 minutes, otherwise it won't start the update. Otherwise a from scratch installation using the `img.xz` file and restoring is probably the only way.

bschatzow, 2021-03-04:
@agners Thanks.  This should be a great test!
I created a special build now: It is based on the current release 6 dev branch, but uses the Raspberry Pi Linux 4.19 kernel as has been used in OS release 5.4. With that, we can definitly pinpoint or rule out kernel issues:
https://os-builds.home-assistant.io/hassos_rpi4-64-6.0.dev20210304.raucb
https://os-builds.home-assistant.io/hassos_rpi4-64-6.0.dev20210304.img.xz
Do I update via CLI or is there another way.

bschatzow, 2021-03-04:
I did the systemctl restart hassos-config after I copied the file over and it did not work.  Left it unable to boot.

bschatzow, 2021-03-04:
Figured out how to use the img file.  Need to reimage the drive and then restore snapshot.  Doing that now.

agners, 2021-03-04:
Did work on my end, but only tested using SD card:

```
# uname -a
Linux homeassistant 4.19.127-v8 #1 SMP PREEMPT Thu Mar 4 16:09:35 UTC 2021 aarch64 HassOS
# cat /etc/os-release 
NAME="Home Assistant OS"
VERSION="6.0.dev20210304 (RaspberryPi 4 64bit)"
```

Not sure why that would not boot from SSD, maybe some strange "new firmware/old kernel" interaction :man_shrugging:

bschatzow, 2021-03-04:
I can put it back in and take a picture when it stops if that helps?

agners, 2021-03-05:
Well, maybe, but I don't want to spend time to fix a test build which we only ever will use for debugging. It was a shot in the dark, but it seems that other things are broken with that old 4.19 kernel :man_shrugging:

bschatzow, 2021-03-05:
I'll send you a picture later today. Maybe it will tell you something.

bschatzow, 2021-03-05:
A couple of things I can try if you think it can help.  I can try the SD and data on the an SSD.  Also I noticed that the config. txt file was changed.  The dtparam=sd_poll_once=on was removed.  I can edit this and try it.  What do you think?

TuomasPakkanen, 2021-03-07:
> @TuomasPakkanen if it works, why not :) It will be slower of course...
> 

I have now updated to 5.12 and plugged the SSD to an USB2 port. I think the performance difference probably doesn't matter in my use case. I also did a database wipe for other reasons and updated the RPI4 EEPROM to latest stable version (16th Feb.). Otherwise running the latest core version and have updated add-ons and such to latest versions.

Let's see how it goes. I'll try to fetch the logs if it crashes again. I guess via monitor you can't see any info but I did setup the SSH access to 22222.

bschatzow, 2021-03-08:
@agners agners> You can reenable all logs by using:
> 
> ```
> login: root
> ...
> ha > login
> # echo 7 > /proc/sys/kernel/printk
> ```
This did not put the logs back into scrolling for me.  I want to test a couple of other options, but I want to see the scrolling screen incase there is another lockup.  Maybe the screen will have additional information that the logs do not have?

TuomasPakkanen, 2021-03-08:
> echo 7 > /proc/sys/kernel/printk

This command worked for me when I ran it via connecting to port 22222 via SSH. Can see the logs via monitor now.

muzzak123, 2021-03-10:
Hi guys, I was just reading all of the above after coming here from https://github.com/home-assistant/core/issues/43155
I'm curious as to why you suspect it's USB3 related ? Or is it just the boot problem you're talking about ?
I had my RPI4 (4G) running on an SD  with nothing plugged into any USB port, when I first experienced this problem. 
In fact one of the things I tried to resolve it was move to SSD to try to see if that fixed it, but it didnt. So far the only "fix" that works is back to 5.4. I've been running 5.4 with Frigate NVR monitoring 3 cameras for objects and it still hasnt fallen over. CPU gets up above 80% and automations start to run a lot slower, like lights taking seconds to turn on. But it keeps running.

TuomasPakkanen, 2021-03-10:
> Hi guys, I was just reading all of the above after coming here from [home-assistant/core#43155](https://github.com/home-assistant/core/issues/43155)
> I'm curious as to why you suspect it's USB3 related ? Or is it just the boot problem you're talking about ?
> I had my RPI4 (4G) running on an SD with nothing plugged into any USB port, when I first experienced this problem.
> In fact one of the things I tried to resolve it was move to SSD to try to see if that fixed it, but it didnt. So far the only "fix" that works is back to 5.4. I've been running 5.4 with Frigate NVR monitoring 3 cameras for objects and it still hasnt fallen over. CPU gets up above 80% and automations start to run a lot slower, like lights taking seconds to turn on. But it keeps running.

I have now been running 5.12 via USB2 for a couple of days without a crash where as previously I have been getting crashes on OS versions above 5.4 when running the SSD via USB3. It is not a boot issue for me, the system boots completely fine on both USB versions. 

I will try next weekend changing the SSD to USB3 to make sure something else has not fixed the issue for me.

bschatzow, 2021-03-10:
@muzzak123 > Hi guys, I was just reading all of the above after coming here from [home-assistant/core#43155](https://github.com/home-assistant/core/issues/43155)
> I'm curious as to why you suspect it's USB3 related ? Or is it just the boot problem you're talking about ?
I believe people are making educated guesses on what the issue is. No one has been about to capture any useful logs to help the developers determine the root cause. People are using trial and error to try and fix the issue. I tried the SD / ssd method and this failed for me.  @TuomasPakkanen  is trying usb 2 and he has had better success with this than usb3. Checkout the 1119 issue. There is a spreadsheet with over 30 users' data and I believe all but one had usb3 configuration.

muzzak123, 2021-03-10:
Thanks @bschatzow , 
It may be worth you browsing my link. Myself and others have tried a number of things as well. Also mentioned in that thread is the spreadsheet which I'm already in. I don't recall it containing mentioning specific info about USB or version ?

bschatzow, 2021-03-10:
@muzzak123 > It may be worth you browsing my link. Myself and others have tried a number of things as well. Also mentioned in that thread is the spreadsheet which I'm already in. I don't recall it mentioning specific info about USB or version ?
You are correct about the USB3 not in a separate column.  I believe at the time the sheet started that it was (wrongly, I see now) an assumption that if you were using the PI4 on SSD it was on USB3.  Still not understanding why so few are having this issue?  If it was on SD like you are having and I had on SD / SSD this issue should be flooded with complaints.
@TuomasPakkanen Should we had an USB column or cover it with hardware?

bschatzow, 2021-03-10:
@muzzak123 are you on 32 bit or 64?

bschatzow, 2021-03-10:
@agners <Initially only 64-bit worked with USB boot yes. But today 32-bit does work as well. I mean if it helps, maybe worth a try. But I am also ok to stay with 64-bit testing.
I just tried the 32 bit version on my SSD and it does not boot off of USB3.  Nothing shows on the monitor.  I can try the 32 bit on the SD card and try everything else on the SSD if you think this will help figure out what the issue is?  Last time I tried the 64 bit and it failed in a few hours.  Let me know what you think.

muzzak123, 2021-03-11:
@bschatzow - I think you are presumptuous in you're assumption . 
> By the way, I did read your thread and it seems like the people there are not putting in the time to figure out how to get the logs from the previous boot. Without these I don't see how any developer can help?

A lot of people including myself put a lot of time and effort into capturing what logs we could but the logs we did capture held no useful information. I summarised the situation at the end of the link and @agners agreed with me. Doesn't look like this thread is going anywhere so I'm unsubscribing.

TuomasPakkanen, 2021-03-11:
OK so USB2 did not fix the issue. Here are the logs just before the crash:

`command: journalctl --since "2021-03-11 09:55:00" --until "2021-03-11 10:10" >/mnt/data/supervisor/homeassistant/jour8.txt 2>&1
crash at 10:00 sharp UTC time (based on ping based healtch check done with 5min intervals)

-- Logs begin at Thu 2021-01-21 04:47:04 UTC, end at Thu 2021-03-11 10:35:37 UTC. --
Mar 11 09:55:00 homeassistant aa84cd1e1859[412]: 2021-03-11 11:55:00 INFO (MainThread) [homeassistant.components.automation.health_check] Health check: Running automation actions
Mar 11 09:55:00 homeassistant aa84cd1e1859[412]: 2021-03-11 11:55:00 INFO (MainThread) [homeassistant.components.automation.health_check] Health check: Executing step call service
Mar 11 09:55:02 homeassistant systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:255 ZCL attribute report 0x588E81FFFE156034 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:265 ZCL attribute report 0xCCCCCCFFFEA901FE for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:294 ZCL attribute report 0xCCCCCCFFFEA9E25B for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:309 ZCL attribute report 0x588E81FFFE266EA9 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:315 ZCL attribute report 0xCCCCCCFFFEC36AC4 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:406 ZCL attribute report 0xCCCCCCFFFEA9E25B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:512 ZCL attribute report 0x588E81FFFE266EA9 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:520 ZCL attribute report 0xCCCCCCFFFEA901FE for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:04 homeassistant 5c7c92aaa3a5[412]: 11:55:04:527 ZCL attribute report 0x588E81FFFE0EA096 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:07 homeassistant 5c7c92aaa3a5[412]: 11:55:07:173 [INFO] - No button map for: SML001, unicast to: 0x0000, endpoint: 0x02, cluster: 0x0402, command: 0x0A, payload: 0000297508, zclSeq: 130
Mar 11 09:55:07 homeassistant 5c7c92aaa3a5[412]: 11:55:07:175 ZCL attribute report 0x0017880106F64B5D for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:21 homeassistant 5c7c92aaa3a5[412]: 11:55:21:797 ZCL attribute report 0xEC1BBDFFFEA8CC50 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:136 binding for cluster 0x0000 of 0xCCCCCCFFFEA9E25B exists (verified by reporting)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:139 binding for cluster 0x0006 of 0xCCCCCCFFFEA9E25B exists (verified by reporting)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:141 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:143 binding for cluster 0x0008 of 0xCCCCCCFFFEA9E25B exists (verified by reporting)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:145 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:147 binding for cluster 0x0300 of 0xCCCCCCFFFEA9E25B exists (verified by reporting)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:150 skip configure report for cluster: 0x0300 attr: 0x0007 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:151 skip configure report for cluster: 0x0300 attr: 0x0003 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:152 skip configure report for cluster: 0x0300 attr: 0x0004 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:153 skip configure report for cluster: 0x0300 attr: 0x0008 of node 0xCCCCCCFFFEA9E25B (seems to be active)
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:675 verified group capacity: 255 and group count: 3 of LightNode 0xccccccfffea9e25b
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:676 0xccccccfffea9e25b found group 0xFFF0
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:678 0xccccccfffea9e25b found group 0x000C
Mar 11 09:55:31 homeassistant 5c7c92aaa3a5[412]: 11:55:31:679 0xccccccfffea9e25b found group 0x000D
Mar 11 09:55:40 homeassistant 5c7c92aaa3a5[412]: 11:55:40:160 Current channel 15
Mar 11 09:55:40 homeassistant 5c7c92aaa3a5[412]: 11:55:40:167 CTRL got nwk update id 0
Mar 11 09:55:40 homeassistant 5c7c92aaa3a5[412]: 11:55:40:173 Device TTL 4015 s flags: 0x7
Mar 11 09:55:40 homeassistant 5c7c92aaa3a5[412]: 11:55:40:319 ZCL attribute report 0xEC1BBDFFFEA8CC50 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
[... cut ...]
Mar 11 10:00:25 homeassistant 5c7c92aaa3a5[412]: 12:00:25:316 ZCL attribute report 0xEC1BBDFFFE469636 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
Mar 11 10:00:25 homeassistant 5c7c92aaa3a5[412]: 12:00:25:338 ZCL attribute report 0xEC1BBDFFFED441B9 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
`
I see no further messages when viewed via plugged in monitor, but a temperature icon started flashing there (due to pi4 getting hot after the crash and staying hot until power plug is pulled:
[image]

What's next? @agners  @pvizeli

bschatzow, 2021-03-11:
@muzzak123 
>@bschatzow - I think you are presumptuous in you're assumption .
I am not.  I think you are taking my questions wrong.  I am not assuming anything.  I am like many others trying to piece together what the issue is what what the solution is.  
Currently many people's RPI 4 work with both USB3 SSD booting and SD booting.  But many like you and I do not.  I am trying to understand the differences.  People are doing trial and error (mostly errors) to try and come up with a solution that works.  So far I have tried 
SSD all versions since 5.4.
SD / SSD split on 5.12 - Crashed it 23 hours
SSD on 32 bit (no boot)
6.x developers version on SSD (no boot)
I was asking why you were on 64 bit for SD as this was not needed.  My system ran with no issues on 32 bit on 5.8 (last I tried).

tobias-kuendig, 2021-03-11:
I do have the same problem. Running on a Raspberry Pi 4, 4 GB Model. Booting from a SSD via USB3, but I had the same problem with an SD card as well. Everything is up-to-date.

```
core-2021.3.3
supervisor-2021.03.4
Home Assistant OS 5.12
```

I can provide logs for a crash I had just a few hours ago. I cannot confirm that the device was still running as I don't have a micro HDMI cable at hand right now, so I cannot connect a display. The device is not reachable via the network once the system freezes. My router does not show it as connected even though it is connected using a ethernet cable.

Crash at 10:34, power cycle around 12:13:

```
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-deconz with version 6.7.2
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-aarch64 with version 0.103.1
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-aarch64 with version 0.103.1
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.2.0
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.2.0
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-check_config with version 3.6.0
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-check_config with version 3.6.0
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.0.2
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.0.2
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.snapshots] Found 4 snapshot files
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.snapshots] Found 4 snapshot files
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.discovery] Loaded 1 messages
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.ingress] Loaded 0 ingress sessions
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.check] System checks complete
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.check] System checks complete
Mar 11 10:19:08 homeassistant 473d4a70daba[415]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
Mar 11 10:19:08 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:08 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
Mar 11 10:19:09 homeassistant 473d4a70daba[415]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Mar 11 10:19:09 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
Mar 11 10:19:09 homeassistant 473d4a70daba[415]: 21-03-11 10:19:09 INFO (MainThread) [__main__] Running Supervisor
Mar 11 10:19:09 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:09 INFO (MainThread) [__main__] Running Supervisor
Mar 11 10:19:09 homeassistant rauc[316]: rauc mark: marked slot kernel.0 as good
Mar 11 10:19:09 homeassistant 473d4a70daba[415]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
Mar 11 10:19:09 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.hassos] Rauc: A - marked slot kernel.0 as good
Mar 11 10:19:09 homeassistant hassos-supervisor[1114]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
Mar 11 10:19:09 homeassistant 473d4a70daba[415]: 21-03-11 10:19:09 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
[... cut ...]
Mar 11 12:15:44 homeassistant hassos-supervisor[1114]: 21-03-11 12:15:44 INFO (MainThread) [supervisor.host] Host information reload completed
Mar 11 12:15:44 homeassistant 473d4a70daba[415]: 21-03-11 12:15:44 INFO (MainThread) [supervisor.host] Host information reload completed
```

During the outtage no new data points were recorded:

[image]

[image]

CPU and memory usage are absolutely normal before and after the crash. There were no spikes or memory issues.

[image]

[image]

I cannot confirm that the device got unusually hot during downtime.

bschatzow, 2021-03-14:
@agner,
I updated the firmware on the StarTech controller this morning, rebooted then updated to OS 5.12.  Froze 4 hours later.  Is there anything I can do to help troubleshoot this issue?

bschatzow, 2021-03-15:
@agners,
Is there a way to get the 32 bit version to boot my SSD?  You said above that it now supports SSD booting.  I have tried several versions and I can't get it to boot.  
Since I have not been able to boot the SSD with a 32 bit version, do you think the next test I should try is going back to 32 bit SD ( which worked up to 5.8 (last version I tried)) and do another spit test?

agners, 2021-03-17:
> @agners,
> Is there a way to get the 32 bit version to boot my SSD? You said above that it now supports SSD booting. I have tried several versions and I can't get it to boot.

Last time I checked 32-bit worked on the devices I test with usually. With OS release 6 we get a new U-Boot, and there have been fixes which might improve the situation. Just tested a local build with U-Boot 2021.04-rc4 now with my Samsung T1 USB 3.0 SSD, and 32-bit seems to work well. I'll let you know once I have some builds online.

> @agner,
> I updated the firmware on the StarTech controller this morning, rebooted then updated to OS 5.12. Froze 4 hours later. Is there anything I can do to help troubleshoot this issue?

Just Googled quickly, have you noticed this thread? https://www.raspberrypi.org/forums/viewtopic.php?t=297562

To me it seems that this adapter is unstable in general, so this seems not really be HAOS related.

bschatzow, 2021-03-17:
> Last time I checked 32-bit worked on the devices I test with usually. With OS release 6 we get a new U-Boot, and there have >been fixes which might improve the situation. Just tested a local build [...]
I will test this as soon as you tell me it is available. I believe the memory handling or timing of the 64 bit vs 32bit maybe the issue.  If you look at the people where they are having no issues, most are using the 8GB RPI4 which is at PCB 1.4 vs 1.2 on the 4GB.  I have been in contract with some with the identical hardware as me (same StarTech controller) and they have no issue.  I have left messages in the Raspberry forums to try and get clarification.  Will pass on any useful information I receive back to you.
I did test the 64 bit SD /SSD setup and this failed as well.

>Just Googled quickly, have you noticed this thread? https://www.raspberrypi.org/forums/viewtopic.php?t=297562

I saw this previously, and I have updated to the new firmware and it made no difference.  I also have the older 3.0 adapter and it also fails the same way and supposedly there is no issue.  If you feel it is the controller, what controller should I use?

>To me it seems that this adapter is unstable in general, so this seems not really be HAOS related.
To me if it is not HAOS related why does it only fail on OS above 5.4?

tobias-kuendig, 2021-03-21:
I habe disabled the Home Assistant Google Drive Backup addon and haven't had a crash in over a week. This is definitely a new record.

Not sure how this would be connected but maybe someone else can give it a try.

bschatzow, 2021-03-21:
@tobias-kuendig,
I would think that this would be a coincidence. I also have this add-on that runs daily at 3AM. I don't see how it could crash the system when not running? I will test it Monday or Tuesday and get back to you.

bschatzow, 2021-03-22:
@agners @pvizeli 
I am trying to understand why most people using the 8G PI4 boards with the same hardware I have are not seeing any issues.  I have asked in the raspberry forums 
```https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=307132&e=1&view=unread#unread```
And one of the responses seems to make sense 
"even if your only using 30% of the 4gig, linux may randomly use all of the 8gig, in any order, and may run into compatibility problems with 64bit addressing"
Since I can't get the 32 bit version to boot the SSD Is there anyway to test this idea?

bschatzow, 2021-03-25:
@agners 
I have not heard back from you on the 32bit OS that can boot off of the SSD.  I tried the spit SD / SSD with the 64 bit OS and it failed.  I am going to try today the 32bit split SD / SSD and let you know if this works.

tobias-kuendig, 2021-03-31:
Just to report back on my `Home Assistant Google Drive Backup` tests: I did disable the plugin and had no issues for over two weeks. I did re-enable the addon last Monday and had my first crash a little more than 30 hours later.

> I do use google drive add on and it is set to run daily at 3AM. I don't see how it could crash the system when not running? 

I run the same config, and the crash does not happen at the time of the backup or after a specific interval. It seems to happen randomly and it does not happen every time a backup is made.

But I wonder if the backup job might overwhelm the Pi in some way, which causes this issue later on?
After all, my snapshots are around 800 MB in size with a 4 GB `home-assistant.db` to compress. But it's true, I don't have any idea what the exact issue would be, especially as it manifests at random times, so the Backup plugin as a culprint sounds wrong.

bschatzow, 2021-03-31:
I looked at the 1119 issue again with the spreadsheet mentioned 
``` https://docs.google.com/spreadsheets/d/1iHTVvaNlTUqwFUgsUhUNws2Sw115INIx5ChEgTnIfoc/edit#gid=0 ```
And many do not use the google drive add on and have the same issue.  It would be interesting to see why it works for you?  Maybe some feedback from @agners can give us some understanding on this?

lwolfs, 2021-03-31:
Since the upgrade to core-2021.3.4 I never experienced a hung state again :-) 
Before that daily hung states.

bschatzow, 2021-03-31:
@lwolfs Had no effect on me.

agners, 2021-04-01:
@bschatzow 

> I have not heard back from you on the 32bit OS that can boot off of the SSD. I tried the spit SD / SSD with the 64 bit OS and it failed. I am going to try today the 32bit split SD / SSD and let you know if this works. Can't do this as import with USB is broke so I can't run SSH 22222.

I had it booting off a USB 3.0 flash stick (I think I used a Kingston model I have here on my table) as well as my Samsung T1 250GB SSD. Both worked fine with 32-bit OS using USB 3.0. But as we know, this means nothing in the world of USB boot :cry:

[73 more replies]

closed 2022-01-03 by agners