home-assistant/operating-system #1257: HASSio not booting from NVMe Drives

ur7x, 2021-03-02:
**Hardware Environment**

- [X ] Raspberry Pi [4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

agners, 2021-03-02:
I assume your NVMe is connected via USB 3.0?

Unfortunately not all USB 3 to NVMe adapters seem to work reliable. This is a Raspberry Pi and/or U-Boot issue. We cannot support them all, see also our FAQ: https://www.home-assistant.io/faq/#is-usb-boot-for-the-raspberry-pi-4-supported

What I suggest to do is use the USB data disk feature: That way you get reliably boot the Operating System from SD card, but Home Assistant Core etc. will still run off the speedy USB SSD: https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk

ur7x, 2021-03-02:
Yes using USB3.0  port... I have tried a few different USB-> NVMe adapters and ironically ALL of them will boot when flashed with a base Raspberian OS, but as soon as I flash HASSio it fails to boot.  Also ironic the  HASSio will boot to almost every SATA SSD.  including the M.2 versions of the SATA SSD, just not the NVMe.

cogneato, 2021-03-02:
@ur7x I had a setup with NMMe which would boot, but only from a pi4's USB 2 port. Is that something you tried?

ur7x, 2021-03-05:
Yup tried the USB 2 port and you are right.  It boots... once... weird.  As soon as it boots  HASSio does a bunch of updates to the drive and then... it is a brick... no second boot.

ur7x, 2021-03-06:
Yes I determined the issue is with the NVMe Drive to USB adapter.  These use a JMS583 chip that comes with Various levels of firmware.  HASSio will only work with adapters running firmware version 2.00.09 or better. Once I flashed my adapter(s) with the newer firmware they were happy to boot with HASSio.

kdelios, 2021-03-12:
> @ur7x that is an interesting find, how did you flash the adapter firmware?

I used "JMMassProd_v1_16_15_12.exe" along with the "v00.02.00.09.bin" (No MVMe drive ssd during flashing).
Boot with HASSos is now possible.

agners, 2021-03-16:
@kdelios this is with the drive attached directly to USB 3.0?

Just upgraded my JMS583 based Icy Box (see #796) to 00.02.00.09 (without NVMe drive during flashing). Things seem to work well on my Desktop PC. But for me it seems the RPi 4 (4GB) cannot boot from it at all. It looks like the firmware has issues again (besides Feb 16 2021 I also tried with the Sep 03 EEPROM firmware release). Maybe something else is broken with this Icy Box enclosure.

[image]

This shows that the whole USB boot is a huge hit and miss :tired_face: 

Please use the `datactl` feature as much as possible https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk.

closed 2021-03-16 by agners

kdelios, 2021-03-16:
> @kdelios this is with the drive attached directly to USB 3.0?
> 
> Just upgraded my JMS583 based Icy Box (see #796) to 00.02.00.09 (without NVMe drive during flashing). Things seem to work well on my Desktop PC. But for me it seems the RPi 4 (4GB) cannot boot from it at all. It looks like the firmware has issues again (besides Feb 16 2021 I also tried with the Sep 03 EEPROM firmware release). Maybe something else is broken with this Icy Box enclosure.
> 
> [image]
> 
> This shows that the whole USB boot is a huge hit and miss 😫
> 
> Please use the `datactl` feature as much as possible https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk.

Yes is attached to USB3 with TRIM enabled.
Tested at my Pi4 2GB, and also at my "production" Pi4 8GB. In both worked smoothly. In my "test" setup I've tried many scenarios (soft & hard restarts etc), no problems., every singe time booted. My eeprom version was  2021-01-11. I didn't tried with newer.
Tested also with the datactl feature for a couple of months, no problems also.

[image]

agners, 2021-03-17:
@kdelios which enclosure do you use exactly? Looks like there is variance even with the same chipset.

Interestingly enough, today after power cycling the RPi 4 it booted once just fine. But even after toggling a couple of times I have not been able to get it booting a second time (so it seems there is some type of race condition).

The one time it booted things failed after a short period though:
```
# [  387.040051] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[  387.047959] xhci_hcd 0000:01:00.0: HC died; cleaning up
```

But I assume this is since I did not disable UAS using `usb-storage.quirks=152d:0583:u`. Is yours running with or without such usb-storage quirk?

kdelios, 2021-03-17:
> @kdelios which enclosure do you use exactly? Looks like there is variance even with the same chipset.
>
> Interestingly enough, today after power cycling the RPi 4 it booted once just fine. But even [...]

I am using the following enclosure:

https://www.amazon.com/Protection-Enclosure-Innovative-Connecting-Thunderbolt/dp/B07Q7H5QLX

with the following SSD:

Is running without usb-storage quirk setting.