home-assistant/operating-system #1259: Cannot mount /dev/sda1 after update to 5.12

quarcko, 2021-03-02:
**Hardware Environment**

- [ x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ x] Updated from version 5.11
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I use my own custom modified addon version of MariaDB
(please keep in mind this is not a problem with addon itself)

So i run it with protection mode disabled, and i mount external USB drive
to offload files to it, so microSD card is not wearing much.

Maybe it's coincidence, but everything was working fine on 5.11
and when i updated to 5.12, suddenly i can't mount /dev/sda1 anymore!

i tried rolling back to 5.11, but no luck, some kind of "damage" already done..
I tried logging in with portainer, and trying to do:

bash-5.0# mount /dev/sda1 /share/usb
mount: permission denied (are you root?)
bash-5.0# whoami
root

What has changed? How can i again mount my USB drive and use it inside my addon?
Also - USB drive is healthy, i tried it on my PC and it mounts RW okay.
file system check also did not find any errors on it.
File system on it is EXT4 if this matters.

If this does not belong here, please advise where should i request help
as there are so many layers in this project, but i write here because
the only thing that i updated was OS.

agners, 2021-03-03:
I don't think this is OS related, there has been no change in this domain. That is also why rolling back to 5.11 did not fix the issue. This is likely cause by a Supervisor update.

quarcko, 2021-03-04:
OK, i filled in a report for Supervisor, will close this issue when someone there confirms that it is actually caused by Supervisor.

agners, 2021-03-05:
Supervisor issue: https://github.com/home-assistant/supervisor/issues/2677

quarcko, 2021-03-17:
Yeah, sorted it out. Apparently it was supervisor thing. 
Sorry for bugging and closing.

closed 2021-03-17 by quarcko