home-assistant/operating-system #1262: OS ignores some custom settings in NetworkManager.conf

tdejneka, 2021-03-06:
**Hardware Environment**

- [x] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Intel NUC (or compatible)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.2.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.83-v7
arch | armv7l
timezone | America/NewYork

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.03.4
docker_version | 19.03.13
disk_total | 13.8 GB
disk_used | 3.7 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (8.0.3), Portainer (1.4.0), File editor (5.2.0), Samba share (9.3.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 2
views | 6
mode | storage

</details>

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I followed the [instructions here](https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md) for modifying **NetworkManager.conf**. I added a static IP address (and DNS and Gateway addresses) and changed the polling interval for Connectivity.

<details>
<summary>Click to show NetworkManager.conf</summary>

```text
[main]
dns=default
plugins=keyfile
autoconnect-retries-default=0
rc-manager=file

[keyfile]
unmanaged-devices=type:bridge;type:tun;type:veth

[logging]
backend=journal

[connection]
connection.mdns=2
connection.llmnr=2

[connectivity]
uri=https://version.home-assistant.io/online.txt
interval=3600

[device]
wifi.scan-rand-mac-address=no

[connection]
id=hassos-network
uuid=<redacted>
type=802-3-ethernet

[ipv4]
method=manual
address=192.168.1.50/24;192.168.1.1
dns=192.168.1.2

[ipv6]
addr-gen-mode=stable-privacy
method=auto
```
</details>

The OS _accepts_ the static addresses but _ignores_ the `interval` option (3600 seconds) for `connectivity`. It continues to use the default value of 300 seconds (5 minutes).

I have tried several variations:
- Set `interval=0` to disable connectivity check.
- Set `uri` to an internal address without an `interval` option.
- Remove the `connectivity` section entirely.

However, all variations have been ignored; it continues to poll at 5-minute intervals.

Is it not possible to instruct the OS to alter the behavior of its `connectivity` check in **NetworkManager.conf**?

tdejneka, 2021-03-13:
Is there any additional information required to help someone reply to this Issue?

agners, 2021-03-16:
> I followed the instructions here for modifying NetworkManager.conf. 

This instructions show how to add/change the system-connections keyfiles. So this is not the main NetworkManager.conf file you are adjusting, but connection settings. From what I can tell, there is no `connectivity` section supported for those files. Currently its not possible to adjust the main `NetworkManager.conf` file.

agners, 2021-03-17:
Closing as adjusting `NetworkManager.conf` is currently not an implemented feature.

closed 2021-03-17 by agners

tdejneka, 2021-03-17:
Thank you for taking the time to investigate and reply.

The original documentation led one to believe that `NetworkManager.conf` can be modified (it even linked to the documentation for NetworkManager). However, thank you for clarifying that it does not, in fact, support all functionality. Additional thanks for revising the documentation, notably the addition of this line:

> Note that changes to `NetworkManager.conf` are not supported currently, only connection keyfiles are supported.

---

My original goal was to reduce the polling frequency of the `connectivity` check (currently it is the default value of every 5 minutes). That's far too frequent for my purposes plus there is yet another connectivity check performed by Supervisor (every 10 minutes).

Given that there appears to be no control over the OS's connectivity check, I switched from Home Assistant OS to Home Assistant Supervised on Debian (where `NetworkManager.conf` is accessible and the connectivity check is disabled by default).

agners, 2021-03-17:
Having a custom setting to adjust connectivity interval sounds a somewhat special use case, but it could be implemented if this is a common feature request.

> Given that there appears to be no control over the OS's connectivity check, I switched from Home Assistant OS to Home Assistant Supervised on Debian (where NetworkManager.conf is accessible and the connectivity check is disabled by default).

Using Supervised on Debian is a good way to get more control over OS behavior. So I think going this direction is the right approach here.

Thanks for understanding!