home-assistant/operating-system #1264: Bluetooth issue after 5.12 upgrade

Toukite, 2021-03-09:
**Hardware Environment**
- Raspberry Pi 4b

**Home Assistant OS release:**
- Updated from version 5.11
- Additional information (if accessible):
## System Health

version | core-2021.2.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.83-v8
arch | aarch64
timezone | Europe/Paris

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4982
Installed Version | 1.11.3
Stage | running
Available Repositories | 751
Installed Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.03.4
docker_version | 19.03.13
disk_total | 219.4 GB
disk_used | 11.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.2.0), Terminal & SSH (9.0.2), Samba share (9.3.1), Mosquitto broker (5.1.1), Node-RED (8.1.3), InfluxDB (4.0.3), phpMyAdmin (0.2.0), Grafana (6.1.3), Zigbee2mqtt (1.18.1-1), MariaDB (2.2.2), Z-Wave JS (0.1.10), Z-Wave JS to MQTT (0.6.1), Glances (0.11.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 11
views | 11
mode | storage

</details>

**Description of problem:**
Opening an issue here as it was suggested to me on this issue: https://github.com/custom-components/ble_monitor/issues/295
After upgrade to 5.12, bluetooth works for some time then suddenly crash after a few hours and requires a host reboot to work again for a few hours before crashing again.
Rollback to 5.11 works (managed to have a whole week without issues after rollback)
And before 5.12 upgrade, everything had been working for months
When crashing, I have following error message in the logs:
`**ERROR (Thread-3) [custom_components.ble_monitor] HCIdump thread: Runtime error while stop scan request on hci0: Event loop stopped before Future completed.**`

Additional details can be found in the issue mentioned above, but don't hesitate if I can provide any more details !
Thanks for the help

agners, 2021-03-11:
Hm, I guess this is related to the latest Bluetooth firmware update #1250. I wonder if this is really a issue with the new driver or just some different timing which causes issues in that custom component?

Toukite, 2021-03-12:
Anything I can do to help answer this question ?

Ernst79, 2021-03-12:
I'm one of the developers of the custom component, and I was running out of idea's. As @Toukite told me that it was running fine on 5.11, and not on 5.12, so I suspected it had something to do with the 5.12 update. This solution helped for some people, but that was specific for venv installations, after they upgraded python, to point setcap to the right python version after python update. https://github.com/custom-components/ble_monitor/issues/167#issuecomment-739541674 But I don't think python got updated in 5.12. 

I also have to mention that it seems to affect only a very limited number of people (only 2 at the moment), so it doesn't seem to be an general issue for everyone. It is running fine on my end for example (Raspberry Pi 3b).

agners, 2021-03-12:
There was no Python upgrade in 5.12.

To me it looks like some race condition in that custom component. `Event loop stopped`, why did the event loop stop? I mean other than on shutdown or similar it should continue to run, I would expect? Is there another error which teared down the event loop before that?

Ernst79, 2021-03-12:
Ok, I think we can close the issue here. 

The custom component uses a slightly modified version of aioblescan (Python 3/asyncio library to listen for BLE advertized packets) to listen to the Bluetooth packets. Is it failing somewhere in the aioblescan, perhaps the developer of this library can help us. 

@Toukite Let's discuss further in https://github.com/custom-components/ble_monitor/issues/295 on how to proceed

pvojnisek, 2021-03-13:
@Ernst79 I am affected also with it. It is clear that the 5.12 upgrade brought this bug, but..
I have a 1 month old clean installation of HA supervised on a clean install debian. Everything was working perfectly before. (my heating system is controlled by HA and I was lucky that I was able to start the old rapberry with domoticz which is stable like a stone)

Ernst79, 2021-03-13:
@agners When installing 5.11, is the firmware update reverted as well, or does it keep the firmware version?

Wampukas, 2021-07-08:
I can confirm this issue also. I am running the newest ova version on VMware and this happens as well while in 5.11 it was very stable.

Toukite, 2021-07-12:
Does anybody know what kind of issues we might get into by not upgrading post 5.11 due to this ?

We are already at OS 6.1 / core-2021.7.1, will there be a point where core version won't be compatible anymore with an older OS release (5.11) and we will be forced to upgrade OS ?

agners, 2021-07-29:
To me it seems that this issue is a collection of different issues with the same/similar error.

Those using a Raspberry Pi, HAOS 6.2 comes with new Bluetooth firmware for Raspberry Pi's. Please test with 6.2 again.

Draghmar, 2021-08-15:
I have HAOS 6.2 and I recently discovered that I had (and still have) this error. Actually anything BT related simply doesn't work.

mcc05, 2021-08-19:
I'm using HAOS 6.2 to read an SMA inverter using bluetooth, Pi 4 I've noticed there appears to be a bug. If I reboot the host on the first communication to the inverter everything is ok, but on subsequent requests it fails randomly. It like resources are not (disconnect) being freed. Is there an easy way to roll back to older OS, I'm sure it  was rock solid before?

Ernst79, 2021-10-30:
For the users of BLE monitor we have now implemented a workaround, that powers on Bluetooth and unblocks Bluetooth. (`bluetoothctl power on` and `rfkill unblock bluetooth`) in case of this error. It seems that for some reason, the power state of the Bluetooth adapter is turned off, so the first command is tuning it back on. This workaround works if the error occurs occasionally, but one of the users has reported that in his case, it is powered off so often that this workaround is still unusable. 

This “fix” is of course a workaround, it is not solving the actual problem why Bluetooth is powered off in the first place. I have no clue why this happens.

closed 2022-02-24 by github-actions[bot]