home-assistant/operating-system #1268: New ODROID N2+ installation failed

hghaken, 2021-03-12:
- [ ] Raspberry Pi [1/2/3/4]
- [x] ODROID [N2+]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ x] Fresh installation of release: hassos_odroid-n2-5.12.img.xz

Just received my new ODROID N2+ (new stock)
Flashed 128Gb eMMC card with the USB3.0 to eMMC Reader and balenaEtcher v1.5.116.
Flashed ok, no errors.

After boot you get error:
hdmi-audio-codec hdmi-audio-codec.4.auto: Not able to map channels to speakers (-22)

Then after root login:
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x3f08ec]
See picture

[image]

agners, 2021-03-17:
There are three issues here:

1. The banner not showing on start (this should be resolved with v4.11.0 release https://github.com/home-assistant/cli/releases/tag/4.11.0)
2. The kernel messages with regards to HDMI
3. No IP address assigned when no HDMI connected

For number 2: Does you HDMI device has Audio capabilities? This might be related to #1216.

For number 3: That is odd really, I run Blue always without HDMI, never saw this issue. Is this reproducible? Any messages in `dmesg`?

hghaken, 2021-03-18:
I have installed the developper version haos_odroid-n2-6.0.dev20210317.img.xz
This version still gives me the HDMI audio error, but it will boot the kernel.

IP address is received from the router and Home Assistant can be installed.
(With no HDMI monitor attached)
I can live with the HDMI error because i will run the Odroid N2+ headless.

Thanks for the support!

agners, 2021-03-19:
Ok, thanks for the update. Closing this, as we track the HDMI error on #1216.

closed 2021-03-19 by agners