home-assistant/operating-system #1270: 5.12 OVA will not boot properly

Stapleton, 2021-03-13:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x] OVA (Open Virtualization Applicance, VirtualBox 6.1)

OS: Ubuntu Server 20.04.2 LTS
Kernel: 5.4.0-66
HV: Virtualbox 6.1

**Home Assistant OS release:**
- [x] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

OS in Emergency mode

**Journal logs:**

Dont really know how I am supposed to get these logs off when the OS is in a partial working state

**Kernel logs:**

Dont really know how I am supposed to get these logs off when the OS is in a partial working state

**VirtualBox logs:**
2 logs and vbox config
https://gist.github.com/Stapleton/dbc25d2e12d3a946a19caac8de854dc5

**Description of problem:**

https://i.stapleton.pw/2021-03-12%2019-58-08.mp4
Fails to load the data partition in both VDI and VMDK formats.

I had a previous install that was version 5.8 and I updated to 5.10 and the VM never booted again, really getting fed up with the reliability of this project.

I hope I have supplied enough information in hopes to get to the bottom of this.

agners, 2021-03-16:
> I had a previous install that was version 5.8 and I updated to 5.10 and the VM never booted again, really getting fed up with the reliability of this project.

I am sorry about this, this is likely what is described in #1125. We made some fixes which might address the problem, but I haven't been able to reproduce the problem here so far, so its hard to tell.

Back to the case at hand: Just tested the vdi and ova image on VirtualBox 6.1.18 r142142, and they seem to work on my end (following https://www.home-assistant.io/installation/linux).

Do you use a special hard drive configuration (e.g. second hard drive?)

Stapleton, 2021-03-17:
> Just tested the vdi and ova image on VirtualBox 6.1.18 r142142, and they seem to work on my end
I havent been able to get VBox to install nicely on my system. It seems to expect the generic kernel but I have the liquorix kernel installed. So I use VMWare on my main machine and Virtualbox on my server.

I experienced the same, it worked for both VDI and VMDK but then I put those exact same images on the server and they either boot to the UEFI shell or not at all.

> We made some fixes which might address the problem, but I haven't been able to reproduce the problem here so far, so its hard to tell.
I would be happy to help figure out if there is a pattern for the systems that are having this issue.
Better off that I try to contribute rather than complain.

> Do you use a special hard drive configuration (e.g. second hard drive?)
Sorta?
https://gist.github.com/Stapleton/74f2dbed64edacdcce4502a56ee6488a

Stapleton, 2021-04-06:
Alright, so I stopped my UnRAID array, and started it again and now the HA VM is busted.
It seems like some of the services fail in a "one and done" kind of sitch.
The first boot, it works awesome. Every time. Like this could be solved with a brand new image every time I start the VM but a VM without persistence in this situation is really lame

Stapleton, 2021-04-06:
My Server setup has changed. Here is the [QEMU XML](https://gist.github.com/Stapleton/14e7710c81430dedc91b0e4980edc880)

**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance)

OS: UnRAID 6.9.1
Kernel: Linux 5.10.21-Unraid x86_64
HV: qemu x86-64

**Home Assistant OS release:**
- [x] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

agners, 2021-04-07:
> Alright, so I stopped my UnRAID array, and started it again and now the HA VM is busted.
> It seems like some of the services fail in a "one and done" kind of sitch.
> The first boot, it works awesome. [...]

I am not quite sure what happens exactly, but in both cases you have a second hard drive attached to the virtual machine (hdc in qemu case). If you have been using the data disk feature, it could be that a file system on that partition has the label `hassos-data`, which we use to identify the data partition. Now when you use a new ova images, it comes with a partition named `hassos-data` on its own. So you end up with two partitions named the same, which can cause troubles.

Can you try and make sure that the OVA images behaves well in your environment without second disk?

We currently don't have a documented/established process to use a data disk from an existing/previous installation. However, it should work by simply renaming the internal data partition. So basically (but only use this if are sure the second disk has a partition named `hassos-data`!):

1. Boot **without** second disk
2. Login as `root`, if the HA CLI appears use login to enter the OS shell
3. Use: `e2label $(findfs LABEL=hassos-data) hassos-data-old` to rename the data partition on the OS image
4. Shutdown using `poweroff`
5. Attach the second disk again
6. Boot **with** second disk. Now the second disk should be used as data partition reliably on every boot.

> Using a VNC connection and logging into HA with `root` the only thing that gets printed is
> `[WARN] Home Assistant CLI is not running! Jump into emergency console...`

On first boot it takes a while until Supervisor and all its plug-ins are setup, so this is normal. Waiting a while and logout and login should not show the message again.

closed 2022-01-11 by github-actions[bot]