home-assistant/operating-system #1271: gpio issues

comet424, 2021-03-13:
so there is 2 issues  gpios are broken.. and reboots of HA trigger gpio pins when its not supposed to
here i have the logs and such  that remote gpio is broken  and currently using 4.3
https://community.home-assistant.io/t/can-not-get-remote-rpi-gpio-to-work/289509/5

and the link for gpio going high when supposed to be low
https://community.home-assistant.io/t/how-do-you-stop-gpio-pins-going-high-low-on-a-configuration-restart/288758/7

agners, 2021-03-16:
What OS version are you using exactly? Is it really 4.3? That is a rather old version at this point with known issues.

comet424, 2021-03-16:
sorry my dislexis
its 3.4  not 4.3   so 2021.3.4  i tried to installed 3.6  but it wouldnt update and then it was gone..
i tried to do what you do with raspbian  where you set  in the rc local  or so  to force the pin up or down on boot up

i also having issues with the remote_rpi_gpio    its throwing errors 
and trying to get   mqtt for remote rpi monitor to work..
so i been working on 3 different rpi gpio issues all the same time..

still new at this stuff.. i still new with HA  and trying to get comfortable using it etc.. and my dislexia  messes up at times

comet424, 2021-03-16:
oh and thats the core.. i dunno if you need to know the supervisor version but its 3.6  and operating system is 5.12

comet424, 2021-03-16:
what i did now  was i right clicked on each file..  saved it and put it in the remote_rpi_gpio folder... i reboot HA  and i got similar errors?  
`2021-03-16 08:03:04 ERROR (SyncWorker_2) [homeassistant.loader] Error parsing manifest.json file at /config/custom_components/remote_rpi_gpio/manifest.json: Expecting value: line 7 column 1 (char 6)
2021-03-16 08:03:04 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-03-16 08:03:15 ERROR (MainThread) [homeassistant.components.binary_sensor] remote_rpi_gpio: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 465, in async_device_update
await task
File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
result = self.fn(*self.args, **self.kwargs)
File "/usr/src/homeassistant/homeassistant/components/remote_rpi_gpio/binary_sensor.py", line 100, in update
self._state = remote_rpi_gpio.read_input(self._button)
File "/usr/src/homeassistant/homeassistant/components/remote_rpi_gpio/__init__.py", line 61, in read_input
return button.is_pressed
AttributeError: 'NoneType' object has no attribute 'is_pressed'
2021-03-16 08:03:16 ERROR (MainThread) [homeassistant.components.binary_sensor] remote_rpi_gpio: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 465, in async_device_update
await task
File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
result = self.fn(*self.args, **self.kwargs)
File "/usr/src/homeassistant/homeassistant/components/remote_rpi_gpio/binary_sensor.py", line 100, in update
self._state = remote_rpi_gpio.read_input(self._button)
File "/usr/src/homeassistant/homeassistant/components/remote_rpi_gpio/__init__.py", line 61, in read_input
return button.is_pressed
AttributeError: 'NoneType' object has no attribute 'is_pressed'`

agners, 2021-03-16:
> so there is 2 issues gpios issues 1 is its broken in the OS

I guess with that you refer to the remote API stack trace?

Why do you think this is broken in Home Assistant OS specifically? The stack trace looks more like a bug in the integration/library. Is the same functionality working on other OS?

> reboots of HA trigger gpio pins when its not supposed to
This part does not belong to Home Assistant OS issue tracker, it belongs to the core repository or the library which is used for that integration which causes the problem.

comet424, 2021-03-16:
since i realtivilty new to home assistant 
on the home assistant page.. from what i understand
remote_rpi_gpio  is a home assistant application that they write...  and when i first installed HA  the remote_rpi_gpio_   like the instructions say 
https://www.home-assistant.io/integrations/remote_rpi_gpio/
but nothing on the page says  when you  update home assistant  remote rpi is broken  or what to do to make it work again..

thats why i figure its a home assistant thing..  its a raspberry pi thing and home assistant is on raspberry pi

the other issue   i do not know what the home assistant tracker is 
but what i talking about .. is on the each time Home assistant reboots..  it trigglers the GPIO pins when it shouldnt.
on raspbian  you tell it in like the rc.local  or something like that i forget   you say the OP= 23 D  or something i guessing right now but to force the pin to stay low on boot up  before the OS loads.    but home assistant doesnt have this . least its not in  any of the configurations  to tell ya this is what you need to do  to ensure on a home assistant reboot  the pin stays in LOW or High or stays in the state it was in kinda thing..

so  if my question doesnt beling on the HA OS issue tracker.
can you direct me to the HA OS  Core Repository community help page to ask my question about gpio tripping
sorry for being in the wrong forum
so i guess HA OS Core Reposiity/Library forum be where to ask

but when i said there is 2 issues 
its because i showed   2  different gpio issues   with 2 different topics  
1 is for local raspberry pi  the other one is  remote raspberry pi

hope that clears it up.. but if you can direct me to the HA OS Core  to see if they can tell me how to fix the gpio issue trigger issue

comet424, 2021-03-24:
so far have not found any fix  or a fix for remote_rpi_gpio  since that no longer works with HA   
ill have to try different route to fix my 2 issues 
of HA triggering the gpio  on reboot..as it doesnt stay low like its supposed to  and 2nd  is the remote_rpi_gpio

agners, 2021-03-30:
Note that some GPIO on RPi4 are pulled high in hardware (see chapter 5.1.2 GPIO Alternate Functions in https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0_preliminary.pdf).

FrancescoBarbieri1, 2021-04-13:
From the 3-4-2021 I have this error and binary_sensor don't run any more.
I haven't change anything... Sigh sigh sigh :(

_ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up rpi_gpio platform for binary_sensor
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 205, in _async_setup_platform
await asyncio.shield(task)
File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
result = self.fn(*self.args, **self.kwargs)
File "/usr/src/homeassistant/homeassistant/components/rpi_gpio/binary_sensor.py", line 48, in setup_platform
RPiGPIOBinarySensor(
File "/usr/src/homeassistant/homeassistant/components/rpi_gpio/binary_sensor.py", line 81, in __init__
rpi_gpio.edge_detect(self._port, edge_detected, self._bouncetime)
File "/usr/src/homeassistant/homeassistant/components/rpi_gpio/__init__.py", line 48, in edge_detect
GPIO.add_event_detect(port, GPIO.BOTH, callback=event_callback, bouncetime=bounce)
RuntimeError: Failed to add edge detection_

FrancescoBarbieri1, 2021-04-13:
May be an upgrade that need an upgrade of the library for binary_sensor....
But I don't know what I must do

FrancescoBarbieri1, 2021-04-13:
If I take away all the code that use binary_sensor of course it run without errors

FrancescoBarbieri1, 2021-04-13:
If I re add:
```
binary_sensor:
  - platform: rpi_gpio
    ports:
      11: IN1
```

without use it I have not Errors

FrancescoBarbieri1, 2021-04-13:
If I add only one:

```
  - alias: 'Avviso Allarme Portone'
    trigger:
      - platform: state
        entity_id: binary_sensor.in2
        to: 'on'
    condition:
      - condition: state
        entity_id: input_select.allarme
        state: 'ON'
      - condition: state
        entity_id: input_select.abportone
        state: 'ON'
    action:
      - service: notify.francesco
        data:
          message: "Allarme portone"
      - service: notify.cristina
        data:
          message: "Allarme portone"
```
I have the Error

FrancescoBarbieri1, 2021-04-13:
Mmmm I have not error if I use other port different from 7

FrancescoBarbieri1, 2021-04-13:
OK!!!! I find the error, don't use rpi_gpio port 7 and 8. Not work with new release OS!!!!!

```
binary_sensor:
  - platform: rpi_gpio
    ports:
      11: IN1 #cavo blu portoncino ingresso? non ricordo
#      7: IN2 #portone ingresso   **(Don't run any more!!!!!!!! Take away the seven and all the code that call it)**
      5: IN3 #finestre e portone garage
      6: IN4 #PIR garage
      12: IN5 #balconi notte
      13: IN6 #campanello cancello
      16: IN8 #finestre notte
      26: IN9 #PIR soggiorno
      20: IN10 #balconi giorno
      21: IN11 #finestre giorno e cucina
```

comet424, 2021-04-13:
i have   8 working with the latest OS of HA    i use it to detect a mometary switch.. i know for it 
i had to set   pull_mode: up    
to get it to work properly  

and i still have remote_rpi_gpio  issues.. i ended up  installing  HA  on the remote pi   and let HA  access the Remote HA .. and that works for me.. had too much headaches the gpios  not working right etc

comet424, 2021-04-13:
this is how i got GPIO 8 to work 
# Display State of Pump Switch
binary_sensor:
  - platform: rpi_gpio
    pull_mode: up
    invert_logic: true
#    bouncetime: 1000
    ports:
      8: Pump Switch     # Mometary Physical Switch

comet424, 2021-04-13:
i cant post code right even inside the code  option doesnt post right