home-assistant/operating-system #1272: NUC Compatible - Shutdown doesn't power it off

marciogranzotto, 2021-03-14:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [x] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

[image]

**Journal logs:**

**Kernel logs:**

**Description of problem:**

I'm using an off-brand (Shenzhen Tengyatong TG-J1800) Mini PC that uses an Intel Celeron J1800, 4GB of RAM, 128GB SSD.
The issue is if I try to shut it off, using either the Supervisor UI ( Supervisor > System > Shutdown Host ), or `ha host shutdown`, or even pressing the physical power button, the console shows all the shutdown logs, but then gets stuck on `Reached target Power-off` and doesn't power down the PC.
I tried to wait for like an hour, but it never powers down.

Here's a picture of the actual board for reference:
[image]

marciogranzotto, 2021-03-15:
On other OS', like Proxmox, the same hardware has no problem powering down after the shutdown procedure.

agners, 2021-03-16:
On x86 with modern ACPI a poweroff should be triggered by the kernel and lead to poweroff on ACPI side.

Do you use EFI boot for other OS' as well? Does maybe a BIOS update help? Another thing you can try is using one of the latest nightly builds which comes with a newer Linux kernel version: https://os-builds.home-assistant.io/6.0.dev20210311/

marciogranzotto, 2021-03-16:
> Do you use EFI boot for other OS' as well?

yeah, its a regular Proxmox 6.3 installation, so I do think so.

> Does maybe a BIOS update help?

I tried to find a newer BIOS, but I had no luck, since this board is so off-brand

> Another thing you can try is using one of the latest nightly builds which comes with a newer Linux kernel version: https://os-builds.home-assistant.io/6.0.dev20210311/

I'll do that and report back

marciogranzotto, 2021-03-22:
@agners Proxmox started to behave the same way. Also, the nightly build didn't work.
I guess the issue is on the BIOS 😕 
I'll close the issue for now. Thanks!

closed 2021-03-22 by marciogranzotto