home-assistant/operating-system #1273: Realtek bluetooth 5 dongle fails from time to time

clau-bucur, 2021-03-15:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [x ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.12
- Additional information (if accessible):

## System Health

version | core-2021.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.4.99
arch | x86_64
timezone | Europe/Bucharest

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4774
Installed Version | 1.11.3
Stage | running
Available Repositories | 833
Installed Repositories | 26

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 11

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 12 April 2021, 03:00
relayer_connected | true
remote_enabled | false
remote_connected | false
alexa_enabled | false
google_enabled | true
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.03.4
[... 122 lines cut ...]
**Journal logs:**

**Kernel logs:**

**Description of problem:**

From time to time the Bluetooth USB dongle stops working and I simply can't get it to work again.
This is the host log when the device is inserted:
```
[  559.500685] usb 1-2: USB disconnect, device number 4
[  587.116457] usb 1-2: new full-speed USB device number 5 using xhci_hcd
[  587.444249] usb 1-2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[  587.444250] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  587.444251] usb 1-2: Product: Bluetooth Radio
[  587.444252] usb 1-2: Manufacturer: Realtek
[  587.444253] usb 1-2: SerialNumber: 00E04C239987
[  587.449150] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[  587.450112] Bluetooth: hci0: RTL: rom_version status=0 version=1
[  587.450113] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761a_fw.bin
[  587.450123] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761a_fw.bin failed with error -2
[  587.450125] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761a_fw.bin not found
```

This is not HassOS 5.12 related, it behaved the same in older versions also.
I cannot say how do I get it to work again, I simply try countless times to plug it back in or restart the VM and it will start to work at some point.

agners, 2021-03-17:
This sounds like a missing firmware.

clau-bucur, 2021-03-17:
It works most of the time, so shouldn't mean that the firmware is there?

agners, 2021-03-18:
`-2` means "No such file or directory", and from what I can see Home Assistant OS has that firmware not installed. Not sure why it works sometime, some devices have a default firmware (which might be buggy and hence work only sometimes?), or it could be that your host Operating System's driver loads the firmware beforehand maybe?

clau-bucur, 2021-03-18:
I see. Is there a way that I can add the firmware manually until it will be a standard part of HassOS ?

kelos-01, 2021-03-29:
Not sure if I should log a bug/enhancement as well, seem to be having the same issue. I see the commit is completed but still in Dev. (btw thanks @agners)

```
[16263.319768] usb 1-1.4: USB disconnect, device number 4
[16642.754635] usb 1-1.2: new full-speed USB device number 5 using dwc_otg
[16642.887871] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=8771, bcdDevice= 2.00
[16642.887899] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[16642.887919] usb 1-1.2: Product: Bluetooth Radio
[16642.887936] usb 1-1.2: Manufacturer: Realtek
[16642.887953] usb 1-1.2: SerialNumber: 00E04C239987
[16642.907465] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[16642.909456] Bluetooth: hci0: RTL: rom_version status=0 version=1
[16642.909486] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[16642.923271] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[16642.923306] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

clau-bucur, 2021-04-03:
@agners is this fix included in 5.13?
The 5.13 update does not show up in HA though.

clau-bucur, 2021-05-04:
Looks like 5.13 does not contain the fix, as the behavior remains unchanged after 5.13 upgrade.

GreyTeardrop, 2021-05-14:
I've upgraded to [OS 6.0.rc1](https://github.com/home-assistant/operating-system/releases/tag/6.0.rc1) that is supposed to have Realtek Bluetooth drivers bundled according to the release notes, but alas, I see the same issue. My host is ODroid N2+ (Home Assistant Blue)
```
~ $ bluetoothctl list
~ $ ha os info
board: odroid-n2
boot: B
update_available: false
version: 6.0.rc1
version_latest: 6.0.rc1
~ $ ha host logs|grep -i bluetooth
[    3.461272] Bluetooth: Core ver 2.22
[    3.465123] Bluetooth: HCI device and connection manager initialized
[    3.471016] Bluetooth: HCI socket layer initialized
[    3.475603] Bluetooth: L2CAP socket layer initialized
[    3.482178] Bluetooth: SCO socket layer initialized
[    3.497825] Bluetooth: hci0: RTL: examining hci_ver=0a hci_rev=000b lmp_ver=0a lmp_subver=8761
[    3.504819] Bluetooth: hci0: RTL: rom_version status=0 version=1
[    3.509350] Bluetooth: hci0: RTL: loading rtl_bt/rtl8761b_fw.bin
[    3.517721] bluetooth hci0: Direct firmware load for rtl_bt/rtl8761b_fw.bin failed with error -2
[    3.524450] bluetooth hci0: Falling back to sysfs fallback for: rtl_bt/rtl8761b_fw.bin
[   64.482850] Bluetooth: hci0: RTL: firmware file rtl_bt/rtl8761b_fw.bin not found
```

wzaatar, 2021-06-02:
Confirming the issue is still there with OS 5.13 and Supervisor 2021.05.4 also on ODROID (HA Blue). 

I tried following the instructions at https://community.home-assistant.io/t/hass-io-and-usb-bluetooth-dongle/217917/4 but a reboot erases the folder and files.

The issue title should be changed to "fails all the time"...

agners, 2021-06-05:
@clau-bucur @wzaatar the fix is in the development branch, so no changes in 5.13 or future 5 releases.

@GreyTeardrop @kelos-01 your devices seem to request a different firmware (note the `b` in `rtl_bt/rtl8761b_fw.bin`). I'll check if I can add this firmware as well.

agners, 2021-06-05:
It seems that `rtl_bt/rtl8761b_fw.bin` is not available in the upstream linux-firmware repository. Splitting this firmware into a separate issue #1392.

The firmware @clau-bucur was mentioning (`rtl_bt/rtl8761a_fw.bin`) will be shipped in release 6 (see #1283). Hence closing.

closed 2021-06-05 by agners