home-assistant/operating-system #1281: Cannot boot CM4 from USB

fencer, 2021-03-18:
**Hardware Environment**

- [ ] Raspberry Pi Compute 4 Lite 4GB WiFi

**Home Assistant OS release:**
- [ ] Fresh installation of release 5.12 x64
- Additional information (if accessible):

**Serial Debug log:**
```
U-Boot 2020.10 (Feb 18 2021 - 14:33:54 +0000)

DRAM:  3.9 GiB
RPI Compute Modue 4 (0xc03140)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link down
starting USB...
No working controllers found
Card did not respond to voltage select!
Card did not respond to voltage select!
starting USB...
No working controllers found
USB is stopped. Please issue 'usb start' first.
starting USB...
No working controllers found
PCIe BRCM: link down
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/01-dc-a6-32-fe-78-6b
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
ethernet@7d580000 Waiting for PHY auto negotiation to complete......... TIMEOUT !
```

Trying to boot from USB SSD getting the above. (Official RaspberryOS works)

agners, 2021-03-19:
> Official RaspberryOS works

Raspberry OS does not use U-Boot, so that make sense.

It looks like U-Boot cannot see the USB controller. The latest dev builds have a new U-Boot, maybe it has been fixed. Can you try a 6.0 development release from here?
https://os-builds.home-assistant.io/6.0.dev20210317/

fencer, 2021-03-19:
Same issue with that version, only difference I got the U-Boot output on the HDMI display and didn't need to connect via serial. Could it be related to the fact that CM4 does not have the USB2.0 enabled by default?

agners, 2021-03-20:
> Could it be related to the fact that CM4 does not have the USB2.0 enabled by default?

Quite possible, I think U-Boot reuses the device tree passed from firmware.

Try adding the USB 2.0 `dtoverlay` in `config.txt` as described here: https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=292632

fencer, 2021-03-20:
I've tried it out but the result was the same. My thought is that if the USB controller is not found then the dto will never be read.

fencer, 2021-05-20:
Seems someone has solved the issue in U-Boot here: https://www.raspberrypi.org/forums/viewtopic.php?t=308542 will try to have a shot at fixing it in the coming weeks if no one fixes it by then :)

fencer, 2021-10-05:
I have had a shot the[ latest  v7 build](https://os-builds.home-assistant.io/7.0.dev20211004/haos_generic-x86-64-7.0.dev20211004.img.xz) hoping that this has been address however there has been no joy  although I now have something on screen rather than just a blank screen.
[image]

agners, 2021-10-05:
It seems to confirm what the user in the Raspberry Pi forum said. It seems that the necessary driver for the Broadcom XHCI controller is not enabled. It should be trivial to enable it, I'll have a look at it.

agners, 2021-10-06:
@fencer can you give the latest nightly build `7.0.dev20211005` a try? It includes the patch and enables the BRCM XHCI driver.

agners, 2021-10-07:
There was one more fix missing. Tested with the changes made in #1577, with that it seems to work here. I'll trigger a dev build later today.

closed 2021-10-07 by agners