home-assistant/operating-system #1298: Blue ODROID hangs with a memory problem systemd-coredump

xerxel, 2021-04-03:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [*] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [*] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):

## System Health

version | core-2021.3.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.7
os_name | Linux
os_version | 5.9.16
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4868
Installed Version | 1.11.3
Stage | running
Available Repositories | 841
Installed Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | undefined

</details>

<details><summary>Hass.io</summary>

host_os | Home Assistant OS 5.12
-- | --
update_channel | stable
supervisor_version | supervisor-2021.03.6
docker_version | 19.03.13
disk_total | 113.9 GB
disk_used | 14.3 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.3.1), Home Assistant Google Drive Backup (0.103.1), Check Home Assistant configuration (3.6.0), Duck DNS (1.12.5), File editor (5.2.0), SSH & Web Terminal (8.1.0), AdGuard Home (4.0.0), DHCP server (1.2), ESPHome (1.16.2), Log Viewer (0.9.1), Node-RED (8.2.1)

</details>

<details><summary>Lovelace</summary>

[... 94 lines cut ...]
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.check] System checks complete
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
21-04-03 21:06:13 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
21-04-03 21:06:13 INFO (MainThread) [supervisor.host.services] Updating service information
21-04-03 21:06:13 INFO (MainThread) [supervisor.host.network] Updating local network information
21-04-03 21:06:14 INFO (MainThread) [supervisor.api.security] /dns/info access from a0d7b954_logviewer
21-04-03 21:06:15 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-04-03 21:06:15 INFO (MainThread) [supervisor.host] Host information reload completed
21-04-03 21:06:30 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request initialize
21-04-03 21:06:30 INFO (MainThread) [supervisor.api.proxy] WebSocket access from a0d7b954_nodered
21-04-03 21:06:31 INFO (MainThread) [supervisor.api.proxy] Home Assistant WebSocket API request running
```

**Journal logs:**

**Kernel logs:**

**Description of problem:**

From time to time, everything will suddenly stop including core, DHCP, DNS etc. Tonight is the first time I managed to capture an error - see screenshot attached. Looks like a memory leak perhaps?

[image]

xerxel, 2021-04-04:
Attached is the result of `journalctl -b 1 ` albeit not from the previous boot which had the crash but it does show quite a few errors.
[log.txt](https://github.com/home-assistant/operating-system/files/6253795/log.txt)

xerxel, 2021-04-06:
OK - It just crashed again/locked up and brought the internet down due to DNS/ DHCP etc! When the system froze at around 22:52:30 there are audio errors in the log. Then a did a manual hard reboot and on startup there are lots of audio errors. Could these be a cause?

[logaudio.txt](https://github.com/home-assistant/operating-system/files/6261178/logaudio.txt)

xerxel, 2021-04-06:
Also getting these errors in Core log:

```
2021-04-05 23:51:59 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
2021-04-05 23:51:59 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
2021-04-05 23:52:01 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
2021-04-05 23:52:03 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
2021-04-05 23:52:04 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
2021-04-05 23:52:06 ERROR (MainThread) [homeassistant.components.apple_tv] Failed to connect
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 245, in _connect_loop
    await self._connect(conf)
  File "/usr/src/homeassistant/homeassistant/components/apple_tv/__init__.py", line 340, in _connect
    'Connection was re-established to Apple TV "%s"', self.atv.service.name
AttributeError: 'MrpService' object has no attribute 'name'
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[00:07:59] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration garbage_collection which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration sonoff which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration breaking_changes which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration samsungtv_smart which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration alexa_media which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-04-06 00:08:08 WARNING (MainThread) [homeassistant.loader] You are using a custom integration automower which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
[... cut ...]
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1131, in sendfile
    return await self._sendfile_fallback(transport, file,
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 1161, in _sendfile_fallback
    await proto.restore()
  File "/usr/local/lib/python3.8/asyncio/base_events.py", line 263, in restore
    self._transport.resume_reading()
  File "/usr/local/lib/python3.8/asyncio/sslproto.py", line 344, in resume_reading
    self._ssl_protocol._transport.resume_reading()
AttributeError: 'NoneType' object has no attribute 'resume_reading'
```

agners, 2021-04-06:
It seems that the system runs out-of-memory. From the picture attached it seems that a python3 process was using a lot of memory, so that could be Home Assistant Core or the Supervisor. Do you have add-ons running? I would try disabling add-ons/integrations to isolate the component which is causing the memory issue.

closed 2022-01-11 by github-actions[bot]