home-assistant/operating-system #1299: Odroid N2+ HASSOS cannot boot from USB SSD

ced1442, 2021-04-03:
Hi, I've bought an Odroid N2+ and try to make it boot on HassOS from a SSD USB drive using the SPI boot, but it is not booting at all... Petitboot didn't find any bootable partition.
Developers are pushing for this hardware instead of Raspberry Pi but it is for me a drawback compared to Raspberry Pi that allow booting HassOS from USB.
Do I miss something ? How can we push for that feature  ?
For the moment, I'm running it in supervised mode on docker within a debian light OS (dietpi), so I'm not blocked.

agners, 2021-04-06:
Unfortunately Petitboot only supports booting a Linux kernel from a USB drive. However, Home Assistant OS needs U-Boot logic for failsafe mechanism etc. I tried making Petitboot to load U-Boot, but wasn't successful.

However, I also think booting from USB is really not needed anyways. The eMMC has quite a long lifetime. Also you can move data to a USB drive using the datactl method (see https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#data-disk). With this, the OS can use the pretty fast internal eMMC, and at the same time use a fast external USB 3.0 drive for data.

pepsonEL, 2021-04-07:
I think that it is need. SSD Disk is lower cost than eMMC. Some people has free SSD which we need to connect to boot Home Assistant. My eMMC i have about 1 year on ODroid N2 and now show me 70-80% life...
Now in fact that Odroid no boot from USB i think to migrate on RPI4 8G to use SSD.

agners, 2021-04-07:
> My eMMC i have about 1 year on ODroid N2 and now show me 70-80% life...

If you use data disk feature, all data will be written to a USB disk. There will be next to zero writes on the eMMC (unless you update the OS or configure aspects of the OS). Even now that 80% of the eMMC life time is used, if you use that method you will be able to use your ODROID-N2 for a long time still.

pepsonEL, 2021-04-07:
I dont understand..... can you explaine me more ?

agners, 2021-04-07:
The picture in the link above is the best way to explain it:

[image]

This setup is better than boot from USB SSD since the OS can read with full IO performance from the on-board eMMC while the rest of Home Assistant (Core, database etc) has full performance of the USB SSD.

pepsonEL, 2021-04-07:
Ok but how i can configure it ?

agners, 2021-04-07:
https://github.com/home-assistant/operating-system/blob/dev/Documentation/partition.md#using-datactl-to-move-the-data-partition

pepsonEL, 2021-04-07:
But if HA on odroid N2 show me 70-80% life time what i must understand about it ?Remaining 70-80% life ? Or use 70-80% and remaining  only 20% ?

agners, 2021-04-08:
The title in the Supervisor System Tab says "eMMC Lifetime Used". So yes, remaining only 20%.

But again, if you only run the **read-only** OS that will be plenty of write cycles! Also this is a _worst case_ estimate. And third, even if you run into problems, you data is on the data disk, you would only use the OS which you can replace easily.

closed 2021-04-08 by agners

pepsonEL, 2021-04-08:
but if i move this data also custom_componetnt and other also will be move ? And when i create snapshot it will be work ? And after accident when i restore snapshot it will be work ?