home-assistant/operating-system #1303: Blue ODROID no supervisor internet connection

thebino, 2021-04-07:
**Hardware Environment**
- [ ] Raspberry Pi [1/2/3/4]
- [X] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [x] Fresh installation of release 5.12
- [ ] Updated from version x.y
- Additional information (if accessible):
- 
**Supervisor logs:**
```
21-04-07 12:02:15 INFO (MainThread) [__main__] Initializing Supervisor setup
21-04-07 12:02:15 INFO (MainThread) [supervisor.docker.network] Can't find Supervisor network, creating a new network
21-04-07 12:02:15 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-04-07 12:02:20 WARNING (MainThread) [supervisor.utils.dt] Can't fetch freegeoip data: Cannot connect to host ip-api.com:80 ssl:default [Try again]
21-04-07 12:02:20 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: odroid-n2
21-04-07 12:02:20 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.03.6
21-04-07 12:02:20 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
21-04-07 12:02:20 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-04-07 12:02:20 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-04-07 12:02:20 INFO (MainThread) [__main__] Setting up Supervisor
21-04-07 12:02:20 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-04-07 12:02:20 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-04-07 12:02:21 INFO (MainThread) [supervisor.host.info] Updating local host information
21-04-07 12:02:21 INFO (MainThread) [supervisor.host.services] Updating service information
21-04-07 12:02:21 INFO (MainThread) [supervisor.host.network] Updating local network information
21-04-07 12:02:22 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-04-07 12:02:22 INFO (MainThread) [supervisor.host] Host information reload completed
21-04-07 12:02:22 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-04-07 12:02:22 INFO (MainThread) [supervisor.host.services] Reloading local service hassos-apparmor.service
21-04-07 12:02:22 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-dns versions: [<AwesomeVersion CalVer '2021.01.0'>]
21-04-07 12:02:22 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-dns with version 2021.01.0
21-04-07 12:02:22 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-04-07 12:02:22 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS homeassistant/aarch64-hassio-dns with version 2021.01.0 - 172.30.32.3
21-04-07 12:02:22 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-04-07 12:02:22 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-audio versions: [<AwesomeVersion CalVer '2021.02.1'>]
21-04-07 12:02:22 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-audio with version 2021.02.1
21-04-07 12:02:22 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-04-07 12:02:23 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio homeassistant/aarch64-hassio-audio with version 2021.02.1 - 172.30.32.4
21-04-07 12:02:23 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-cli versions: [<AwesomeVersion CalVer '2021.03.1'>]
21-04-07 12:02:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-cli with version 2021.03.1
21-04-07 12:02:23 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-04-07 12:02:24 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI homeassistant/aarch64-hassio-cli with version 2021.03.1 - 172.30.32.5
21-04-07 12:02:24 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-observer versions: [<AwesomeVersion CalVer '2020.10.1'>]
21-04-07 12:02:24 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-observer with version 2020.10.1
21-04-07 12:02:24 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
21-04-07 12:02:24 INFO (SyncWorker_1) [supervisor.docker.observer] Starting Observer homeassistant/aarch64-hassio-observer with version 2020.10.1 - 172.30.32.6
21-04-07 12:02:24 INFO (SyncWorker_0) [supervisor.docker.interface] Found homeassistant/aarch64-hassio-multicast versions: [<AwesomeVersion BuildVer '3'>]
21-04-07 12:02:24 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-hassio-multicast with version 3
21-04-07 12:02:24 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
21-04-07 12:02:25 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast homeassistant/aarch64-hassio-multicast with version 3 - Host
21-04-07 12:02:25 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-04-07 12:02:35 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: 
21-04-07 12:02:35 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
21-04-07 12:02:35 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
21-04-07 12:02:35 INFO (SyncWorker_1) [supervisor.docker.interface] No version found for homeassistant/odroid-n2-homeassistant
21-04-07 12:02:35 INFO (MainThread) [supervisor.homeassistant.core] No Home Assistant Docker image homeassistant/odroid-n2-homeassistant found.
21-04-07 12:02:35 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to homeassistant/odroid-n2-homeassistant with version landingpage
21-04-07 12:02:35 INFO (MainThread) [supervisor.homeassistant.core] Using preinstalled landingpage
21-04-07 12:02:35 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
21-04-07 12:02:35 INFO (MainThread) [supervisor.homeassistant] Update pulse/client.config: /data/tmp/homeassistant_pulse
21-04-07 12:02:35 INFO (SyncWorker_0) [supervisor.docker.homeassistant] Starting Home Assistant homeassistant/odroid-n2-homeassistant with version landingpage
21-04-07 12:02:35 INFO (MainThread) [supervisor.hassos] Detect HassOS 5.13 / BootSlot A
21-04-07 12:02:35 WARNING (MainThread) [supervisor.jobs] 'StoreManager.update_repositories' blocked from execution, no supervisor internet connection
21-04-07 12:02:35 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
21-04-07 12:02:35 INFO (MainThread) [supervisor.snapshots] Found 0 snapshot files
21-04-07 12:02:35 INFO (MainThread) [supervisor.discovery] Loaded 0 messages
[... 70 lines cut ...]
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_interface: true
interfaces:
- connected: true
  enabled: true
  interface: eth0:
  ipv4:
    addresses:
     172.17.2.145/20
     gateway: 172.17.2.1
     method: auto
    nameservers:
      - 8.8.8.8
      - 84.200.69.80
      - 9.9.9.9
    ipv6:
    ...
    primary: true
    type: ethernet
    vlan: null
    wifi: null
supervisor_internet: false
```

agners, 2021-04-07:
This looks quite a bit like #1304. Can you connect a keyboard/HDMI monitor to your ODROID to access the system console? Login using root, and from the Home Assistant CLI use login to get to the OS shell. Run the following commands:

```
nslookup version.home-assistant.io
ping version.home-assistant.io
docker exec -it hassio_supervisor /usr/bin/nslookup version.home-assistant.io
```

thebino, 2021-04-07:
when loggin on the cli, i'm locked inside the `ha cli`
`exit` or `quite` will logout the user instead of leaving the cli.

agners, 2021-04-07:
You need to use the `login` command in the HA CLI to get to the OS shell.

agners, 2021-04-07:
```
eth0: 172.17.2.145/20
hassio: 172.30.32.1/23
docker0: 172.17.0.1/16
```

It seems that your network uses an overlapping IP address range to what Docker is using. Currently this can't be changed on the Home Assistant OS side, so you'll have to change the IP range of your network...

agners, 2021-06-11:
Marking this invalid as it seems to be related to environment issue (IP addresses).

closed 2021-06-11 by agners