home-assistant/operating-system #1304: fresh installation fails on Raspberry Pi 4

waffel, 2021-04-07:
### Describe the issue you are experiencing

after a really fresh installation following the documentation https://www.home-assistant.io/installation/raspberrypi using https://github.com/home-assistant/operating-system/releases/download/5.12/hassos_rpi4-64-5.12.img.xz 

I have added a CONFIG/network/my-network file to connect to my WLAN which seems to work (see log).

But the supervisor will not come up ... it stops after

21-02-02 17:18:50 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]

Got the information while accessing the device through http://<raraspberrypi IP>:4357

### What is the used version of the Supervisor?

unknown

### What type of installation are you running?

Home Assistant OS

### Which operating system are you running on?

Home Assistant Operating System

### What is the version of your installed operating system?

unknown

### What version of Home Assistant Core is installed?

unknown

### Steps to reproduce the issue

1. Download latest image and install as described in the documentatio
2. start PI 4 and take a look into the supervisor log
3. when using the mentioned URL in Browser (same network!) I get the JSON response from version.home-assistant.io:443 ... so the DNS (from my network) is ok

I guess there is an issue with the DNS setup when the PI starts.

I have also tried the same with local eternet LAN connection .. with the same result ... the PI gets an IP and I see the same errors from supervisor logs.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
# Put your logs below this line

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[17:18:27] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:18:27] INFO: Starting local supervisor watchdog...
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing... 
[17:18:30] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[... 90 lines cut ...]
21-02-02 17:19:55 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:20:00 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:20:00 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:20:30 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:20:35 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:20:35 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:21:05 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:21:10 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:21:10 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:21:40 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:21:45 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:21:45 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:22:15 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:22:20 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:22:20 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:22:50 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:22:55 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:22:55 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:23:25 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:23:30 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
21-02-02 17:23:30 WARNING (MainThread) [supervisor.homeassistant.core] Found no information about Home Assistant. Retry in 30sec
21-02-02 17:24:00 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
21-02-02 17:24:05 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]
...
```

agners, 2021-04-07:
Hm, not quite clear to me why the Supervisor can't connect to version.home-assistant.io:443. I assume there is no firewall in place?

Can you connect a keyboard/HDMI monitor to the Raspberry Pi to access the system console? Login using root, and from the Home Assistant CLI use `login` to get to the OS shell. Run the following commands:

```
nslookup version.home-assistant.io
ping version.home-assistant.io
docker exec -it hassio_supervisor /usr/bin/nslookup version.home-assistant.io
```

waffel, 2021-04-07:
OK ... I need first order a HDMI adapter and will test then with the console.

There is no firewall and from my laptop in the same network I can reach the URL ... but lets see what the console will say.

Thank you :-)

waffel, 2021-04-10:
So now I got my HDMI adapter and was able to login and provide you the requested information. I was able to create screenshots but hope this helps. I also took a look at the supervisor log which seems still have the same issue as reported. But it seems from docker container and also host system, the DNS seems to be OK.
[image]

I have also added 2 additional screenshot ... maybe they are related .. one shows an error after logging into hal with root (SEGFAULT) and one I saw when running `df -ah` on the host system.
[image]
[image]

Hope this helps a bit to find out the reason for the startup.

I forgott to mention: When I try any command from hal > I get a message, that the system is in "setup" state ... also after hours ... I need to login on the host system to halt the system.

agners, 2021-04-15:
> I have also added 2 additional screenshot ... maybe they are related .. one shows an error after logging into hal with root (SEGFAULT)

That was a bug in the Home Assistant CLI and will be resolved with the latest version:
https://github.com/home-assistant/cli/releases/tag/4.12.0

>  and one I saw when running `df -ah` on the host system.
We use a squashfs, that root is at 100% is expected.

> I forgott to mention: When I try any command from hal > I get a message, that the system is in "setup" state ... also after hours ... I need to login on the host system to halt the system.

This matches with what your initial log is showing: The system is stuck in setup state because it cannot resolve DNS names.

Is this still the case?
Can you check this command?

```
docker exec -it hassio_supervisor /usr/bin/nslookup version.home-assistant.io
```

waffel, 2021-04-16:
thanks for the explanation. As you see in the first screenshot I already have run the mentioned command.

It returns:
```
Server: 172.30.32.3
Address: 172.30.32.3#53

Non-authoritative answer:
Name: version.home-assistant.io
Address: 172.67.68.90
Name: version.home-assistant.io
Address: 104.26.4.238
Name: version.home-assistant.io
Address: 104.26.5.238
```

What is strange (I'm not an network expert) ... when running the same from the host the order is inverse:

```
Server: 192.168.1.1
Address: 192.168.1.1#53

Non-authoritative answer:
Name: version.home-assistant.io
Address: 104.26.5.238
Name: version.home-assistant.io
Address: 104.26.4.238
Name: version.home-assistant.io
Address: 172.67.68.90
```

When I'm running nslookup from another computer in the same network (windows laptop) I get the following result:
```
C:\Users\me>nslookup version.home-assistant.io
Server:  www.routerlogin.com
Address:  192.168.1.1

Nicht autorisierende Antwort:
Name:    version.home-assistant.io
Addresses:  2606:4700:20::ac43:445a
          2606:4700:20::681a:4ee
          2606:4700:20::681a:5ee
          172.67.68.90
          104.26.5.238
          104.26.4.238

C:\Users\me>ping version.home-assistant.io

Ping wird ausgeführt für version.home-assistant.io [172.67.68.90] mit 32 Bytes Daten:
Antwort von 172.67.68.90: Bytes=32 Zeit=27ms TTL=56
Antwort von 172.67.68.90: Bytes=32 Zeit=20ms TTL=56
Antwort von 172.67.68.90: Bytes=32 Zeit=17ms TTL=56

Ping-Statistik für 172.67.68.90:
    Pakete: Gesendet = 3, Empfangen = 3, Verloren = 0
    (0% Verlust),
Ca. Zeitangaben in Millisek.:
    Minimum = 17ms, Maximum = 27ms, Mittelwert = 21ms

```
So from this perspective everything looks OK. 

But I do not understand why the supervisor comes up with this warning:

`02 17:19:25 WARNING (MainThread) [supervisor.updater] Can't fetch versions from https://version.home-assistant.io/stable.json: Cannot connect to host version.home-assistant.io:443 ssl:default [Try again]`

Meanwhile I will try to jump into the container and try to fetch the json per hand. Lets see. Maybe you have another idea what to try?

waffel, 2021-04-16:
I have also tried (from PI console) to jump into the hassio_dns container ... from within this container I get correct answers from `dig `command. But when running `docker logs hassio_dns` I get many errors. See the screenshot attached ... maybe in the dns setup there is a problem?

[image]

ntindle, 2021-04-16:
Is this possibly related to #1306 as well?

agners, 2021-04-19:
It seems the container has problems connecting to a/the DNS server. Can you ping hosts from within the hassio_dns container (e.g. your local DNS `192.168.1.1` or the fallback `1.1.1.1`?)

agners, 2021-04-19:
Btw, to debug the issue setting up Home Assistant OS SSH access might be worthwhile: https://developers.home-assistant.io/docs/operating-system/debugging

ntindle, 2021-04-19:
I cannot ping hosts using 

```bash
ping 1.1.1.1
```

ntindle, 2021-04-21:
Slightly off topic but probably useful to others as well: Do you know where I can find documentation on downgrading versions?

waffel, 2022-04-21:
I can confirm, that in the meanwhile the issue seems to be solved. 

Thx