home-assistant/operating-system #1306: Unable to boot 5.13 on USB SSD on Rpi3

inverse, 2021-04-07:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [ ] Updated from version x.y
- Additional information (if accessible):

Raspberry PI3
With USB SSD

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**

nanowireUK, 2021-04-09:
Seeing same thing on a raspi4 booting from SD card. Upgraded to 5.13 over an hour ago and it's not booted back up.

jrousek, 2021-04-10:
Same on rpi4 and usb ssd boot. Afternoon test hdmi output.

jrousek, 2021-04-10:
[image] constantly around this.  don't boot.

carlhye, 2021-04-11:
Same issue here.

1. Installed 5.12 on Samsung T5 500GB USB SSD, worked fine
2. Upgraded to 5.13 and could not get to boot
3. Flashed 5.13 to see if this was related to disk failure, then not even monitor output
4. I Flashed 5.12 to start over, then everything worked fine 

R Pi 3B+
Samsung T5 500 GB USD SSD

inverse, 2021-04-11:
Exactly - I tried to see if there was any output on the monitor, and there was nothing :(

bobruddy, 2021-04-22:
I had the same issue on a pi4 4gb ram. Rpi64. 

Did an upgrade and wouldn't boot. This was a new install so I just reflashed. 

It boots. But behavior such as going to devices is broken.

caesar1987sk, 2021-06-11:
today i was forced to restart raspbery pi 3 with usb drive 30 times to start it after update to os 6.0

inverse, 2021-06-11:
ouch - yeah I stayed on `5.12` until this ticket is resolved.

bdurrer, 2021-06-22:
Updated to latest yesterday on PI3 B v1.2 with USB SSD (no sdcard) and it won't come up. It keeps logging OOM errors and attempts to start endlessly
[image]

Sigh, I switched to SSD because SDcards sooner or later failed me, but HA always finds ways to waste my time

agners, 2021-06-22:
> Sigh, I switched to SSD because SDcards sooner or later failed me, but HA always finds ways to waste my time

Sorry about that :(

Unfortunately SSD has proven to be a bit unreliable with Raspberry Pi as well, just search through the issue tracker... 

This is really odd, as at this point in boot process not much stuff is running. It seems as if a process goes awry already early during boot. 

After three unsuccessful boot the system should go back to the last OS installation (which hopefully shouldn't show that problem anymore). If that doesn't help, you can set `systemd.unit=rescue.target` in `cmdline.txt` of the boot partition drops you to a Linux shell. Maybe it helps doing `ext4.fsck -y /dev/disk/by-label/hassos-data`. Otherwise a new installation is probably your best bet.

bdurrer, 2021-06-23:
@agners Thanks for the input. I managed to get into the console with this, but I don't really have an idea what to do with it.
I forgot to mention that this is already reboot >6. I saw the console once, I think at the third reboot (but it HA never started). 

I am migrating to the NAS now, IMO ha-os is just asking for problems

inverse, 2021-06-23:
has anyone tried the newer builds with a PI + SSD boot? I stuck with `5.12`

User5981, 2021-06-23:
> has anyone tried the newer builds with a PI + SSD boot? I stuck with `5.12`
I had problems with 6.0 not starting (HDD on USB hub) then bought an SSD which had the same problem but I then tried without the USB hub and it booted. 
I'm now on 6.1 without any problem. 
Could it be the USB-micro B cable between the Pi3b and the USB hub which was causing problems⸮ :raised_eyebrow:

agners, 2021-06-24:
FWIW, I tried to reproduce this with a Raspberry Pi 3 Model B+, using the 5.12 32-bit as well as 64-bit image. Installed a fresh installation on a USB mass storage device (Samsung flash drive) and booted it. I then upgraded to 5.13 and in both cases it successfully rebooted.

So it seems *not* to be a general problem, it (once again) seems to depend on the exact environment etc.

It also seems somewhat unlikely that 5.13 makes a difference here, since there is no change in the relevant components (same Raspberry Pi firmware, same U-Boot bootloader and Linux kernel version):
https://github.com/home-assistant/operating-system/compare/5.12...5.13

@inverse @carlhye can you also share the *exact* model of SSD you are using? Also, are you using 32-bit or 64-bit image? Thre was a change in 5.11, so if 5.12 does not work, it could be a problem with the firmware. Are you sure that 5.12 works reliably? (e.g. is software rebooting working reliably on 5.12 multiple times?)

inverse, 2021-06-24:
Hi @agners 

Thanks for investigating - I am using

This SSD: https://www.amazon.de/exec/obidos/ASIN/B07G3KRZBX
This Adaptor: https://www.amazon.de/exec/obidos/ASIN/B06XCV1W97

I'm on a RPi4 8GB now running the 64bit version but still on that `5.12` version... I vaguely remember having weird reboot issues, but tbf it's been up for a long while now so I haven't seen issues.

caesar1987sk, 2021-06-25:
Same issue on external USB HD with own power supply on raspbery pi 3b+.
Latest version of everything and its not possible to start hassio on first, sometimes its needed more than 10 power cicles.
No issue with raspbian.

inverse, 2021-06-26:
@agners just upgraded to `6.1` and it booted first time.

BertrumUK, 2021-07-02:
I was running a Pi3B+ on SSD using Raspberry Pi OS so tried to move over to HAOS starting with 6.1. The Pi would ping but it would not boot using an Orico enclosure. 4.20 worked fine.

Bought a Sabrent EC-UK30, as it works fine with HAOS 6.1 on my Pi4, but it still would not boot the Pi3B+ using 6.1. 

I got it to boot by copying "fixup.dat" & "start.elf" from 4.20 to a 6.1 build and the Pi3B+ starts up and HAOS works as expected.

Leapo, 2021-08-12:
I'm seeing similar issues on a Pi 3B (original). It fails to boot from USB SSD with anything newer than 4.19.

4.20 tries to boot, but locks up. Anything newer than that doesn't even try to boot, and leaves the Pi on a black/blank screen.

inverse, 2021-09-07:
@agners feel free to close this. just went from 6.1 -> 6.3 - all worked

closed 2021-12-06 by inverse