home-assistant/operating-system #1308: Camera snapshots stopped working

MaxDucy, 2021-04-08:
**Hardware Environment**

- [x] Raspberry Pi [4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.12
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**
 
```
2021-04-08 12:17:50 ERROR (MainThread) [homeassistant.components.generic.camera] Error getting new camera image from Back Left: peer closed connection without sending complete message body (received 11236 bytes, expected 11284)
2021-04-08 12:17:51 ERROR (MainThread) [homeassistant.components.generic.camera] Error getting new camera image from Back Right: peer closed connection without sending complete message body (received 12612 bytes, expected 12660)
2021-04-08 12:17:52 ERROR (MainThread) [homeassistant.components.generic.camera] Error getting new camera image from Side: peer closed connection without sending complete message body (received 15072 bytes, expected 15120)
2021-04-08 12:17:53 ERROR (MainThread) [homeassistant.components.generic.camera] Error getting new camera image from Front: peer closed connection without sending complete message body (received 12012 bytes, expected 12060)
```

**Kernel logs:**

**Description of problem:**
 

Camera snapshots stopped working. Been broken since last core updates as well. I've heard that using older version of camera.py will solve the problems.

agners, 2021-04-08:
> Been broken since last core updates as well.
So it used to work with OS 5.12? Seems to be a Home Assistant Core issue then?

MaxDucy, 2021-04-08:
> 
> 
> > Been broken since last core updates as well.
> > So it used to work with OS 5.12? Seems to be a Home Assistant Core issue then?

I'm sorry, I wasn't clear. What I meant was since OS 5.12. So it used to work with 5.11.

agners, 2021-04-08:
Can you retry with OS 5.11 with the current setup?

You can downgrade using `ha os update --version=5.11`.

MaxDucy, 2021-04-08:
> 
> 
> Can you retry with OS 5.11 with the current setup?
> 
> You can downgrade using `ha os update --version=5.11`.

I've downgraded to 5.11 and now for some reason now it doesn't work anymore.

I went through HA Community forum threads and one of the member said that using this version of camera.py solved the issue for them. [https://github.com/home-assistant/core/blob/e1060f154e1dd70e80276eeedef2eb039ba3af29/homeassistant/components/generic/camera.py](url)