home-assistant/operating-system #1309: Fresh install on new Intel NUC Panther Canyon doesn't show up on network

mdelecate, 2021-04-08:
**Hardware Environment**

- [ ] Raspberry Pi [1/2/3/4]
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [X] Generic x86-64 **(Intel NUC NUC11PAQi7 Panther Canyon)**
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [X] Fresh installation of release 5.13
- [ ] Updated from version x.y
- Additional information (if accessible):

**Supervisor logs:**

**Journal logs:**

**Kernel logs:**

**Description of problem:**
Home Assistant OS running on an Intel NUC NUC11PAQi7 Panther Canyon does not show up on the network. 
Working ethernet cable is plugged in, http://homeassistant.local:8123 can’t be reached, network scanner can't discover the device. 

I've tested Ubuntu Desktop on the same NUC to verify that WiFi and Ethernet are both working properly to rule out a hardware issue. 

Install procedure was as follows: 
1. Flashed hassos_intel-nuc-5.13.img.xz to the M.2 SSD with Balena Etcher and an external M.2 USB-C enclosure.
2. Connected a working Ethernet cable
3. Waited >30 minutes after booting the NUC

A connected monitor displays:
```
Welcome to Home Assistant
home assistant login: [ 62.6480831] hdaudio hdaudioC0D2: Unable to bind the codec
```

Logging in with root and entering “network info” displays:
```
docker: 
 address: 172.30.32.0./23
 dns: 172.30.32.3
 gateway: 172.30.32.1
 interface: hassio
host_internet: false
interfaces: []
supervisor_internet: false
```

agners, 2021-04-08:
Hm, it seems that no IP has been obtained.

Can you post the network information from the OS shell? You can login using `root` and then `login` to access the OS shell. In the OS shell, use:

```
ip addr
```

Also kernel messages might reveal something:
```
dmesg
```

mdelecate, 2021-04-08:
Sure, although I can only post photos (since I can’t copy paste the output to here): 

ip addr 
[image]

dmesg (only the last visible part) 
[image]

agners, 2021-04-08:
Hm, it seems that Home Assistant OS fails to load a Ethernet driver for that device. I wonder what driver Ubuntu is using.

Can you maybe upload the output of `lsmod` and maybe also `dmesg` from Ubuntu? You can redirect the output to a file using `dmesg > nuc-panther-dmesg.txt` and attach the output as a file.

agners, 2021-04-08:
Hm, it seems that a I-225-V is used in that particular device. The driver `CONFIG_IGC` is currently not enabled.

At this point the output of `lspci` would be more helpful.

mdelecate, 2021-04-08:
Here's the output of lspci in Ubuntu: 

```
00:00.0 Host bridge: Intel Corporation 11th Gen Core Processor Host Bridge/DRAM Registers (rev 01)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics (rev 01)
00:06.0 PCI bridge: Intel Corporation 11th Gen Core Processor PCIe Controller (rev 01)
00:07.0 PCI bridge: Intel Corporation Tiger Lake-LP Thunderbolt PCI Express Root Port #0 (rev 01)
00:07.3 PCI bridge: Intel Corporation Tiger Lake-LP Thunderbolt PCI Express Root Port #3 (rev 01)
00:08.0 System peripheral: Intel Corporation Device 9a11 (rev 01)
00:0d.0 USB controller: Intel Corporation Tiger Lake-LP Thunderbolt USB Controller (rev 01)
00:0d.2 USB controller: Intel Corporation Tiger Lake-LP Thunderbolt NHI #0 (rev 01)
00:0d.3 USB controller: Intel Corporation Tiger Lake-LP Thunderbolt NHI #1 (rev 01)
00:14.0 USB controller: Intel Corporation Tiger Lake-LP USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
00:14.2 RAM memory: Intel Corporation Tiger Lake-LP Shared SRAM (rev 20)
00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201 (rev 20)
00:15.0 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP Serial IO I2C Controller #0 (rev 20)
00:16.0 Communication controller: Intel Corporation Tiger Lake-LP Management Engine Interface (rev 20)
00:1c.0 PCI bridge: Intel Corporation Device a0bc (rev 20)
00:1c.6 PCI bridge: Intel Corporation Device a0be (rev 20)
00:1f.0 ISA bridge: Intel Corporation Tiger Lake-LP LPC Controller (rev 20)
00:1f.3 Audio device: Intel Corporation Tiger Lake-LP Smart Sound Technology Audio Controller (rev 20)
00:1f.4 SMBus: Intel Corporation Tiger Lake-LP SMBus Controller (rev 20)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Tiger Lake-LP SPI Controller (rev 20)
01:00.0 Non-Volatile memory controller: Intel Corporation SSD 600P Series (rev 03)
58:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
59:00.0 Ethernet controller: Intel Corporation Intel(R) Ethernet Controller I225-V (rev 03)
```

doronazl, 2021-05-07:
also been waiting for this matter to be resolved, if theres any log i can provide that will help please let me know
im on a NUC11PAHI3

agners, 2021-05-19:
@mdelecate @doronazl the driver is enabled in 6.0.rc1. I assume with that networking should work.

Please test and reopen in case you still have no network using 6.0.rc1 (or later) releases.

closed 2021-05-19 by agners

mdelecate, 2021-06-19:
Thanks a lot for your work, it's working perfectly on 6.0!