home-assistant/operating-system #1320: Both Dropbear and getty are ran after docker, which makes it impossible to troubleshoot if docker fails

sandhose, 2021-04-10:
**Hardware Environment**

- [x] Raspberry Pi 3
- [ ] ODROID [C2/C4/N2(+)/XU4]
- [ ] ASUS Tinker [S]
- [ ] Generic x86-64 (like Intel NUC)
- [ ] OVA (Open Virtualization Applicance, on Intel NUC or any other hardware, please add the Hypervisor you are using)

**Home Assistant OS release:**
- [ ] Fresh installation of release x.y
- [x] Updated from version 5.10 -> 5.13

**Supervisor logs:**
N/A

**Journal logs:**
[image]

**Kernel logs:**
N/A

**Description of problem:**
I'm in a state where Docker won't start because processes get killed with OOM. Because Docker does not start, it does not start neither the SSH server or the `getty` service, making it very hard to debug anything.
Sorry for the bad photo of the logs. I'm helping a friend remotely, this is the best I could get.

This is partly because [`getty@.service`](https://github.com/home-assistant/operating-system/blob/be2a64f4d23a4edf09713ccfc256721cf4769a75/buildroot-external/rootfs-overlay/etc/systemd/system/getty%40.service.d/hassos.conf) is started after `hassos-supervisor.service`, which itself is started only after the [`multiuser` stage](https://github.com/home-assistant/operating-system/blob/be2a64f4d23a4edf09713ccfc256721cf4769a75/buildroot-external/rootfs-overlay/usr/lib/systemd/system/hassos-supervisor.service#L21).
[`dropbear.service` is started after `docker.service`](https://github.com/home-assistant/operating-system/blob/be2a64f4d23a4edf09713ccfc256721cf4769a75/buildroot-external/rootfs-overlay/etc/systemd/system/dropbear.service.d/docker.conf#L2-L3).

Maybe it would be worth at least having the SSH daemon starting earlier to help debugging in such cases? Or even better, a way to get a TTY after some time if the system does not get up?

agners, 2021-04-11:
Hm, this ordering is likely to make sure the CLI container is ready. But even when ordering after Docker, this is actually not guaranteed, since the container takes a while until it is up and running. We also have a mechanism to fallback to a shell when the CLI container is not up. For troubleshooting etc. this is much more useful. 

I'll check if that ordering is really necessary and drop it to make at least Dropbear available earlier.

What you can do to access the system is using `systemd.unit=rescue.target` in the `cmdline.txt` configuration file to let systemd start in a rescue shell. Not sure if that is really helpful though.

sandhose, 2021-04-12:
> What you can do to access the system is using `systemd.unit=rescue.target` in the `cmdline.txt` configuration file to let systemd start in a rescue shell. Not sure if that is really helpful though.

That was one of my leads to debug that, unfortunately it tries to run `/sbin/sulogin`, which does not exist, so it resumes the normal boot process

sandhose, 2021-04-12:
Some update here: adding [`systemd.debug_shell`](https://www.freedesktop.org/software/systemd/man/systemd-debug-generator.html) to the cmdline helped a bit more: it spawns a shell early on `tty9`. For some reason, the `overlays` file on my boot partition were missing/corrupted, which made the pi boot with only ~100Mb of RAM available, hence the OOM kills. Restoring the boot parition from the stock image helped resolving the issue.