home-assistant/operating-system #1324: Login (onboard) in endless loop without possibility to login

dadaloop82, 2021-04-13:
**Hardware Environment**
- Home Assistant OS 5.13 (2021.3.4)
- Home Assistant in Microsoft Hyper-V Virtual Machine
- aarch: amd64
- image: homeassistant/qemux86-64-homeassistant

**Home Assistant OS release:**
- [X] Updated from version 5.12

**Description of problem:**
I updated the system and I can no longer log in, with any browser (I've tried Chrome, Firefox and Edge).
Basically the login screen (onboarding.html) keeps reloading in an endless loop.

I don't have any error messages in the logs.

I have access to the virtual machine via the console, but not via SSH. I don't know what to do, but I don't want to lose all the configuration.

As long as the session (the connection cookie) was there it worked, as soon as the cookie expired it no longer worked.
I don't know if it's a configuration issue, but it was working fine before.

Please advise me how to fix this problem.
If you need more information please ask.

dadaloop82, 2021-04-14:
## UPDATE ##
This morning I ran 

`has update`

from the HassOS console of the virtual machine, the system has updated to the 2021.4.4 version and now it seems to work correctly, the problem seems to be solved.

agners, 2021-04-15:
I don't think this was related to the OS update, probably another component. But glad it has been resolved. Thanks for the update.

closed 2021-04-15 by agners