home-assistant/operating-system #1330: Feature request: Support for additional internal drives for media storage on NUC platform

jorgen666, 2021-04-16:
Hi I'm running latest release off Home Assistant operation system, 5.13 on a intel i3 desktop with on SSD as main drive, and wish to add an exstra very large HDD  for media storage and the plex addon. 

It would have been very nice to be able to mount additional internal discs the normal linux way for this purpose.

Moondevil-ha, 2021-12-09:
+1 for this. I have used and adapted a udev workaround. Could this be integrated into the install process, or a configuration item/flag be introduced to allow a drive to be mounted to /media? https://gist.github.com/eklex/c5fac345de5be9d9bc420510617c86b5#gistcomment-3921831

Tekno-man, 2022-01-27:
This really applies to the OS on any hardware or VM. There is a growing trend to want to be able to do this as HA becomes the central point for peoples needs https://community.home-assistant.io/t/solved-mount-usb-drive-in-hassio-to-be-used-on-the-media-folder-with-udev-customization/258406/49 
The udev solution posted by @Moondevil-ha has worked for quite a few people but it would be good to have this integrated some how if possible.

sairon, 2026-09-17:
This is being worked on in [home-assistant/supervisor#7137](https://github.com/home-assistant/supervisor/pull/7137), which adds support for mounting internal and USB drives for media, backups and shared storage. The PR is still open and has not been merged yet, but will likely make its way soon. This functionality is managed by Supervisor, so there is nothing to change in HAOS for this request.

Closing as not planned in the OS repository, please follow the Supervisor PR for progress.

closed 2026-09-17 by sairon