home-assistant/operating-system #1331: Unable to edit Add-On Config in fresh VM install

feueru, 2021-04-16:
**Hardware Environment**

ESXi 7.0.1

**Home Assistant OS release:**
https://github.com/home-assistant/operating-system/releases/download/5.13/hassos_ova-5.13.ova

**Description of problem:**

1. Create a new VM using default settings in ESXi 7.0.1 using https://github.com/home-assistant/operating-system/releases/download/5.13/hassos_ova-5.13.ova, with all default settings
2. When VM starts complete initial setup and install an Add On such as Duck DNS
3. Edit Add On Configuration in yaml mode
4. When configuration is saved, it reverts to default settings

I've  been able to re-create this multiple times with mutable Add Ons with multiple fresh VMs 
but some Add Ons work as expected such as Terminal & SSH

agners, 2021-04-18:
You are editing the configuration through the Supervisor fronted I assume? It sounds more like the Supervisor not saving the configuration for some reason. Can you check if the file has the changes on hard disk? Login using `root`, get shell access using `login`, and then check the content of the `options.json` file using `cat /mnt/data/supervisor/addons/data/core_<youraddon>/options.json`.

feueru, 2021-04-18:
Thanks for the reply. That is correct, the initial issue was though the web ui supervisor frontend on multiple browsers. I'm not seeing options.json in the addons folders.

2/3 addons on this install have changes made to the options, 1 is experiencing the issue.

[image]

agners, 2021-04-18:
I'd not recommend to edit the json directly, as it does not get read back by Supervisor/might get written delayed. I meant it more to check if saving is actually done by the Supervisor.

Are you able to edit the settings for your third addon when making sure that no other tabs are open?

feueru, 2021-04-18:
@agners, 
No i'm not able to change configuration for quite a few add ons at all. I think the gif in my last post illustrates the issue well. Its very easy to reproduce,  just spin up a VM with the .ova and install duckdns,zwave js, and/or a few other addons, and configuration it only changeable in a few of them.

agners, 2021-04-19:
This seems like a Supervisor or even Frontend issue. Moving.