home-assistant/operating-system #1335: Operating system not found on VMWare Workstation Player 16

whereizben, 2021-04-19:
**Hardware Environment**
Running an Intel NUC with Ubuntu 18.04 LTS with VMWare Workstation 16 Player

**Home Assistant OS release:**
Downloaded hassos_ova-5.13.vmdk direct from home assistant site

I followed the install instructions (as closely as I could, since they seem to be for Workstation 15, not 16) from the Home Assistant site: https://www.home-assistant.io/installation/linux - I always get an error as soon as it tries to boot "Operating system not found"

There is no place in the VMWare Workstation 16 settings for the VM to change the firmware type to UEFI, and as far as selecting the image, I had to do that after creating the VM (not during the setup process). When selecting the vmdk, it asks "Convert existing virtual disk to newer format?" and states that if I do so, it won't work with older VMs. I've tried converting as well as keeping the existing format, as well as setting the virtual HD to be SCSI, SATA and IDE. 

None of these things result in a different outcome - in all cases, it always says "Operating system not found" on boot.

Any thoughts? I've tried to find the bundle for VMWare Workstation 15 Player but had no luck either. I've got HA Core installed on my Ubuntu machine, and really want to get the OS installed in a VM (so that I can get Supervisor).

agners, 2021-04-19:
Hm, my best guess is that its UEFI related, like you already assumed. What you could try is create again and make sure to set to UEFI before starting the first time (the instructions say "After creation of VM go to “Settings” and “Options” then “Advanced” and select “Firmware type” to “UEFI”.".). Converting the disk image should not be necessary.

whereizben, 2021-04-19:
Unfortunately in VMWare Workstation Player 16 there doesn't even seem to be
a setting (anywhere) to select the firmware type. I looked before trying to
start it as well as after.

bert269, 2021-05-03:
I get exactly the same problem, 14days later - have you found a solution? I am using VMWare Workstation Pro 16.x but still cannot get this HA_5.13 system to boot. I have tried everything.

What are we missing?

whereizben, 2021-05-03:
I never figured this out, sadly. I ended up installing VBoxManage on my
Ubuntu machine (NOT via the Ubuntu software app but directly from the
VirtualBox site, as the former didn't work to be able to autorun on
startup).
Happy to give thoughts on VBoxManage if you need, and good luck with VMWare!

closed 2021-05-19 by agners

alexabreupassos, 2022-08-23:
This change to vmx config file saved my day.
**firmware = "efi"**