home-assistant/operating-system #1339: HA not booting in KVM

MrksHfmn, 2021-04-27:
**Description of problem:**
I am trying to boot HA in a VM (KVM), however it only boots up to "Loading MS-DOS executable '/mnt/system/bzImage'" and stops there (I think). Also no IP is obtained from the DHCP server. The PCI device is a network card that is passed through via PCI passthrough.

**Hardware Environment**
- [ x] Generic x86-64 (like Intel NUC)

**Home Assistant OS release:**
hassos_ova-5.13.qcow2.xz

**virsh import**
```
virt-install \
        --import \
        --name home-assistant \
        --memory 4096 \
        --vcpus 4 \
        --cpu host \
        --disk home-assistant.qcow2,format=qcow2,bus=virtio \
        --nonetworks \
        --os-type=linux \
        --os-variant=debian9 \
        --graphics none \
        --noautoconsole \
        --boot uefi \
        --host-device=pci_0000_01_00_1
```

**virsh dumpxml home-assistant**

```
<domain type='kvm' id='17'>
  <name>home-assistant</name>
  <uuid>13357bd0-000c-496c-95af-602ebdb7999c</uuid>
  <metadata>
    <libosinfo:libosinfo xmlns:libosinfo="http://libosinfo.org/xmlns/libvirt/domain/1.0">
      <libosinfo:os id="http://debian.org/debian/9"/>
    </libosinfo:libosinfo>
  </metadata>
  <memory unit='KiB'>4194304</memory>
  <currentMemory unit='KiB'>4194304</currentMemory>
  <vcpu placement='static'>4</vcpu>
  <resource>
    <partition>/machine</partition>
  </resource>
  <os>
    <type arch='x86_64' machine='pc-q35-5.2'>hvm</type>
    <loader readonly='yes' type='rom'>/usr/share/ovmf/OVMF.fd</loader>
    <nvram>/var/lib/libvirt/qemu/nvram/home-assistant_VARS.fd</nvram>
    <boot dev='hd'/>
  </os>
  <features>
    <acpi/>
    <apic/>
  </features>
  <cpu mode='custom' match='exact' check='full'>
    <model fallback='forbid'>Snowridge</model>
    <vendor>Intel</vendor>
    <feature policy='require' name='ss'/>
    <feature policy='require' name='vmx'/>
    <feature policy='require' name='pdcm'/>
    <feature policy='require' name='hypervisor'/>
    <feature policy='require' name='tsc_adjust'/>
    <feature policy='require' name='mpx'/>
    <feature policy='require' name='rdpid'/>
    <feature policy='require' name='md-clear'/>
    <feature policy='require' name='stibp'/>
    <feature policy='require' name='xsaves'/>
    <feature policy='require' name='ibpb'/>
    <feature policy='require' name='amd-stibp'/>
    <feature policy='require' name='amd-ssbd'/>
[... 151 lines cut ...]

barebox 2020.04.0 #1 Thu Apr 1 15:58:01 UTC 2021

Board: barebox EFI payload
EFI v2.70 by EDK II v65536
EFI Event timer too slow freq = 100 Hz
acpi2: Found XSDT (OEM: BOCHS BX) with 4 entries
efi-fs handle-000000007e9cbb98: mounted on /boot
malloc space: 0x30000000 -> 0x3fffffff (size 256 MiB)
state: New state registered 'state'
state: Using bucket 0@0x00000000
description: UEFI Misc Device
path: pci_root(0)/Pci(0x1,0x2)/Pci(0x0,0x0)
description: EFI Internal Shell
path: Path(132,7,0xc9bdb87cebf8344faaea3ee4af6516a10000000084061800)/Path(132,6,0x83a5047c3e9e1c4fad65e05268d0b4d100000000ffff0800)
environment load /efivars/barebox-env-5b91f69c-8b88-4a2b-9269-5f1d802b5175: No such file or directory
Maybe you have to create the partition.
running /env/bin/init...
- Hit m for menu or wait for autoboot -
Booting entry 'bootchooser'
Booting entry 'system0'
ext4 ext40: EXT2 rev 1, inode_size 128, descriptor size 32

Loading MS-DOS executable '/mnt/system/bzImage'
```

agners, 2021-04-28:
Hm, I test HAOS quite regularly using KVM here. I can't spot a problem really. This is my XML:

```
<domain type="kvm">
  <name>HomeAssistant</name>
  <uuid>cb1f810b-d1cd-443e-913c-b019b207b82a</uuid>
  <memory unit="KiB">4193280</memory>
  <currentMemory unit="KiB">4193280</currentMemory>
  <vcpu placement="static">3</vcpu>
  <os>
    <type arch="x86_64" machine="pc-q35-5.1">hvm</type>
    <loader readonly="yes" type="pflash">/usr/share/edk2-ovmf/x64/OVMF_CODE.fd</loader>
    <nvram>/var/lib/libvirt/qemu/nvram/HomeAssistant_VARS.fd</nvram>
    <boot dev="hd"/>
  </os>
  <features>
    <acpi/>
    <apic/>
    <vmport state="off"/>
  </features>
  <cpu mode="host-model" check="partial"/>
  <clock offset="utc">
    <timer name="rtc" tickpolicy="catchup"/>
    <timer name="pit" tickpolicy="delay"/>
    <timer name="hpet" present="no"/>
  </clock>
  <on_poweroff>destroy</on_poweroff>
  <on_reboot>restart</on_reboot>
  <on_crash>destroy</on_crash>
  <pm>
    <suspend-to-mem enabled="no"/>
    <suspend-to-disk enabled="no"/>
  </pm>
  <devices>
    <emulator>/usr/bin/qemu-system-x86_64</emulator>
    <disk type="file" device="disk">
      <driver name="qemu" type="qcow2" discard="unmap"/>
      <source file="/var/lib/libvirt/images/hassos_ova.qcow2"/>
      <target dev="vdc" bus="virtio"/>
      <address type="pci" domain="0x0000" bus="0x04" slot="0x00" function="0x0"/>
    </disk>
    <controller type="sata" index="0">
      <address type="pci" domain="0x0000" bus="0x00" slot="0x1f" function="0x2"/>
    </controller>
    <controller type="pci" index="0" model="pcie-root"/>
    <controller type="pci" index="1" model="pcie-root-port">
      <model name="pcie-root-port"/>
      <target chassis="1" port="0x10"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x02" function="0x0" multifunction="on"/>
    </controller>
    <controller type="pci" index="2" model="pcie-to-pci-bridge">
      <model name="pcie-pci-bridge"/>
      <address type="pci" domain="0x0000" bus="0x01" slot="0x00" function="0x0"/>
    </controller>
    <controller type="pci" index="3" model="pcie-root-port">
      <model name="pcie-root-port"/>
      <target chassis="3" port="0x11"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x02" function="0x1"/>
    </controller>
    <controller type="pci" index="4" model="pcie-root-port">
      <model name="pcie-root-port"/>
      <target chassis="4" port="0x12"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x02" function="0x2"/>
    </controller>
    <controller type="pci" index="5" model="pcie-root-port">
      <model name="pcie-root-port"/>
      <target chassis="5" port="0x13"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x02" function="0x3"/>
    </controller>
    <controller type="pci" index="6" model="pcie-root-port">
[... 57 lines cut ...]
      <image compression="off"/>
    </graphics>
    <sound model="ich9">
      <address type="pci" domain="0x0000" bus="0x00" slot="0x1b" function="0x0"/>
    </sound>
    <video>
      <model type="qxl" ram="65536" vram="65536" vgamem="16384" heads="1" primary="yes"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x01" function="0x0"/>
    </video>
    <redirdev bus="usb" type="spicevmc">
      <address type="usb" bus="0" port="2"/>
    </redirdev>
    <redirdev bus="usb" type="spicevmc">
      <address type="usb" bus="0" port="3"/>
    </redirdev>
    <memballoon model="virtio">
      <address type="pci" domain="0x0000" bus="0x05" slot="0x00" function="0x0"/>
    </memballoon>
    <vsock model="virtio">
      <cid auto="yes" address="3"/>
      <address type="pci" domain="0x0000" bus="0x06" slot="0x00" function="0x0"/>
    </vsock>
  </devices>
</domain>
```

agners, 2021-05-26:
Ok, looked a bit closer into this issue, and the relevant difference here is that your configuration doesn't has a video card section:

```
    <video>
      <model type="qxl" ram="65536" vram="65536" vgamem="16384" heads="1" primary="yes"/>
      <address type="pci" domain="0x0000" bus="0x00" slot="0x01" function="0x0"/>
    </video>
```

The virsh console is the serial console, and it seems that the boot loader has serial console enabled by default. However, Home Assistant OS only has tty1 as default console in the ova machine. I think it should be fairly safe to enable serial console by default. I'll do some tests and create a PR.

agners, 2021-06-17:
Assuming this is resolved with the enabled serial console.

closed 2021-06-17 by agners

rolf-tx, 2023-08-12:
@agners 
I'm using homeassistant x86 OS.
I would like to use a USB serial cable and have console output to it.
I have "console=tty1  console=ttyUSB0,115200n8" to the cmdline.txt but get nothing on the serial.
```
[    0.000000] Command line: BOOT_IMAGE=(hd1,gpt2)/bzImage root=PARTUUID=8d3d53e
3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 net.nam
ing-scheme=v250 systemd.machine_id=1233797a03a14c4e9ba0926b05bac010 fsck.repair=
yes console=tty1 console=ttyUSB0,115200 rauc.slot=A
```

I'm seeing tty1 being setup as console, but not a similar line or ttyUSB0.
`  [    0.065342] printk: console [tty1] enabled`

Is the current version of homeassistant OS compiled with USB serial console enabled?
 CONFIG_USB_SERIAL_CONSOLE=y