home-assistant/operating-system #1340: Docker compilation error: version `GLIBC_2.33' not found

TopSWorld, 2021-04-28:
root@32f514ea6016:/build# make ova
config ova
make -C /build/buildroot BR2_EXTERNAL=/build/buildroot-external "ova_defconfig"
make[1]: Entering directory '/build/buildroot'
/build/buildroot/output/build/buildroot-config/conf: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /build/buildroot/output/build/buildroot-config/conf)
make[1]: *** [Makefile:1030: ova_defconfig] Error 1
make[1]: Leaving directory '/build/buildroot'
make: *** [Makefile:31: ova-config] Error 2
root@32f514ea6016:/build#

agners, 2021-04-28:
Hm, I haven't seen that issue. Is that a fresh checkout, clean output directory?

topsworld, 2021-04-29:
thanks, I tried to clear the output directory( `make clean` ),then compile it using Docker ( `sudo ./scripts/enter.sh` -> `make ova`) and it works.

elsbrock, 2023-07-23:
FYI for anyone else stumbling over this: This happens if you're trying to build a specific board on a different architecture than the builder (eg. building X86_64 on M1/ARM).