home-assistant/operating-system #1350: USB Sound card does not appear in HA

philibertc, 2021-05-03:
### The problem

When a sound card is connected, it appears well when `dmesg` is run but does not appear anywhere on HA (Audio Add-ons, `ha audio info`).

### What is version of Home Assistant Core has the issue?

core-2021.4.6

### What type of installation are you running?

Home Assistant OS

### Integration causing the issue

system

### Anything in the logs that might be useful for us?

```txt
`ha audio logs` output:
https://imgur.com/a/K9nddQg
I can't copy the text inside the terminal add-on.
```

balloob, 2021-05-03:
What sound card and hardware platform?

philibertc, 2021-05-03:
@balloob Just a classic usb sound card with optical and coaxial outputs.
It is working perfectly on Ubuntu and other any (linux or not) distro.

frenck, 2021-05-03:
@philibertc Please, be specific... We can't do anything with just that information.

philibertc, 2021-05-04:
@frenck Hello, I don't know more than it's a normal usb card, but the bug is here with all usb sound cards.
I have this from `dmesg` [image]

philibertc, 2021-05-04:
This bug is not for any particular extension or integration. It's at the system level.

frenck, 2021-05-04:
> but the bug is here with all usb sound cards.

The reason we ask, is that that statement is incorrect. Could you please provide the exact model/type/manufacturer/version of the soundcard you are using?

philibertc, 2021-05-04:
All is in the logs.
I don't know more than "C-Media USB Sound Device"
The chip is a CM102-A+

philibertc, 2021-05-04:
http://www.aqvox.de/download/CM102S-USB-DAC10.pdf

philibertc, 2021-05-04:
> is that that statement is incorrect.
@frenck Yes, I've corrected this : The problem is here with all the sound cards that I've tested.

agners, 2021-05-06:
I assume this is Home Assistant OS in a virtual machine?

The sound card you mentioned is supported by `CONFIG_SND_USB_AUDIO`, which is enabled in the kernel config. From your `dmesg` snipped it also seems that the USB sound card driver got loaded correctly. So I assume this is a problem on a higher level (user space).

@pvizeli do we have USB audio handled in general in PulseAudio? Maybe something wrong with the PulseAudio configuration/USB audio device not available in the Audio container?

pvizeli, 2021-05-07:
Can you post the `ha audio logs` after you plug it in?